ucb-bar/chipyard #1209: Private or missing submodules

esromneb, 2022-08-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

main branch
Hash: [684a02a](https://github.com/ucb-bar/chipyard/commit/684a02a10f093860511c23bc8da52c198cd5e879)

### OS Setup

Ex: Output of `uname -a` and `lsb_release -a`
Linux 5.15.0-43-generic #46~20.04.1-Ubuntu SMP Thu Jul 14 15:20:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

### Other Setup

none

### Current Behavior

Cannot checkout submodules. it appears that

https://github.com/ucb-bar/hammer-mentor-plugins
and
https://github.com/ucb-bar/hammer-synopsys-plugins

are both missing or got turned private

### Expected Behavior

submodules that exist

harrisonliew, 2022-08-16:
Indeed, they are private. Please refer to the documentation and request access to these submodules: https://chipyard.readthedocs.io/en/stable/VLSI/Basic-Flow.html

closed 2022-08-16 by harrisonliew

timsnyder-siv, 2022-08-17:
@esromneb if you were following the FireSim tutorial docs, I'm surprised that the scripts were trying to populate those submodules.  Can you provide a pointer to the docs you were following?  Was it https://docs.fires.im/en/stable/Initial-Setup/Setting-up-your-Manager-Instance.html#setting-up-the-firesim-repo or something else?

esromneb, 2022-08-18:
The docs here don't give a specific git tag. (https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html) I was "all over the place" trying to figure out combinations of the AMI and git tag that would work. Once I realized that some of the repos had moved (the libdwarf repo moved as referred to here https://github.com/firesim/firesim/issues/998) I started to check things out manually. I may have manually ran into the private submodule issue which the setup scripts specifically avoid. Sorry I'm not more clear about this but I will need to revisit the whole process from a fresh start once I can get access to the 1.11.1 AMI (https://github.com/firesim/firesim/issues/1172) or maybe I can follow the docs once they are updated. Sorry again if this isn't exactly a clear answer