ucb-bar/chipyard #1210: How to add spi interface

Usutatsu, 2022-08-24:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

 I want to be able to connect my new Pmod module to my FPGA board. 
The PMOD module to be connected must be connected via SPI interface.
So I want to connect the new SPI interface with a vacant PMOD connector.
Does anyone know which part of the description I need to change to create the new SPI interface?

### Motivating Example

I would be very happy if you could tell me.

jerryz123, 2022-09-02:
This is better directed at the mailing list.

Usutatsu, 2022-09-02:
Of course I have asked about this issue on my mailing list.
However, no one was willing to help me on this issue.
So I am asking the question on Github.