ucb-bar/chipyard #1212: Errors with OpenROAD P&R

therbom, 2022-08-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.7.1
Hash: 9926642

### OS Setup

```terminal
$ uname -a
Linux [redacted] 3.10.0-1160.15.2.el7.x86_64 #1 SMP Wed Feb 3 15:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	CentOS
Description:	CentOS Linux release 7.9.2009 (Core)
Release:	7.9.2009
Codename:	Core
```

### Other Setup

Based on:
* https://chipyard.readthedocs.io/en/1.7.1/VLSI/Sky130-Tutorial.html
* https://docs.hammer-eda.org/en/latest/Examples/openroad-nangate45.html
* #1183
* #608

### Current Behavior

I am trying to perform place-and-route with OpenROAD with both the Sky130 and Nangate45 PDKs, but both do not succeed.

#### Sky130
I had to make some changes:
* In `vlsi/example-vlsi-sky130`, undo the removal of the `place_bumps` hook.
* In `vlsi/hammer/src/hammer-vlsi/synthesis/yosys/__init__.py`, check for "not driver_cells" instead of "driver_cells is None" (because an empty list will be returned)

Then, `openroad` runs and fails with:
```terminal
# Floorplan automatically generated from HAMMER
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_0 -origin { 50 100 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_1_0 -origin { 50 700 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_2_0 -origin { 50 1300 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_3_0 -origin { 50 1900 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_4_0 -origin { 1000 1900 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_5_0 -origin { 1000 1300 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_6_0 -origin { 1000 700 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_7_0 -origin { 1000 100 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/mem_0_0 -origin { 3700 100 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/mem_1_0 -origin { 3700 700 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/frontend/icache/tag_array/tag_array_ext/mem_0_0 -origin { 3000 100 } -orient R0 -status FIRM
place_cell -inst_name system/tile_prci_domain/tile_reset_domain/tile/ptw/l2_tlb_ram/l2_tlb_ram_ext/mem_0_0 -origin { 1900 1900 } -orient R0 -status FIRM
[ERROR PAD-0197] Instance system/tile_prci_domain/tile_reset_domain/tile/ptw/l2_tlb_ram/l2_tlb_ram_ext/mem_0_0 not in the design, -cell must be specified to create a new instance.
Error: floorplan.tcl, 14 PAD-0197
```

After uncommenting in `floorplan.tcl` what `openroad` believes is not in the design, `openroad` crashes with:
```terminal
[INFO RSZ-0026] Removed 1574 buffers.
Found 11 macro blocks.
Using 1u default distance from corners.
Using 2 tracks default min distance between IO pins.
/.../gcc-9.1.0/include/c++/9.1.0/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace: ...
```

#### Nangate45
I had to make some changes:
* In `vlsi/example-vlsi`, undo the removal of the `place_bumps` hook.
* In `vlsi/hammer/src/hammer-vlsi/technology/nangate45/defaults.yml`, change the stackup from `nangate45_3Ma_3Mb_2Mc_2Md` to `nangate45_3Ma_2Mb_2Mc_2Md` following the tech json.
* Manually copy `sky130A.lyt` to the `tech-sky130-cache`.

Then, at sub-step `place_tap_cells`:
```terminal
Traceback (most recent call last):
  File "./example-vlsi", line 62, in <module>
    ExampleDriver().main()
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1471, in main
    sys.exit(self.run_main_parsed(vars(parser.parse_args(args))))
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 1376, in run_main_parsed
    output_config = action_func(driver, errors.append)  # type: Optional[dict]
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/cli_driver.py", line 488, in action
    list(extra_hooks or []))
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/driver.py", line 991, in run_par
    run_succeeded = self.par_tool.run(hooks_to_use)
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/hammer_tool.py", line 116, in run
    if not self.run_steps(self.steps, hook_actions):
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/hammer_tool.py", line 631, in run_steps
    func_out = step.func(self)  # type: bool
  File "/chipyard-171/vlsi/hammer/src/hammer-vlsi/hammer_vlsi/hammer_tool.py", line 675, in wrapper
    return func.__func__(x)  # type: ignore # no type stub for builtin __func__
  File "hammer/src/hammer-vlsi/par/openroad/__init__.py", line 408, in place_tap_cells
    endcap_cell=endcap_cells[0].name[0]
IndexError: list index out of range
```

### Expected Behavior

P&R should finish for both Sky130 and Nangate45.

harrisonliew, 2022-09-16:
We are in the process of fixing some known issues with OpenROAD with Sky130 with macro placement. You can follow along here: ucb-bar/hammer#666.

Unfortunately, we broke nangate45 when we updated the OpenROAD plugin. This is because the old plugin basically only called the OpenROAD-flow-scripts Makefiles, whereas the new one somewhat replaces it.

harrisonliew, 2022-09-29:
Closed by #1226

closed 2022-09-29 by harrisonliew