ucb-bar/chipyard #1215: Rename generators

jerryz123, 2022-09-02:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

N/A

### Other Setup

N/A

### Current Behavior

Current generator directories:
- sifive-blocks
- sifive-cache

### Expected Behavior

Rename to
- rocket-chip-blocks
- rocket-chip-cache

abejgonzalez, 2022-09-19:
@jerryz123 Can you do this part of the RC bump #1216

jerryz123, 2022-09-19:
I don't want to complicate the cy/fs deps further. This can be done in a separate PR.