ucb-bar/chipyard #1263: Removing c++ top level with verilator 5.0

tianrui-wei, 2022-10-20:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Verilator 5.0 introduces timing support. As such, a verilog top level testbench can be used instead of maintaining a separate c++ top level wrapper. After we bump verilator in conda we could remove https://github.com/ucb-bar/chipyard/blob/main/generators/chipyard/src/main/resources/csrc/emulator.cc

cc @abejgonzalez 

### Motivating Example

Removing https://github.com/ucb-bar/chipyard/blob/main/generators/chipyard/src/main/resources/csrc/emulator.cc

jerryz123, 2022-11-22:
Lets get rid of the top-level verilator wrapper in rocket-chip as well.

jerryz123, 2023-03-15:
Resolved by #1398

closed 2023-03-16 by tianrui-wei