ucb-bar/chipyard #1267: Build ESP Tools fail on ARM64 Linux

willliu, 2022-11-06:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: -----

Follow Gemmini tutorial:

git checkout 117624d8eea27bafd613eec09e9b9b3e31239e08

### OS Setup

`uname -a`
Linux ubuntu-linux-22-04-desktop 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:04:57 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

`lsb_release -a`
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

`conda list`
_openmp_mutex             4.5                       2_gnu    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
brotlipy                  0.7.0           py310h761cc84_1005    conda-forge
bzip2                     1.0.8                hf897c2e_4    conda-forge
ca-certificates           2022.9.24            h4fd8a4c_0    conda-forge
cachecontrol              0.12.12            pyhd8ed1ab_1    conda-forge
cachy                     0.3.0              pyhd8ed1ab_1    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310hf0c4615_2    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3           unix_pyhd8ed1ab_2    conda-forge
click-default-group       1.2.2              pyhd8ed1ab_1    conda-forge
clikit                    0.6.2              pyh9f0ad1d_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     22.9.0          py310h4c7bcd0_1    conda-forge
conda-lock                1.2.1              pyhd8ed1ab_1    conda-forge
conda-package-handling    1.9.0           py310h761cc84_1    conda-forge
crashtest                 0.3.1              pyhd8ed1ab_0    conda-forge
cryptography              38.0.2          py310h674ca28_2    conda-forge
dbus                      1.13.6               h12b9eeb_3    conda-forge
distlib                   0.3.5              pyhd8ed1ab_0    conda-forge
ensureconda               1.4.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                ha18d298_0    conda-forge
filelock                  3.8.0              pyhd8ed1ab_0    conda-forge
gettext                   0.21.1               ha18d298_0    conda-forge
html5lib                  1.1                pyh9f0ad1d_0    conda-forge
idna                      3.4                pyhd8ed1ab_0    conda-forge
importlib-metadata        5.0.0              pyha770c72_1    conda-forge
jaraco.classes            3.2.3              pyhd8ed1ab_0    conda-forge
jeepney                   0.8.0              pyhd8ed1ab_0    conda-forge
jinja2                    3.1.2              pyhd8ed1ab_1    conda-forge
keyring                   23.9.3          py310h4c7bcd0_1    conda-forge
ld_impl_linux-aarch64     2.39                 ha75b1e8_0    conda-forge
libffi                    3.4.2                h3557bc0_5    conda-forge
libgcc-ng                 12.2.0              h607ecd0_19    conda-forge
libglib                   2.74.1               h58953e2_0    conda-forge
libgomp                   12.2.0              h607ecd0_19    conda-forge
libiconv                  1.17                 h9cdd2b7_0    conda-forge
libnsl                    2.0.0                hf897c2e_0    conda-forge
libsqlite                 3.39.4               hf9034f9_0    conda-forge
libstdcxx-ng              12.2.0              hc13a102_19    conda-forge
libuuid                   2.32.1            hf897c2e_1000    conda-forge
libzlib                   1.2.13               h4e544f5_4    conda-forge
[... 107 lines cut ...]
	--disable-gas \
	--disable-binutils \
	--disable-ld \
	--disable-gold \
	--disable-gprof
/bin/sh: 1: /home/parallels/Git/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/../riscv-gdb/configure: not found
gmake: *** [Makefile:388: stamps/build-gdb-newlib] Error 127
gmake: *** Waiting for unfinished jobs....
cd build-binutils-newlib && CC_FOR_TARGET=riscv64-unknown-elf-gcc /home/parallels/Git/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/../riscv-binutils/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/home/parallels/Git/chipyard/esp-tools-install \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline
/bin/sh: 1: /home/parallels/Git/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/../riscv-binutils/configure: not found
gmake: *** [Makefile:369: stamps/build-binutils-newlib] Error 127

### Expected Behavior

Should build successful.

jerryz123, 2023-02-17:
We'll stop maintaining esp-tools soon. Esp-tools is not longer required for the latest versions of gemmini

closed 2023-02-17 by jerryz123

willliu, 2023-02-17:
👍