ucb-bar/chipyard #1270: Boom+Gemmini java.lang.IllegalArgumentException: requirement failed

YingkunZhou, 2022-11-18:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.8.1
Hash: 004297b6a8c01be1b2110c4cf4f9393ae1ff8805

### OS Setup

```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.4 LTS
Release:        20.04
Codename:       focal

CONDA_EXE=/home/xxx/miniforge3/bin/conda
_CE_M=
_CE_CONDA=
CONDA_PYTHON_EXE=/home/xxx/miniforge3/bin/python
CONDA_SHLVL=2
CONDA_PREFIX=/home/xxx/work/chipyard/.conda-env
CONDA_DEFAULT_ENV=/home/xxx/work/chipyard/.conda-env
CONDA_PROMPT_MODIFIER=(/home/xxx/work/chipyard/.conda-env) 
CONDA_PREFIX_1=/home/xxx/miniforge3
RISCV=/home/xxx/work/chipyard/.conda-env/riscv-tools
CONDA_BACKUP_PATH=/home/xxx/work/chipyard/software/firemarshal:/home/xxx/work/chipyard/bin:/home/xxx/work/chipyard/.conda-env/bin:/home/xxx/miniforge3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LD_LIBRARY_PATH=/home/xxx/work/chipyard/.conda-env/riscv-tools/lib
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
GSETTINGS_SCHEMA_DIR=/home/xxx/work/chipyard/.conda-env/share/glib-2.0/schemas
JAVA_HOME_CONDA_BACKUP=
JAVA_HOME=/home/xxx/work/chipyard/.conda-env/lib/jvm
JAVA_LD_LIBRARY_PATH_BACKUP=
JAVA_LD_LIBRARY_PATH=/home/xxx/work/chipyard/.conda-env/lib/jvm/lib/server

$ conda list
# packages in environment at /home/xxx/work/chipyard/.conda-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h5bd9786_13    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.7.2              h166bdaf_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argcomplete               1.12.3             pyhd8ed1ab_2    conda-forge
asn1crypto                1.5.1              pyhd8ed1ab_0    conda-forge
asttokens                 2.0.8                    pypi_0    pypi
atk-1.0                   2.36.0               h3371d22_4    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
autoconf                  2.69            pl5321hd708f79_11    conda-forge
aws-sam-translator        1.52.0             pyhd8ed1ab_0    conda-forge
aws-xray-sdk              2.10.0             pyhd8ed1ab_0    conda-forge
awscli                    1.22.21          py39hf3d152e_0    conda-forge
babel                     2.10.3             pyhd8ed1ab_0    conda-forge
bash                      5.1.16               hd863ed2_0    conda-forge
bash-completion           2.11                          0    conda-forge
bc                        1.07.1               h7f98852_0    conda-forge
bcrypt                    4.0.0                    pypi_0    pypi
binutils                  2.36.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_2    conda-forge
bison                     3.8                  h9c3ff4c_0    conda-forge
boto3                     1.20.21            pyhd8ed1ab_0    conda-forge
boto3-stubs               1.21.6             pyhd8ed1ab_0    conda-forge
botocore                  1.23.21            pyhd8ed1ab_0    conda-forge
botocore-stubs            1.24.6                   pypi_0    pypi
brotlipy                  0.7.0           py39hb9d737c_1004    conda-forge
[... 664 lines cut ...]
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error]         at logger.Logger$.makeScope(Logger.scala:135)
[error]         at firrtl.options.Stage.transform(Stage.scala:47)
[error]         at firrtl.options.Stage.execute(Stage.scala:58)
[error]         at firrtl.options.StageMain.main(Stage.scala:71)
[error]         at chipyard.Generator.main(Generator.scala)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
[error]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.base/java.lang.reflect.Method.invoke(Method.java:568)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
```

### Expected Behavior

no error and generate verilog files

### Other Information

add ` --full-stacktrace` flag to see the full stack trace

jerryz123, 2022-11-18:
Use LargeBooms or above. This is a symptom of a longstanding wart in some of the designs. We hope to fix this in the future.

closed 2022-11-18 by jerryz123

YingkunZhou, 2022-11-18:
thanks a lot