ucb-bar/chipyard #1281: vlsi flow run with lots of errors, what could i do for solving the problem?

zhpeasd, 2022-12-02:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: latest version(clone just now)
Hash: 6929ea0

### OS Setup

[root@xxx home]# conda --version
conda 22.9.0
[root@xxx home]# git clone https://github.com/ucb-bar/chipyard.git
[root@xxx home]# cd chipyard
[root@xxx chipyard]# ./build-setup.sh riscv-tools
[root@xxx chipyard]# conda env list
*                /home/chipyard/.conda-env
                /home/miniconda3

### Other Setup

[root@xxx chipyard]# source ./env.sh
[root@xxx chipyard]# ./scripts/init-vlsi.sh asap7
[root@xxx chipyard]# cd vlsi
[root@xxx vlsi]# export HAMMER_HOME=$PWD/hammer
[root@xxx vlsi]# source $HAMMER_HOME/sourceme.sh
[root@xxx vlsi]# make buildfile CONFIG=TinyRocketConfig

### Current Behavior

(/home/chipyard/.conda-env) [zhpeasd@xxx vlsi]$ make buildfile CONFIG=TinyRocketConfig

Running with RISCV=/home/chipyard/.conda-env/riscv-tools
mkdir -p /home/chipyard/vlsi/generated-src/chipyard.TestHarness.TinyRocketConfig
cd /home/chipyard && java -jar /home/chipyard/generators/rocket-chip/sbt-launch.jar ";project chipyard; runMain chipyard.Generator --target-dir /home/chipyard/vlsi/generated-src/chipyard.TestHarness.TinyRocketConfig --name chipyard.TestHarness.TinyRocketConfig --top-module chipyard.TestHarness --legacy-configs chipyard:TinyRocketConfig "
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/chipyard/.java_tmp
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by sbt.TrapExit$ (file:/home/.sbt/boot/scala-2.12.14/org.scala-sbt/sbt/1.5.5/run_2.12-1.5.5.jar)
WARNING: Please consider reporting this to the maintainers of sbt.TrapExit$
WARNING: System::setSecurityManager will be removed in a future release
[info] welcome to sbt 1.5.5 (N/A Java 17.0.3-internal)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project constellation from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (36668 settings) ...
[info] set current project to chipyardRoot (in build file:/home/chipyard/)
[info] set current project to chipyard (in build file:/home/chipyard/)
[info] compiling 25 Scala sources to /home/chipyard/generators/testchipip/target/scala-2.12/classes ...
[error] /home/chipyard/generators/testchipip/src/main/scala/SPIFlash.scala:9:30: object spi is not a member of package sifive.blocks.devices
[error] import sifive.blocks.devices.spi.{PeripherySPIFlashKey}
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/SPIFlash.scala:78:15: not found: value PeripherySPIFlashKey
[error] spi.zip(p(PeripherySPIFlashKey)).zipWithIndex.foreach { case ((port, params), i) =>
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/SPIFlash.scala:79:56: value fSize is not a member of Any
[error] val spi_mem = Module(new SimSPIFlashModel(params.fSize, i, rdOnly))
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/SPIFlash.scala:79:63: type mismatch;
[error] found : Any
[error] required: Int
[error] val spi_mem = Module(new SimSPIFlashModel(params.fSize, i, rdOnly))
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/SPIFlash.scala:82:22: value csWidth is not a member of Any
[error] require(params.csWidth == 1, "I don't know what to do with your extra CS bits. Fix me please.")
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/UARTAdapter.scala:29:28: not found: type UARTPortIO
[error] val uart = Flipped(new UARTPortIO(UARTParams(address = 0))) // We do not support the four wire variant
[error] ^
[error] /home/chipyard/generators/testchipip/src/main/scala/UARTAdapter.scala:29:39: not found: value UARTParams
[error] val uart = Flipped(new UARTPortIO(UARTParams(address = 0))) // We do not support the four wire variant

### Expected Behavior

expected the flow runs with no errors come up,  but so many errors have arisen !  help...

zhpeasd, 2022-12-05:
can anyone give some suggestions for the error ?

jerryz123, 2022-12-05:
Most likely one of the submodules here was not initialized properly.

zhpeasd, 2022-12-05:
> Most likely one of the submodules here was not initialized properly.

i have run "git submodule update --init --recursive" in root path of repository, and after tens times, now no notice appears after running this command.

harrisonliew, 2022-12-06:
As I mentioned in the other issue, you have a problem with your sifive-blocks repo, it seems. Please make sure that all the submodule commits match: https://github.com/ucb-bar/chipyard/tree/main/generators

zhpeasd, 2022-12-06:
i found that when i run "git submodule update --init --recursive" in root path of the repository,  it will clone folders as following: sims/firesim/target-design/chipyard/sims/firesim/target-design/chipyard/sims/firesim/target-design/chipyard...... almost fall into an infinite loop ?
why ???