ucb-bar/chipyard #1283: VCS simulations ignore non-zero exit codes

tymcauley, 2022-12-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: A few commits past 1.8.1
Hash: 1b7457d2fc894e8a57b1710cb3362e4923a447cd

### OS Setup

Tested on:
Ubuntu 22.04
CentOS 7

### Other Setup

Cloned Chipyard, initialized submodules (`./scripts/init-submodules-no-riscv-tools.sh`)

Tool versions:
- Verilator 4.100
- VCS T-2022.06-SP2

### Current Behavior

When you run a software simulation of a RISC-V program, Verilator accurately reports failures, but VCS does not. Consider this test C program:

```c
int main() {
    return -1;
}
```

When running with Verilator (`make -C sims/verilator run-binary CONFIG=RocketConfig BINARY=/path/to/chipyard/tests/hello.riscv`), the stderr log file (`sims/verilator/output/chipyard.TestHarness.RocketConfig/hello.out`) ends with these two lines:

```
*** FAILED *** (tohost = -1)
*** FAILED *** (code = -1, seed 1670529991) after 10634 cycles
```

Furthermore, the process exits with status `2`, so a shell process can tell the test failed.

When running with VCS (`make -C sims/vcs run-binary CONFIG=RocketConfig BINARY=/path/to/chipyard/tests/hello.riscv`), the stderr log file (`sims/vcs/output/chipyard.TestHarness.RocketConfig/hello.out`) ends with these two lines:

```
*** FAILED *** (tohost = -1)
*** PASSED *** Completed after                11313 simulation cycles
```

Furthermore, the process exits with status `0`, so a shell process _can't_ tell the test failed.

### Expected Behavior

The VCS simulation should behave the same as the Verilator simulation: it should report the failure cause, and exit with a non-zero status code.

### Other Information

Some background on why this happens: The Verilator and VCS simulations are driven by two different testbench files (although perhaps that will change after #1263 is resolved). The Verilator simulation is driven by [`emulator.cc`](https://github.com/ucb-bar/chipyard/blob/43b75640cc5f5c95ede218e53f39976a96623c0f/generators/chipyard/src/main/resources/csrc/emulator.cc) in the Chipyard repo, while the VCS simulation is driven by [`TestDriver.v`](https://github.com/chipsalliance/rocket-chip/blob/21722ddd51957f43b933f910f34b9c645fa54226/src/main/resources/vsrc/TestDriver.v) in the rocket-chip repo.

As I recall, returning a non-zero exit code from a C program causes the RISC-V core to write the exit code to a special `tohost` memory-mapped address, which causes the [FESVR](https://github.com/riscv-software-src/riscv-isa-sim) to interpret that data as a system call, and [print this error message](https://github.com/riscv-software-src/riscv-isa-sim/blob/9b074198357b77189df989f0e274b23353f7f528/fesvr/syscall.cc#L204) (which we see as the second-to-last line in both simulation logs).

### Verilator

[This code at the end of `emulator.cc`](https://github.com/ucb-bar/chipyard/blob/43b75640cc5f5c95ede218e53f39976a96623c0f/generators/chipyard/src/main/resources/csrc/emulator.cc#L363-L386) checks for the various causes of why the test might halt, and reports the exit code. Note that this file is directly interrogating `tsi->exit_code()`, and using that to set the exit status.

### VCS

The signals of interest in `TestDriver.v` are [`failure` and `success`](https://github.com/chipsalliance/rocket-chip/blob/21722ddd51957f43b933f910f34b9c645fa54226/src/main/resources/vsrc/TestDriver.v#L116-L117). Note that both of them are 1-bit signals. The `failure` signal is [only written after a simulation timeout](https://github.com/chipsalliance/rocket-chip/blob/21722ddd51957f43b933f910f34b9c645fa54226/src/main/resources/vsrc/TestDriver.v#L137-L141), so that's the only way to have the VCS simulation report a non-zero exit code.

The `success` signal also causes the simulation to end, but [the result is the test-passed message and a `0` exit code](https://github.com/chipsalliance/rocket-chip/blob/21722ddd51957f43b933f910f34b9c645fa54226/src/main/resources/vsrc/TestDriver.v#L150-L160). The `success` signal is driven by the `TestHarness` module, and has a variety of drivers depending on how the DUT is set up. However, those drivers can only ever tell the `TestHarness` to pass the test, never to fail it (or what the failure cause was).

For this particular test, the test is finished when the [`SimSerial`](https://github.com/ucb-bar/testchipip/blob/f99b1eb59a34d7934059904f03237d5bc8d4a680/src/main/resources/testchipip/vsrc/SimSerial.v) module in the testchipip repo reports an exit code from [the TSI block](https://github.com/ucb-bar/testchipip/blob/f99b1eb59a34d7934059904f03237d5bc8d4a680/src/main/resources/testchipip/csrc/SimSerial.cc). However, `SimSerial` [only gets the LSB of the TSI's exit code](https://github.com/ucb-bar/testchipip/blob/f99b1eb59a34d7934059904f03237d5bc8d4a680/src/main/resources/testchipip/vsrc/SimSerial.v#L65), so it doesn't know if TSI wants the test to pass or fail.

### Path Forward

I don't know exactly how we want to resolve this lack-of-information getting to `TestDriver.v`, but here is a possibility.

- Change the interface between `SimSerial` and the `TestHarness` infrastructure, so we don't just have a `success` signal to drive, but perhaps a `finish` bit and `exit_code` value. Then `TestDriver` could examine the two together, and determine if the test should pass or fail.
- This would require changing `SimSerial` to read the entirety of `__exit` from TSI. Additionally, we'd have to update some of the `TestHarness`/`HarnessBinders` code to use these new `finish`/`exit_code` signals. Finally, we'd have to update `TestDriver` to interpret these signals appropriately and print out the right message (similar to the simulation-finish logic at the end of `emulator.cc`). This might be the most painful part, since this would involve a rocket-chip bump, and those are significantly more involved.

I'm sure there are other ways to solve this problem, and potentially some use-cases that I'm not considering, so I'd love to hear what other suggestions you have. I'm happy to drive forward progress on resolving this. Thanks!

jerryz123, 2022-12-09:
Yeah, this has been bothering me for a while too. Thanks for looking into it.

Your proposal for a path forwards seems reasonable. This issue should be reproducible in bare rocket-chip. I would try to reproduce it, and fix the issue there. Once we merge the fix into rocket-chip, we'll get it into chipyard with the next rocket-chip bump.

tymcauley, 2022-12-12:
After digging into this in the Rocket Chip repo, it looks like they solve this problem through the use of their `SimDTM` and `SimJTAG` classes, which both explicitly print an error message and call `stop(1)` when an error condition is detected: https://github.com/chipsalliance/rocket-chip/blob/21722ddd51957f43b933f910f34b9c645fa54226/src/main/scala/devices/debug/Periphery.scala#L179-L242

However, their VCS simulation still doesn't return a non-zero exit code, but that's just because they're not passing the `-exitstatus` flag to the simulation executable.

To use this same strategy, we could update `connectSimSerial` in `SerialAdapter.scala` to contain some similar logic. That'd certainly make this a quicker fix that waiting for the next rocket-chip update. How does that sound?

tymcauley, 2022-12-12:
Alright, I've re-run this same test using the following branch: https://github.com/ucb-bar/testchipip/pull/151

When running a simulation like this, the test appropriately fails and signal a non-zero exit code:

```
make -C sims/vcs run-binary CONFIG=RocketConfig BINARY=/path/to/chipyard/tests/hello.riscv EXTRA_SIM_FLAGS="-exitstatus"
```

However, due to (I believe) https://github.com/chipsalliance/chisel3/issues/2408, we don't get the error message printed before encountering the `$fatal` statement. That's not an issue with this code, and should be resolved upon updates to Chisel/the FIRRTL compiler.

jerryz123, 2022-12-12:
Thanks for looking at this. What is consuming the `-exitstatus` flag?

tymcauley, 2022-12-14:
If you'd like, we could add `-exitstatus` to the default invocation of the VCS simulators. Just curious, should it go outside of the `+permissive` bounds? I guess it doesn't really matter, since `+permissive` is only used to parse command-line args to the Verilator simulator, right?

jerryz123, 2022-12-15:
I think having it not be default is better. Differentiating between target-errors and simulator errors is useful.

tymcauley, 2022-12-15:
Sounds good. I can put together a PR for chipyard to bump `testchipip` once https://github.com/ucb-bar/testchipip/pull/151 is merged, and then I think we can close this issue.

tymcauley, 2022-12-15:
Awesome, thanks for merging that! #1288 is open now to merge this into Chipyard.

tymcauley, 2022-12-15:
Resolved by #1288.

closed 2022-12-15 by tymcauley