ucb-bar/chipyard #1311: Where is the *.top.v Verilog file in the latest version?

fangwenji, 2023-01-17:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hi! I'm new to the chipyard platform, and I would like to generate Verilog files based on the scala configuration (e.g. a SmallBoom SoC). I followed the instructions in the slides _Configuring and Building Custom RISC-V SoCs in Chipyard_, but I could not find the ***.top.v file** shown in the slides for the SoC after the  command _make verilog CONFIG=SmallBoomConfig_. Instead, there only exists the ***.top.f file**, which is the filelist of all the SystemVerilog design files, rather than a single .v file containing all the submodules. I would like to ask if this is a new feature? Or it's caused by my wrong operation. Thank you so much for your help!

### Motivating Example

An example of run the command _make verilog CONFIG=SmallBoomConfig_, all the verilog files I got are listed in the figure below:
[image]

Seems that there is no *.top.v file in the directory.

RaphaelKlink, 2023-01-17:
in which folder did you execute the command?

fangwenji, 2023-01-17:
I executed the command in this folder ~/chipyard/sims/verilator/

RaphaelKlink, 2023-01-17:
this should work, i use it in the same way and it worked for me. But did you wait until chipyard is finished. For BOOM cores this can take 30 minutes. Could you post your full terminal output for the command execution

fangwenji, 2023-01-17:
Here is the ouput for the command, the only error is the JAVA_TOOL_OPTIONS, which should be ignored according to the slides, and it just took less than 5 minutes for me to generate the verilog.
[image]
Besides, I compared the current common.mk (Latest commit [045f3d7](https://github.com/ucb-bar/chipyard/commit/045f3d7bc04c6ae8d112b87ad2ec6c63f7a26c39) 2 days ago) with a old-version one (commit [781cea2](https://github.com/ucb-bar/chipyard/commit/781cea27494dd1ed75196ef88e5752b4b5a770d7)), and I found that there is no output file like _TOP_FILE_ in the old version, instead, there is the _ALL_MODS_FILELIST_ in it.

jerryz123, 2023-01-17:
The latest commits on main have switched to CIRCT as the verilog-generator backend (#1239 ). Instead of a single `.top.v`, there should be a directory `gen-collateral` under `generated-src/<config>`. This should be updated in the docs.

jerryz123, 2023-01-18:
We've updated the docs for `main` in #1312 to reflect the new file emission pattern.

closed 2023-01-18 by jerryz123