ucb-bar/chipyard #1315: Obtain the synthesizable Rocket core Verilog instead of the whole SoC Verilog

fangwenji, 2023-01-31:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hi! I generated Verilog for the Rocket SoC according to the tutorial below:
[image]

However, when I tried to synthesize the SoC through Yosys, I found that all the registers have been initialized in the SoC (e.g, in the top.v file), and there is no sequential cell in the synthesized netlist Verilog.
[image]

I would like to ask if there is a way to close the initialization in the SoC or could I simply get the Rocket core Verilog without the SoC outside? Thank you so much for your time!

### Motivating Example

Same as above.

jerryz123, 2023-01-31:
For not initializing registers,wouldn't not setting the RANDOMIZE_REG_INIT define be sufficient?

For generating a different top level, if you are on a recent release, you can pass TOP=RocketTile to the makefile.  

The next release will split out the verilog files by module, making this even easier.

fangwenji, 2023-01-31:
Got it! Thank you so much for your reply. In fact, I have also tried the latest committed verision, which splits the modules and generates a filelist. Howvever, the module files are all in System Verilog format (.sv). So is there a way to generate Verilog files rather than the SV ones?

jerryz123, 2023-01-31:
The subset of system verilog in the generated RTL should not present problems for modern commercial or open-source tools. Are you seeing an issue when passing that verilog to a tool?

fangwenji, 2023-01-31:
I tried to use Yosys to read and synthesize the sv files and found no issue so far. Thank you so much for your kind help!

closed 2023-01-31 by fangwenji