ucb-bar/chipyard #1364: Barstools still uses Scala 2.13

tymcauley, 2023-02-23:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: main
Hash: 632a7a93482d5a57607d98d2512cf08f5105f620

### OS Setup

macOS 13.2.1

### Other Setup

Check out fresh Chipyard, run ./scripts/init-submodules-no-riscv-tools.sh

### Current Behavior

When building Chipyard after the Scala 2.13 update (https://github.com/ucb-bar/chipyard/pull/1303), barstools is still compiling with Scala 2.12. If you look in `tools/barstools/target`, there's a `scala-2.12` directory. I've got a project with some custom FIRRTL transforms which the `tapeout` project must depend on, and since `barstools` runs with Scala 2.12, my project must also run with Scala 2.12. I'd like to upgrade it to Scala 2.13 :)

### Expected Behavior

Barstools uses Scala 2.13 like the other Chipyard sub-projects.

I put together this PR for barstools which should resolve the problem: https://github.com/ucb-bar/barstools/pull/127

abejgonzalez, 2023-03-01:
The barstools PR LGTM, can you update the submodule pointer in Chipyard?

tymcauley, 2023-03-02:
Closed by #1367

closed 2023-03-02 by tymcauley