ucb-bar/chipyard #1377: lnuma not found?

noahgaertner, 2023-03-03:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.8.1
Hash: https://github.com/ucb-bar/chipyard/commit/004297b6a8c01be1b2110c4cf4f9393ae1ff8805

### OS Setup

```bash
❯ uname -a
Linux ece015.ece.local.cmu.edu 3.10.0-1160.62.1.el7.x86_64 #1 SMP Wed Mar 23 09:04:02 UTC 2022 x86_64 GNU/Linux
❯ lsb_release -a
LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: RedHatEnterpriseServer
Description:    Red Hat Enterprise Linux Server release 7.9 (Maipo)
Release:        7.9
Codename:       Maipo
❯ printenv
MANPATH=/afs/ece.cmu.edu/usr/ngaertne/.nvm/versions/node/v19.6.0/share/man:/afs/club.cc.cmu.edu/contrib/rhel79/share/man::/opt/puppetlabs/puppet/share/man:/afs/ece.cmu.edu/support/synopsys/synopsys.release/T-Foundation/vcs/T-2022.06/doc/man:/syn/T-2022.03-SP2/doc/syn/man
XDG_SESSION_ID=58236
HOSTNAME=ece015.ece.local.cmu.edu
SELINUX_ROLE_REQUESTED=
TERM=screen-256color
SHELL=/afs/ece.cmu.edu/usr/ngaertne/bin/zsh
HISTSIZE=1000
SSH_CLIENT=128.2.131.23 36818 22
SELINUX_USE_CURRENT_RANGE=
QTDIR=/usr/lib64/qt-3.3
OLDPWD=/afs/ece/usr/ngaertne/725/chipyard/.conda-env/lib/gcc/x86_64-conda-linux-gnu/12.1.0
QTINC=/usr/lib64/qt-3.3/include
SSH_TTY=/dev/pts/107
LC_ALL=en_US.UTF-8
QT_GRAPHICSSYSTEM_CHECKED=1
NO_PROXY=localhost,127.0.0.1,.cmu.edu,.cmu.local
http_proxy=http://proximus.ece.cmu.edu:3128/
USER=ngaertne
LD_LIBRARY_PATH=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/lib:/afs/club.cc.cmu.edu/contrib/rhel79/lib:/afs/club.cc.cmu.edu/contrib/rhel79/lib64:
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
FTP_PROXY=http://proximus.ece.cmu.edu:3128/
ftp_proxy=http://proximus.ece.cmu.edu:3128/
MAIL=/var/spool/mail/ngaertne
PATH=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/oss-cad-suite/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/oss-cad-suite/py3bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/software/firemarshal:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin:/afs/ece.cmu.edu/usr/ngaertne/miniforge3/condabin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/genus-18.14.000/tools.lnx86/bin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/confrml-18.10.300/tools.lnx86/bin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/xcelium-22.03/tools.lnx86/bin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/innovus-19.10.000/tools.lnx86/bin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/ext-17.10.000/tools.lnx86/bin:/afs/ece.cmu.edu/support/cds/share/image/usr/cds/ssv-19.12.000/tools.lnx86/bin:/afs/ece/support/cds/share/image/usr/cds/ic-6.16.090/tools/bin:/afs/ece/support/cds/share/image/usr/cds/ic-6.16.090/tools/dfII/bin:/usr/cds/mmsim-14.10.527/tools/bin:/usr/cds/mmsim-14.10.527/tools/dfII/bin:/afs/ece/support/mgc/mgc.release/cal_2018.2_33.24/aoi_cal_2018.2_33.24/bin:/afs/ece/support/cds/share/image/usr/cds/pvs-16.15.000/bin:/afs/ece.cmu.edu/usr/ngaertne/.nvm/versions/node/v19.6.0/bin:/afs/ece/usr/ngaertne/.local/bin:/afs/ece/usr/ngaertne/bin/jdk-19.0.2+7/bin:/afs/ece.cmu.edu/usr/ngaertne/bin:/afs/ece.cmu.edu/usr/ngaertne/.local/share/zinit/polaris/bin:/afs/ece.cmu.edu/usr/ngaertne/.cargo/bin:/afs/club.cc.cmu.edu/contrib/rhel79/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/opt/puppetlabs/bin:/opt/dell/srvadmin/bin:/afs/ece.cmu.edu/usr/ngaertne/bin:/afs/ece.cmu.edu/class/ece240/bin:/afs/ece.cmu.edu/support/synopsys/synopsys.release/T-Foundation/vcs/T-2022.06/bin:/syn/T-2022.03-SP2/bin:/afs/ece/support/altera/release/16.1.2/quartus/bin:/usr/cds/assura-4.15.001-616/tools/assura/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/bin
PWD=/afs/ece/usr/ngaertne/725/chipyard/.conda-env/lib/gcc/x86_64-conda-linux-gnu/12.1.0/include
LANG=en_US.UTF-8
MODULEPATH=/usr/share/Modules/modulefiles:/etc/modulefiles
LOADEDMODULES=
KDEDIRS=/usr
SELINUX_LEVEL_REQUESTED=
HTTPS_PROXY=http://proximus.ece.cmu.edu:3128/
https_proxy=http://proximus.ece.cmu.edu:3128/
SSH_ASKPASS=/usr/libexec/openssh/gnome-ssh-askpass
HISTCONTROL=ignoredups
KRB5CCNAME=KEYRING:persistent:2671647:krb_ccache_lI97CaO
SHLVL=1
HOME=/afs/ece.cmu.edu/usr/ngaertne
no_proxy=localhost,127.0.0.1,.cmu.edu,.cmu.local
HTTP_PROXY=http://proximus.ece.cmu.edu:3128/
LOGNAME=ngaertne
QTLIB=/usr/lib64/qt-3.3/lib
CVS_RSH=ssh
XDG_DATA_DIRS=/afs/ece.cmu.edu/usr/ngaertne/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share
SSH_CONNECTION=128.2.131.23 36818 172.19.138.17 22
MODULESHOME=/usr/share/Modules
LESSOPEN=||/usr/bin/lesspipe.sh %s
XDG_RUNTIME_DIR=/run/user/2671647
[... 446 lines cut ...]

### Other Information

error:
```bash
/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lnuma
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:118: product_timestamp] Error 1
make[1]: Leaving directory '/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig'
Make exited with status 2
make: *** [Makefile:72: /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig] Error 2
```

config: 
```scala
class TapeoutDoubleMedConfig extends Config(
  new freechips.rocketchip.subsystem.WithNMedCores(2) ++             // two medium rocket-cores
  new freechips.rocketchip.subsystem.WithNBanks(2) ++                // two cache banks
  new ChaCha.withChaCha ++                                                              //add custom accelerator
  new freechips.rocketchip.subsystem.WithCoherentBusTopology ++     // coherent bus
  new testchipip.WithRingSystemBus ++                               // Ring-topology system bus
//  new freechips.rocketchip.subsystem.WithNoMemPort ++ // no external memory port
//  new testchipip.WithBackingScratchpad ++                   // add mbus backing scratchpad
  new chipyard.config.AbstractConfig)
```

noahgaertner, 2023-03-03:
numactl-libs-2.0.12-5.el7.x86_64, numactl-devel-2.0.12-5.el7.x86_64, and numad-0.5-18.20150602git.el7.x86_64 are installed on the host machine as well, so I don't think numa not being installed at all is the issue either (assuming it's looking for a global version it's expecting to be there and not a local version that just wasn't installed for some reason).

jerryz123, 2023-03-15:
The error looks like it's happening when VCS is calling GCC to link together an executable. My suspicion is that the conda sysroot iacks libnuma, while VCS expects it to be present.

You are on a newer VCS than what we are using. In the documentation for VCS, can you find any reference to libnuma? I wonder if there is a way to prevent VCS from attempting to link against it.

noahgaertner, 2023-03-15:
I’ll look into it, but what version of vcs are you using? Our servers have like 12 installed, I can just use another one if that option doesn’t exist and we have it installed.

noahgaertner, 2023-03-15:
Switching to vcs-mx_vK-2015.09 seems to have worked, but now it's throwing a different error related to -fPIE not being enabled:
```bash
c++ -w  -fPIC -std=c++11 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/include -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig -O3 -I/afs/ece.cmu.edu/support/synopsys/synopsys.release/vcs-mx_vK-2015.09/include    -c /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/uart.cc
cc1: warning: command-line option '-std=c++11' is valid for C++/ObjC++ but not for C
/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: warning: libncurses.so.5, needed by /afs/ece.cmu.edu/support/synopsys/synopsys.release/vcs-mx_vK-2015.09/linux64/lib/libuclinative.so, not found (try using -rpath or -rpath-link)
/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.1.0/../../../../x86_64-conda-linux-gnu/bin/ld: /afs/ece.cmu.edu/support/synopsys/synopsys.release/vcs-mx_vK-2015.09/linux64/lib/vcs_save_restore_new.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIE
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:98: product_timestamp] Error 1
make[1]: Leaving directory '/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig'
Make exited with status 2
make: *** [Makefile:72: /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig] Error 2
```

#1011 seems to be mentioning a similar issue, but that solution didn't work for me - our machines don't have gcc-5 or gcc-6 installed, and neither are installed by chipyard (also, chipyard installs gcc and g++, so I feel like those versions should like... work)

I also tried adding -fPIE in `sims/common-sim-flags.mk` in both the `SIM_CXXFLAGS` and `SIM_LDFLAGS` variable sets and neither of those worked either, so I'm not sure what the issue is there

Let me know if this should be a new issue - I'm not sure if it's related or not to the other gcc issue.

noahgaertner, 2023-03-15:
Also saw https://github.com/chipsalliance/rocket-chip/issues/1377, which suggests gcc-4.8 - We do have that, and it just... explodes the log - I'm not actually sure what the issue is here - it looks like the C code is just wrong?
```bash
❯ make CONFIG=TapeoutDoubleMedConfig
Running with RISCV=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools
rm -rf /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig
vcs -full64 -cpp /usr/bin/g++ -cc /usr/bin/gcc -CFLAGS " -O3 -std=c++11 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/include -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig " -LDFLAGS "-L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/lib -Wl,-rpath,/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/lib -L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs -L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2" -lfesvr -ldramsim -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -quiet -q +rad +vcs+lic+wait +vc+list -f /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/sim_files.common.f -sverilog +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -debug_pp +incdir+/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.top.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.harness.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.top.mems.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.harness.mems.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+MODEL=TestHarness +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +define+FSDB  -o /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig -Mdir=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig
234 modules and 14 UDPs read.
Generating code for _VCSgd_B3D0s
Generating code for _VCSgd_reYIK
Generating code for _VCSgd_TqLIZ
Generating code for _VCSgd_TykGD
Generating code for _VCSgd_r6AeS
Generating code for _VCSgd_ZZgEu
Generating code for _VCSgd_WB1J1
Generating code for _VCSgd_NKIiI
Generating code for _VCSgd_LSQDD
Generating code for _VCSgd_v190n
Generating code for _VCSgd_ZrfTa
Generating code for _VCSgd_IVz0R
Generating code for _VCSgd_qW2V4
Generating code for _VCSgd_gKdQ7
Generating code for _VCSgd_wk8Rj
Generating code for _VCSgd_xHIqC
Generating code for _VCSgd_Qr2mQ
Generating code for _VCSgd_A9acY
Generating code for _VCSgd_G51V8
Generating code for _VCSgd_GEmJe
Generating code for _VCSgd_SydhY
Generating code for _VCSgd_f6fpd
Generating code for _VCSgd_U2duu
Generating code for _VCSgd_G5D9s
Generating code for _VCSgd_jrwqy
Generating code for _VCSgd_uAhtB
Generating code for _VCSgd_ciB4Z
Generating code for _VCSgd_D1AUt
Generating code for _VCSgd_btvuP
Generating code for _VCSgd_CmqZi
Generating code for _VCSgd_q5N8I
Generating code for _VCSgd_M10kB
Generating code for _VCSgd_UDVfe
Generating code for _VCSgd_B5NBs
Generating code for _VCSgd_zwSxT
Generating code for _VCSgd_YrSMD
Generating code for _VCSgd_LjkNG
Generating code for _VCSgd_quKip
Generating code for _VCSgd_sEmMJ
Generating code for _VCSgd_r7xua
Generating code for _VCSgd_TKKU8
Generating code for _VCSgd_kzdF0
Generating code for _VCSgd_TQLWP
Generating code for _VCSgd_JTzV3
Generating code for _VCSgd_kQ8Gv
Generating code for _VCSgd_ShsuT
Generating code for _VCSgd_GFDEs
Generating code for _VCSgd_Y32M0
Generating code for _VCSgd_HzqsM
Generating code for _VCSgd_MJipY
Generating code for _VCSgd_rG3pi
Generating code for _VCSgd_e8FM0
Generating code for _VCSgd_jRpyI
Generating code for _VCSgd_xRQvs
Generating code for _VCSgd_qtdnH
Generating code for _VCSgd_fjWz2
Generating code for _VCSgd_Rzena
Generating code for _VCSgd_cyMQa
Generating code for _VCSgd_BbqA7
Generating code for _VCSgd_AFz9Z
Generating code for _VCSgd_uQZHk
Generating code for _VCSgd_TRn6G
Generating code for _VCSgd_x1eQk
[... 730 lines cut ...]
/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/ld: /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2/libdramsim.a(CommandQueue.o): in function `void std::vector<std::vector<std::vector<DRAMSim::BusPacket*, std::allocator<DRAMSim::BusPacket*> >, std::allocator<std::vector<DRAMSim::BusPacket*, std::allocator<DRAMSim::BusPacket*> > > >, std::allocator<std::vector<std::vector<DRAMSim::BusPacket*, std::allocator<DRAMSim::BusPacket*> >, std::allocator<std::vector<DRAMSim::BusPacket*, std:: [... cut ...]
/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/ld: /afs/ece.cmu.edu/support/synopsys/synopsys.release/vcs-mx_vK-2015.09/linux64/lib/libuclinative.so: undefined reference to `tgetnum'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:98: product_timestamp] Error 1
make[1]: Leaving directory '/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig'
Make exited with status 2
make: *** [Makefile:72: /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig] Error 2
```

jerryz123, 2023-03-15:
We are on 20.04 VCS. Does that work for you?

noahgaertner, 2023-03-22:
For anyone that comes across this in the future (especially from CMU/Using AFS - not sure if this is a config issue specific to the ece machines or not) we fixed it by using chipyard's gcc/g++/ld (by NOT setting any additional flags to the VCS variable in `chipyard/sims/vcs/Makefile` (other than `-full64`), and adding `-L/usr/lib64` to `VCS_LDFLAGS` in `chipyard/vcs.mk`

YiKangOY, 2023-06-09:
> ```shell
> testchip_tsi.o:(.data.rel.ro._ZTV14testchip_tsi_t[_ZTV14testchip_tsi_t]+0x78): undefined reference to `htif_t::load_payload(std::string const&, unsigned long*)'
> ```

Hi I have also met this problem that tells me that t`estchip_tsi.o:(.data.rel.ro._ZTV14testchip_tsi_t[_ZTV14testchip_tsi_t]+0x78): undefined reference to `htif_t::load_payload(std::string const&, unsigned long*)'`
May I know how you fix this?

YiKangOY, 2023-06-09:
> Also saw [chipsalliance/rocket-chip#1377](https://github.com/chipsalliance/rocket-chip/issues/1377), which suggests gcc-4.8 - We do have that, and it just... explodes the log - I'm not actually sure what the issue is here - it looks like the C++ code is just wrong?
> 
> ```shell
> ❯ make CONFIG=TapeoutDoubleMedConfig
> Running with RISCV=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools
> rm -rf /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig
> vcs -full64 -cpp /usr/bin/g++ -cc /usr/bin/gcc -CFLAGS " -O3 -std=c++11 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/include -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2 -I/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig " -LDFLAGS "-L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/lib -Wl,-rpath,/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/riscv-tools/lib -L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs -L/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/tools/DRAMSim2" -lfesvr -ldramsim -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -quiet -q +rad +vcs+lic+wait +vc+list -f /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/sim_files.common.f -sverilog +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -debug_pp +incdir+/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.top.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.harness.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.top.mems.v /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig.harness.mems.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+MODEL=TestHarness +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +define+FSDB  -o /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig -Mdir=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig
> 234 modules and 14 UDPs read.
> Generating code for _VCSgd_B3D0s
> Generating code for _VCSgd_reYIK
> Generating code for _VCSgd_TqLIZ
> Generating code for _VCSgd_TykGD
> Generating code for _VCSgd_r6AeS
> Generating code for _VCSgd_ZZgEu
> Generating code for _VCSgd_WB1J1
> Generating code for _VCSgd_NKIiI
> Generating code for _VCSgd_LSQDD
> Generating code for _VCSgd_v190n
> Generating code for _VCSgd_ZrfTa
> Generating code for _VCSgd_IVz0R
> Generating code for _VCSgd_qW2V4
> Generating code for _VCSgd_gKdQ7
> Generating code for _VCSgd_wk8Rj
> Generating code for _VCSgd_xHIqC
> Generating code for _VCSgd_Qr2mQ
> Generating code for _VCSgd_A9acY
> Generating code for _VCSgd_G51V8
> Generating code for _VCSgd_GEmJe
> Generating code for _VCSgd_SydhY
> Generating code for _VCSgd_f6fpd
> Generating code for _VCSgd_U2duu
> Generating code for _VCSgd_G5D9s
> Generating code for _VCSgd_jrwqy
> Generating code for _VCSgd_uAhtB
> Generating code for _VCSgd_ciB4Z
> Generating code for _VCSgd_D1AUt
> Generating code for _VCSgd_btvuP
> Generating code for _VCSgd_CmqZi
> Generating code for _VCSgd_q5N8I
> Generating code for _VCSgd_M10kB
> Generating code for _VCSgd_UDVfe
> Generating code for _VCSgd_B5NBs
> Generating code for _VCSgd_zwSxT
> Generating code for _VCSgd_YrSMD
> Generating code for _VCSgd_LjkNG
> Generating code for _VCSgd_quKip
> Generating code for _VCSgd_sEmMJ
> Generating code for _VCSgd_r7xua
> Generating code for _VCSgd_TKKU8
> Generating code for _VCSgd_kzdF0
> Generating code for _VCSgd_TQLWP
> Generating code for _VCSgd_JTzV3
> Generating code for _VCSgd_kQ8Gv
> Generating code for _VCSgd_ShsuT
> Generating code for _VCSgd_GFDEs
> Generating code for _VCSgd_Y32M0
> Generating code for _VCSgd_HzqsM
> Generating code for _VCSgd_MJipY
> Generating code for _VCSgd_rG3pi
> Generating code for _VCSgd_e8FM0
> Generating code for _VCSgd_jRpyI
> Generating code for _VCSgd_xRQvs
> Generating code for _VCSgd_qtdnH
> Generating code for _VCSgd_fjWz2
> Generating code for _VCSgd_Rzena
> Generating code for _VCSgd_cyMQa
> Generating code for _VCSgd_BbqA7
> Generating code for _VCSgd_AFz9Z
> Generating code for _VCSgd_uQZHk
> Generating code for _VCSgd_TRn6G
[... 734 lines cut ...]
> /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/.conda-env/bin/ld: CommandQueue.cpp:(.text._ZNSt6vectorIS_IS_IPN7DRAMSim9BusPacketESaIS2_EESaIS4_EESaIS6_EE17_M_realloc_insertIJRKS6_EEEvN9__gnu_cxx17__normal_iteratorIPS6_S8_EEDpOT_[_ZNSt6vectorIS_IS_IPN7DRAMSim9BusPacketESaIS2_EESaIS4_EESaIS6_EE17_M_realloc_insertIJRKS6_EEEvN9__gnu_cxx17__normal_iteratorIPS6_S8_EE [... cut ...]
> make[1]: Leaving directory '/afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/generated-src/chipyard.TestHarness.TapeoutDoubleMedConfig/chipyard.TestHarness.TapeoutDoubleMedConfig'
> Make exited with status 2
> make: *** [Makefile:72: /afs/ece.cmu.edu/usr/ngaertne/725/chipyard/sims/vcs/simv-chipyard-TapeoutDoubleMedConfig] Error 2
> ```
Hi I have also met this problem that tells me that `testchip_tsi.o:(.data.rel.ro._ZTV14testchip_tsi_t[_ZTV14testchip_tsi_t]+0x78): undefined reference to htif_t::load_payload(std::string const&, unsigned long*)`
My g++ version is 10.2.0
May I know how you fix this?

jerryz123, 2023-06-09:
Did you update chipyard without reinstalling the toolchain?

Try running `scripts/build-toolchain-extra.sh --prefix $RISCV`

YiKangOY, 2023-06-09:
I tried this but it didn't help.
I use a haky way to solve this, I put my solution here if someone meets the same problem.

In /path-to-chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-gcc/gcc/testsuite/gdc.test/runnable_cxx/extra-files/cppb.cpp.
It says that:
```
Fortunately, it is easily possible to disable the new feature
by defining _GLIBCXX_USE_CXX11_ABI as 0 before including any standard
headers.
```
So I put `#define _GLIBCXX_USE_CXX11_ABI 0` at the first line of t`estchip_tsi.cc` and the problem is sloved.
I'm not sure if I am doing it correctly.