ucb-bar/chipyard #1384: Docker build fail

LIUQyou, 2023-03-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.8.1
Hash: 004297b

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

Try to build a docker image inside dockerfiles/  using
```
sudo docker build --target base-with-tools .
```
But failed with
```
docker build --target base-with-tools .
[+] Building 2.3s (9/10)
 => [internal] load build definition from Dockerfile                                                                                        0.0s
 => => transferring dockerfile: 1.83kB                                                                                                      0.0s
 => [internal] load .dockerignore                                                                                                           0.0s
 => => transferring context: 2B                                                                                                             0.0s
 => [internal] load metadata for docker.io/library/ubuntu:20.04                                                                             0.9s
 => [base 1/5] FROM docker.io/library/ubuntu:20.04@sha256:9fa30fcef427e5e88c76bc41ad37b7cc573e1d79cecb23035e413c4be6e476ab                  0.0s
 => CACHED [base 2/5] RUN apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends     curl     wge  0.0s
 => CACHED [base 3/5] WORKDIR /root                                                                                                         0.0s
 => CACHED [base 4/5] RUN git clone https://github.com/ucb-bar/chipyard.git &&         cd chipyard &&         git checkout $CHIPYARD_HASH   0.0s
 => CACHED [base 5/5] RUN ./chipyard/.github/scripts/install-conda.sh                                                                       0.0s
 => ERROR [base-with-tools 1/2] RUN cd chipyard &&         export MAKEFLAGS=-"j $(nproc)" &&         conda activate base &&         ./setu  1.3s
------
 > [base-with-tools 1/2] RUN cd chipyard &&         export MAKEFLAGS=-"j $(nproc)" &&         conda activate base &&         ./setup.sh --env-name chipyard --skip-validate:
#0 1.228 /bin/bash: ./setup.sh: No such file or directory
------
Dockerfile:42
--------------------
  41 |     # Initialize repo
  42 | >>> RUN cd chipyard && \
  43 | >>>         export MAKEFLAGS=-"j $(nproc)" && \
  44 | >>>         conda activate base && \
  45 | >>>         ./setup.sh --env-name chipyard --skip-validate
  46 |
--------------------
ERROR: failed to solve: process "/bin/bash -cl cd chipyard &&         export MAKEFLAGS=-\"j $(nproc)\" &&         conda activate base &&         ./setup.sh --env-name chipyard --skip-validate" did not complete successfully: exit code: 127

```

./setup.sh: No such file or directory

### Expected Behavior

Docker build successfully

jerryz123, 2023-03-15:
The docker setup has not been maintained well. With the Conda based flow, most users do not need to run within a Docker environment. As long as you can install the Conda package manager on their system, you should be able to setup Chipyard using the default instructions.

Conda is a much lighter environment than Docker, which why we've transitioned to a standard Conda flow for installation. I recommend you try the conda-based flow, if you are able.

Is there a specific reason you want the docker image?

rseac, 2024-10-15:
@jerryz123 Due to restrictions on certain servers, we may only be able to deploy docker containers on them. The servers do not have Conda on them, and most likely would not be installed.

rseac, 2024-10-19:
If it is useful, I created a Docker version here. https://github.com/rseac/chipyard-docker