ucb-bar/chipyard #1388: CIRCT elaboration should uniquify top/model hierarchy

harrisonliew, 2023-03-09:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

With the migration to CIRCT, modules in the hierarchy are now individually in their own Verilog files. However, modules that are common to the top and model's hierarchy are not uniquified. After the top is synthesized, those modules will collide with the same modules in the model's hierarchy, causing simulation to fail.

For now, this commit avoids module collisions: https://github.com/ucb-bar/chipyard/pull/1369/commits/08839930009a6f15f2b3ba73940c243d34750ab6

When this "bug" is fixed, this hack can be removed.

### Motivating Example

Gate-level simulation is altered due to this "bug". If synthesized modules are referenced in the model's hierarchy and timing annotation is turned on, it is uncertain if the timing at the top/model boundary will still be OK. Regardless, a testbench should be fully behavioral.

joey0320, 2023-03-10:
Just to make sure I understood correctly, we need separate file names for top/model because when running gate-level simulation the simulator does not know if it is using the time-annotated synthesized modules or just the verilog files?

harrisonliew, 2023-03-10:
Not quite. Let me try to be clearer. Suppose we had this module hierarchy:

```
   Top    Model
  /   \   /   \
ModA   ModB   Mod C
```
There are instances of `ModA` and `ModB` in `Top`, and there are instances of `ModB` and `ModC` in Model.

In `<long_name>.top.f`:
```
Top.sv
ModA.sv
ModB.sv
```
In `<long_name>.model.f`:
```
Model.sv
ModB.sv
ModC.sv
```

So here's the problem sequence:
1. The files in `<long_name>.top.f` are given to synthesis. We get a single mapped gate-level netlist, say `Top.mapped.v`.
2. We then combine `Top.mapped.v` and the files in `<long_name>.model.f` for gate-level simulation.
3. Problem: `ModB` in `Top.mapped.v` collides with `ModB` in `<long_name>.model.f`.
4. Current hack in linked commit: remove `ModB.sv` from `<long_name>.model.f`.

Why this is not OK: the instance of `ModB` in `Model` now refers to the synthesized version of `ModB`. If the simulation is timing-annotated, the signals crossing the Top/Model boundary will have modified timing compared to the non-timing annotated simulation. Furthermore, if there are more shared modules, internally, the design under `Model` may have timing violations. In general, the testbench (`Model`) should not have any gate-level modules.

Desired solution:
In `<long_name>.top.f`:
```
Top.sv
ModA.sv
ModB_Top.sv
```
In `<long_name>.model.f`:
```
Model.sv
ModB_Model.sv
ModC.sv
```

joey0320, 2023-03-10:
By instance declarations, do you mean that the module names inside the files should also be changes?
For instance, `module Adder` to `module Adder_Top` and `module Adder_Model`? In my understanding, the tools reads the `<long_name>.<top or model>.f` and works on those files so it seems like we don't need to change the module names inside the files.

harrisonliew, 2023-03-10:
You do need to change the module names. This way, `Adder_Top` can be synthesized and simulated inside the Top, while `Adder_Model` remains behavioral and is simulated inside the Model.

joey0320, 2023-03-10:
Hmmm I see. I'll look into this

harrisonliew, 2023-04-13:
Bumping this - it needs to be fixed ASAP. We're running into gate-level synthesis unresolved modules when synthesis modifies a module (e.g. optimizes away a bit, creating a modified module name) or there are unsynthesizable modules (e.g. plusarg_reader used in TLMonitors). Thus, those modules that are also used by the TestHarness are no longer in the mapped Verilog file.

I think this is not quite correct: https://github.com/ucb-bar/chipyard/blob/10c26250c8ad71a6bb7ff5c30f365f8655338c8f/scripts/split-module-files.py#L74 but the proper fix is to deuniquify the modules in the top vs. model.

Related to https://groups.google.com/g/chipyard/c/tXCmoNNpwwg.

joey0320, 2023-04-14:
Sorry for the delay, will work on this today

joey0320, 2023-04-21:
Closing this issue for now

closed 2023-04-21 by joey0320

allpan3, 2023-04-22:
I'm still seeing multiple module declarations errors in post-syn simulation.
I might be wrong, but based on my observation it's because [instances under DUT are not uniquified](https://github.com/ucb-bar/chipyard/blob/15e22d985b29442f168ce2a91771b0e39de816b1/scripts/uniqify-module-names.py#L63).

Taking `AsyncQueueSink*` for example, here are the error messages:
```

Error-[MPD] Module previously declared
The module was previously declared at:
"/bwrcq/C/allpan/chipyard/vlsi/build/chipyard.TestHarness.HPURocketConfig-ChipTop/syn-rundir/ChipTop.mapped.v",
459025
It is redeclared later at:
"/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_1.sv",
71: token is 'AsyncQueueSink_1'
module AsyncQueueSink_1(
^
Please remove one of the declarations and compile again.

Error-[MPD] Module previously declared
The module was previously declared at:
"/bwrcq/C/allpan/chipyard/vlsi/build/chipyard.TestHarness.HPURocketConfig-ChipTop/syn-rundir/ChipTop.mapped.v",
458272
It is redeclared later at:
"/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_2.sv",
71: token is 'AsyncQueueSink_2'
module AsyncQueueSink_2(
^
Please remove one of the declarations and compile again.

Error-[MPD] Module previously declared
The module was previously declared at:
"/bwrcq/C/allpan/chipyard/vlsi/build/chipyard.TestHarness.HPURocketConfig-ChipTop/syn-rundir/ChipTop.mapped.v",
460013
It is redeclared later at:
"/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink.sv",
71: token is 'AsyncQueueSink'
module AsyncQueueSink(
^
Please remove one of the declarations and compile again.
```

In `sim-inputs.yml`, I'm seeing these lines:
```
sim.inputs:
  top_module: ChipTop
  tb_name: ''
  input_files:
    - "/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_1.sv"
    - "/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_2.sv"
    - "/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_3.sv"
    - "/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink_3_TestHarness_UNIQUIFIED.sv"
    - "/bwrcq/C/allpan/chipyard/vlsi/generated-src/chipyard.TestHarness.HPURocketConfig/gen-collateral/AsyncQueueSink.sv"
```

Only `AsyncQueueSink_3` is uniquified. 
I checked, `AsyncQueueSink` is under `ChipTop` but `AsyncQueueSink_3` is not.

joey0320, 2023-04-22:
@allpan3  Can you share your RTL & config so that I can run tests on it? I don't see those issues in the existing configs in CY (as CI seems to be passing) so it would be great if I could actually test the script on your design.

allpan3, 2023-04-22:
@joey0320 This may be a little complicated since I'm using tstech28 and currently the RTL repo is private (and I'm not the owner).

I will ask my advisor if he can make the repo public or invite you, but even if not you can just copy from my area.
And also do you have access to tstech28? I think you have to have permission to the PDK on BWRC machine in order to run through the hammer flow successfully.

joey0320, 2023-04-22:
I think I don't need the PDK. I just want to generate the verilog with your design so that I can see what is happening.

allpan3, 2023-04-22:
Ok, you can try it. I think if you are just running the steps before `sram_generator` then you probably won't need PDK.
Btw my project is written in verilog.

You can copy `generator/hpu` from `/bwrcq/C/allpan/chipyard/generators/hpu`.
You will also need to copy the modified [build.sbt](https://github.com/allpan3/chipyard/blob/hpu/build.sbt) and [RoCCAcceleratorConfigs.scala](https://github.com/allpan3/chipyard/blob/hpu/generators/chipyard/src/main/scala/config/RoCCAcceleratorConfigs.scala).

The config is `HPURocketConfig`.

allpan3, 2023-04-22:
And another issue is `make sim-rtl` now has duplicated declarations.
Both `inputs.yml` and `sim-inputs.yml` are fed to hammer, and in `sim-inputs.yml`, `sim.inputs.input_files_meta` is `append`, so all the files are specified twice in VCS.

```
./example-vlsi -e /bwrcq/C/allpan/new-chipyard/vlsi/env-bwrc.yml -p /bwrcq/C/allpan/new-chipyard/vlsi/example-tools.yml -p /bwrcq/C/allpan/new-chipyard/vlsi/tstech28.yml -p /bwrcq/C/allpan/new-chipyard/vlsi/build/chipyard.TestHarness.RocketConfig-ChipTop/inputs.yml -p /bwrcq/C/allpan/new-chipyard/vlsi/build/chipyard.TestHarness.RocketConfig-ChipTop/sram_generator-output.json -p /bwrcq/C/allpan/new-chipyard/vlsi/tstech28_sram_macros/sram_generator-output.yml -p /bwrcq/C/allpan/new-chipyard/vlsi/build/chipyard.TestHarness.RocketConfig-ChipTop/sim-inputs.yml --sim_rundir /bwrcq/C/allpan/new-chipyard/vlsi/build/chipyard.TestHarness.RocketConfig-ChipTop/sim-rtl-rundir --obj_dir /bwrcq/C/allpan/new-chipyard/vlsi/build/chipyard.TestHarness.RocketConfig-ChipTop sim
```
I ran with RocketConfig this time so it should be a common issue.

harrisonliew, 2023-04-22:
I suspect using DFS is not correct with this check: https://github.com/ucb-bar/chipyard/blob/15e22d985b29442f168ce2a91771b0e39de816b1/scripts/uniqify-module-names.py#L63

If the instance hierarchy is as follows, where ModA is the model, and ModB and ModC need to be uniquified:
```
ModA
-> ModB
    -> ModC
-> ModB
    -> ModC
```
DFS with the visited check will return:
```
ModA
-> ModB_UNIQUIFIED
    -> ModC_UNIQUIFIED
-> ModB
    -> ModC
```
This should actually be done using a BFS, so that all instances of ModB get changed to ModB_UNIQUIFIED, then, ModC.

Regarding plusarg_reader.v missing, I don't see it in the `.model.f` file, so it doesn't get checked. This points to a potential bug in `split-module-files.py`, which I think is [here](https://github.com/ucb-bar/chipyard/blob/15e22d985b29442f168ce2a91771b0e39de816b1/scripts/split-module-files.py#L74), where the model_mods set should not fully subtract the dut_mods set.

davidmlw, 2023-06-23:
Could this be done in Chisel? I remember that a Chisel Module with different parameters will be elaborated to two different Modules. For the `SharedModule` shared between Top and Harness (Model), how about passing a parameter `where` pointing to the current parent Module. Will this solve this problem?

And if it will, how about Chisel adding a scope concept, with synthesized scope, testbench scope, etc, and `SharedModule` has different names in different scopes.

jerryz123, 2023-06-23:
A scope mechanism would fix this, although it would have to block de-uniquification passes at the FIRRTL level too.
 
 Passing in a parameter explicitly, rather than implicitly through a Scope, is not feasible... our TestHarness blocks have their own non-trivial module hierarchies.

joey0320, 2023-07-03:
Yes, the current script is a workaround. The true solution would be to add passes in firtool to emit separate verilog files for top & harness.