ucb-bar/chipyard #1417: MMIO and front bus port not generated in MMIOScratchpadOnlyRocketConfig

zephray, 2023-03-23:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: not on a release
Hash: 26f86d

### OS Setup

Linux wenting-b650m 5.19.0-35-generic #36-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 3 18:36:56 UTC 2023 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.10
Release:	22.10
Codename:	kinetic

# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h5bd9786_13    conda-forge
aiobotocore               2.4.2              pyhd8ed1ab_0    conda-forge
aiohttp                   3.8.4           py310h1fa729e_0    conda-forge
aioitertools              0.11.0             pyhd8ed1ab_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alabaster                 0.7.13             pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argcomplete               3.0.2              pyhd8ed1ab_0    conda-forge
asttokens                 2.2.1                    pypi_0    pypi
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
autoconf                  2.71            pl5321h2b4cb7a_1    conda-forge
aws-sam-translator        1.55.0             pyhd8ed1ab_0    conda-forge
aws-xray-sdk              2.11.0             pyhd8ed1ab_0    conda-forge
awscli                    1.25.60         py310hff52083_0    conda-forge
azure-core                1.26.3             pyhd8ed1ab_0    conda-forge
azure-identity            1.12.0             pyhd8ed1ab_0    conda-forge
babel                     2.12.1             pyhd8ed1ab_1    conda-forge
bash                      5.1.16               hd863ed2_0    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
bc                        1.07.1               h7f98852_0    conda-forge
bcrypt                    3.2.2           py310h5764c6d_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
bison                     3.8                  h9c3ff4c_0    conda-forge
boto3                     1.24.59            pyhd8ed1ab_0    conda-forge
boto3-stubs               1.26.95            pyhd8ed1ab_0    conda-forge
botocore                  1.27.59            pyhd8ed1ab_0    conda-forge
botocore-stubs            1.29.95                  pypi_0    pypi
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
brotlipy                  0.7.0           py310h5764c6d_1005    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cachecontrol              0.12.11            pyhd8ed1ab_1    conda-forge
cachecontrol-with-filecache 0.12.11            pyhd8ed1ab_1    conda-forge
cachy                     0.3.0              pyhd8ed1ab_1    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_3    conda-forge
cfgv                      3.3.1              pyhd8ed1ab_0    conda-forge
cfn-lint                  0.75.1             pyhd8ed1ab_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
[... 450 lines cut ...]
  output [1:0]  axi4_fbus_0_bits_b_bits_resp,
  output        axi4_fbus_0_bits_ar_ready,
  input         axi4_fbus_0_bits_ar_valid,
  input  [7:0]  axi4_fbus_0_bits_ar_bits_id,
  input  [31:0] axi4_fbus_0_bits_ar_bits_addr,
  input  [7:0]  axi4_fbus_0_bits_ar_bits_len,
  input  [2:0]  axi4_fbus_0_bits_ar_bits_size,
  input  [1:0]  axi4_fbus_0_bits_ar_bits_burst,
  input         axi4_fbus_0_bits_ar_bits_lock,
  input  [3:0]  axi4_fbus_0_bits_ar_bits_cache,
  input  [2:0]  axi4_fbus_0_bits_ar_bits_prot,
  input  [3:0]  axi4_fbus_0_bits_ar_bits_qos,
  input         axi4_fbus_0_bits_r_ready,
  output        axi4_fbus_0_bits_r_valid,
  output [7:0]  axi4_fbus_0_bits_r_bits_id,
  output [63:0] axi4_fbus_0_bits_r_bits_data,
  output [1:0]  axi4_fbus_0_bits_r_bits_resp,
  output        axi4_fbus_0_bits_r_bits_last,
  input         custom_boot,
  input         clock_clock,
  input         reset,
  output        uart_0_txd,
  input         uart_0_rxd
);
```

zephray, 2023-03-23:
Hi, thanks for your quick reply! However I am already on the latest main branch (#26f86d). Maybe I am missing something, or the behavior is actually expected and the old 1.8.1 release behavior is wrong?

jerryz123, 2023-03-23:
No, it should definitely be generating ports...this is what I see on 9ef3001ce1aafc5e0d38c5b7d980825c4faac7e3 :
```
9ef3001ce1aafc5e0d38c5b7d980825c4faac7e3
```

zephray, 2023-03-23:
That's interesting, I deleted my local repo and started from scratch with 9ef3001, now I can getting the same result as you have. However, isn't this still wrong? The MMIO port is present, but the front bus port is still missing.

kgreig87, 2023-04-01:
I'm also seeing the same issue as @zephray - DigitalTop contains a MMIO port, but is missing a Front Port.

jerryz123, 2023-04-01:
Yeah, I'm looking into this.

jerryz123, 2023-04-01:
The latest commit on this branch: https://github.com/ucb-bar/chipyard/pull/1399/files#diff-e72d7730b5de7057f15f513757aa310c4a51b21e8d5d0e1cb7b206b132e802d9 should fix this.

kgreig87, 2023-04-02:
Thanks Jerry, this resolved the issue. I also like what you're adding with the custom IO cells

zephray, 2023-04-02:
Awesome, thanks jerry, just tried and it works great! Should I close the issue or wait for the PR to be merged first?

closed 2023-04-20 by abejgonzalez