ucb-bar/chipyard #1423: chipyard.config.WithTileFrequency not working in 1.9.0

noahgaertner, 2023-04-03:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.0
Hash: 7475bfb

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`
uname - a
```bash
Linux ece015.ece.local.cmu.edu 3.10.0-1160.83.1.el7.x86_64 #1 SMP Mon Dec 19 10:44:06 UTC 2022 x86_64 GNU/Linux
```
lsb_release -a
```bash
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	RedHatEnterpriseServer
Description:	Red Hat Enterprise Linux Server release 7.9 (Maipo)
Release:	7.9
Codename:	Maipo
```
printenv (w/o pdk proprientary info)
```bash
MANPATH=/afs/ece.cmu.edu/usr/ngaertne/.nvm/versions/node/v19.6.0/share/man:/afs/club.cc.cmu.edu/contrib/rhel79/share/man::/opt/puppetlabs/puppet/share/man:/afs/ece.cmu.edu/support/synopsys/synopsys.release/T-Foundation/vcs/T-2022.06//doc/man:/syn/U-2022.12-SP2/doc/syn/man
XDG_SESSION_ID=5779
HOSTNAME=ece015.ece.local.cmu.edu
SELINUX_ROLE_REQUESTED=
TERM=xterm-256color
SHELL=/afs/ece.cmu.edu/usr/ngaertne/bin/zsh
HISTSIZE=1000
SSH_CLIENT=128.2.131.23 57514 22
SELINUX_USE_CURRENT_RANGE=
QTDIR=/usr/lib64/qt-3.3
OLDPWD=/afs/ece.cmu.edu/usr/ngaertne
QTINC=/usr/lib64/qt-3.3/include
SSH_TTY=/dev/pts/191
LC_ALL=en_US.UTF-8
QT_GRAPHICSSYSTEM_CHECKED=1
NO_PROXY=localhost,127.0.0.1,.cmu.edu,.cmu.local
http_proxy=http://proximus.ece.cmu.edu:3128/
USER=ngaertne
LD_LIBRARY_PATH=/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/.conda-env/riscv-tools/lib:/afs/club.cc.cmu.edu/contrib/rhel79/lib:/afs/club.cc.cmu.edu/contrib/rhel79/lib64:
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
FTP_PROXY=http://proximus.ece.cmu.edu:3128/
ftp_proxy=http://proximus.ece.cmu.edu:3128/
MAIL=/var/spool/mail/ngaertne
PATH=/afs/ece.cmu.edu/support/cds/share/image/usr/cds/xcelium-22.03/tools.lnx86/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/oss-cad-suite/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/oss-cad-suite/py3bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/.conda-env/riscv-tools/bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/software/firemarshal:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/software/firemarshal:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/.conda-env/riscv-tools/bin:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/software/firemarshal:/afs/ece.cmu.edu/usr/ngaertne/725/chipyard-1.9/.conda-env/bin:/afs/ece.cmu.edu/usr/ngaertne/mambaforge/condabin:/afs/ece.cmu.edu/usr/ngaertne/.nvm/versions/node/v19.6.0/bin:/afs/ece/usr/ngaertne/.local/bin:/afs/ece/usr/ngaertne/bin/jdk-19.0.2+7/bin:/afs/ece.cmu.edu/usr/ngaertne/bin:/afs/ece.cmu.edu/usr/ngaertne/.local/share/zinit/polaris/bin:/afs/ece.cmu.edu/usr/ngaertne/.cargo/bin:/afs/club.cc.cmu.edu/contrib/rhel79/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/opt/puppetlabs/bin:/opt/dell/srvadmin/bin:/afs/club.cc.cmu.edu/contrib/misc/yosys-rhel79/bin:/afs/ece.cmu.edu/usr/ngaertne/bin:/afs/ece.cmu.edu/class/ece240/bin:/afs/ece.cmu.edu/support/synopsys/synopsys.release/T-Foundation/vcs/T-2022.06/bin:/syn/U-2022.12-SP2/bin:/afs/ece/support/altera/release/pro-19.3.0.222/quartus/bin:/afs/club.cc.cmu.edu/projects/stuco-open-eda/bin
PWD=/afs/ece.cmu.edu/usr/ngaertne
LANG=en_US.UTF-8
MODULEPATH=/usr/share/Modules/modulefiles:/etc/modulefiles
LOADEDMODULES=
KDEDIRS=/usr
SELINUX_LEVEL_REQUESTED=
HTTPS_PROXY=http://proximus.ece.cmu.edu:3128/
https_proxy=http://proximus.ece.cmu.edu:3128/
SSH_ASKPASS=/usr/libexec/openssh/gnome-ssh-askpass
HISTCONTROL=ignoredups
KRB5CCNAME=KEYRING:persistent:2671647:krb_ccache_1h0cp43
SHLVL=1
HOME=/afs/ece.cmu.edu/usr/ngaertne
no_proxy=localhost,127.0.0.1,.cmu.edu,.cmu.local
HTTP_PROXY=http://proximus.ece.cmu.edu:3128/
LOGNAME=ngaertne
QTLIB=/usr/lib64/qt-3.3/lib
CVS_RSH=ssh
[... 499 lines cut ...]
  new chipyard.harness.WithSimAXIMemOverSerialTL() ++ // add SimDRAM DRAM model for axi4 backing memory over the SerDes link, if axi4 mem is enabled
  new chipyard.config.WithSerialTLBackingMemory ++ // remove axi4 mem port in favor of SerialTL memory
    // default config
  new chipyard.config.AbstractConfig)
  ``` results in
  ```bash
  buildTopClockGenerator Frequency Summary
  Input Reference Frequency: 375.0 MHz
  Output clock subsystem_sbus_0, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  Output clock subsystem_sbus_1, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  Output clock subsystem_pbus_0, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  Output clock subsystem_fbus_0, requested: 75.0 MHz, actual: 75.0 MHz (division of 5)
  Output clock subsystem_mbus_0, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  Output clock subsystem_cbus_0, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  Output clock implicit_clock, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)

harnessDividerOnlyClockGenerator Frequency Summary
  Input Reference Frequency: 375.0 MHz
  Output clock mem_over_serial_tl_clock, requested: 75.0 MHz, actual: 75.0 MHz (division of 5)
  Output clock buildtop_reference_clock, requested: 375.0 MHz, actual: 375.0 MHz (division of 1)
  ```

### Expected Behavior

Tile Frequency should be 750MHz, and Input Reference Clock should also be 750MHz.

noahgaertner, 2023-04-04:
It doesn't appear to be a config bug, same behavior exists in `MulticlockRocketConfig` - also have confirmed it does not exist in 1.8.1, so we're rolling back to that release for our chip, but I can help debug/isolate the issue if neccesary.

jerryz123, 2023-04-06:
Reorder the `WithAsynchonousRocketTiles` above the `WithNMedCores` in your config

jerryz123, 2023-04-06:
For context, the `WithXRocketTiles` now modifies a local parameter for the tile, rather than global parameters, so those fragments must be added after tiles are insantiated

jerryz123, 2023-04-06:
FYI those reported frequencies are only used by the fake "clock approximator" divideronlyclockgenerator used for RTL sim/FireSim. They otherwise have no bearing on the physical implementation of the chip, in which you'll need to integrate a PLL or a clock receiver. 

#1427 resolves this PR, and also provides a demo `ChipLikeRocketConfig` to show how a tapeout-ready config might appear.

noahgaertner, 2023-04-10:
issue fixed by reordering config.

closed 2023-04-10 by noahgaertner