ucb-bar/chipyard #1429: SBT Download Error

dipsankarb, 2023-04-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.0
Hash: g7475bfb1

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

Linux user 5.15.0-46-generic #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 20.04

Java version 17

### Other Setup

Following steps as per conda flow mentioned here:

https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

### Current Behavior

Running with RISCV=/home/arch_c1/chipyard/.conda-env/riscv-tools
cd /home/arch_c1/chipyard && java -jar /home/arch_c1/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.ivy.home=/home/arch_c1/chipyard/.ivy2 -Dsbt.global.base=/home/arch_c1/chipyard/.sbt -Dsbt.boot.directory=/home/arch_c1/chipyard/.sbt/boot/ -Dsbt.color=always  ";project chipyard; compile"
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Dsbt.supershell=false -Djava.io.tmpdir=/home/arch_c1/chipyard/.java_tmp
[info] welcome to sbt 1.8.2 (N/A Java 17.0.3-internal)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/arch_c1/chipyard/project
[warn] 
[warn] 	Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn] 		com.eed3si9n:sbt-assembly:0.15.0 (sbtVersion=1.0, scalaVersion=2.12)
[warn] 
[warn] 	Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading com.eed3si9n:sbt-assembly;sbtVersion=1.0;scalaVersion=2.12:0.15.0
[error]   Not found
[error]   Not found
[error]   not found: https://repo1.maven.org/maven2/com/eed3si9n/sbt-assembly_2.12_1.0/0.15.0/sbt-assembly-0.15.0.pom
[error]   not found: /home/arch_c1/chipyard/.ivy2/localcom.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml
[error]   checksum format error: /home/arch_c1/.cache/coursier/v1/https/repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml.sha1
[error]   checksum format error: /home/arch_c1/.cache/coursier/v1/https/repo.typesafe.com/typesafe/ivy-releases/com.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml.sha1
[error] 	at lmcoursier.CoursierDependencyResolution.unresolvedWarningOrThrow(CoursierDependencyResolution.scala:344)
[error] 	at lmcoursier.CoursierDependencyResolution.$anonfun$update$38(CoursierDependencyResolution.scala:313)
[error] 	at scala.util.Either$LeftProjection.map(Either.scala:573)
[error] 	at lmcoursier.CoursierDependencyResolution.update(CoursierDependencyResolution.scala:313)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:60)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:59)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:133)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:73)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$20(LibraryManagement.scala:146)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:228)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:146)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:127)
[error] 	at sbt.util.Tracked$.$anonfun$inputChangedW$1(Tracked.scala:219)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:160)
[error] 	at sbt.Classpaths$.$anonfun$updateTask0$1(Defaults.scala:3687)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.Execute.work(Execute.scala:291)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
[error] 	at java.base/java.lang.Thread.run(Thread.java:833)
[error] (update) sbt.librarymanagement.ResolveException: Error downloading com.eed3si9n:sbt-assembly;sbtVersion=1.0;scalaVersion=2.12:0.15.0
[error]   Not found
[error]   Not found
[error]   not found: https://repo1.maven.org/maven2/com/eed3si9n/sbt-assembly_2.12_1.0/0.15.0/sbt-assembly-0.15.0.pom
[error]   not found: /home/arch_c1/chipyard/.ivy2/localcom.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml
[error]   checksum format error: /home/arch_c1/.cache/coursier/v1/https/repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml.sha1
[error]   checksum format error: /home/arch_c1/.cache/coursier/v1/https/repo.typesafe.com/typesafe/ivy-releases/com.eed3si9n/sbt-assembly/scala_2.12/sbt_1.0/0.15.0/ivys/ivy.xml.sha1
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
make: *** [/home/arch_c1/chipyard/common.mk:384: launch-sbt] Error 1

### Expected Behavior

Succesful make

dipsankarb, 2023-04-08:
This page is 404

https://repo1.maven.org/maven2/com/eed3si9n/sbt-assembly_2.12_1.0/0.15.0/sbt-assembly-0.15.0.pom

Is there a way to update the URL?

dipsankarb, 2023-04-09:
The issue has been resolved by updating the plugins.sbt file in the projects directory.

closed 2023-04-09 by dipsankarb

Blameying, 2023-04-10:
I met the same problem recently, and the version for sbt-assembly defined in *plugins.sbt* does not exist in the maven repos. 

I wonder if this is a common issue.