ucb-bar/chipyard #1443: `make sim-rtl` fails due to redeclaration of modules

allpan3, 2023-04-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.0
Hash: 7475bfb1a05802ac4dfc3990a889f93164b8d798

### OS Setup

BWRC machine

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

When running `make sim-rtl` from `vlsi` directory, re-declaration of modules causes VCS to error out.

[image]

### Expected Behavior

In version 1.8.1, this does not happen. Re-declaration is a warning and overrides the previous definition instead.

[image]

### Other Information

Not really a bug, but I'm trying to figure out how to resolve this.
Based on some experiments it seems like this is caused by the omission of `-top TestDriver` in VCS command options. In the previous version of Chipyard `-top TestDriver` is specified. Not sure why this changes an error to a warning.

Commit 83764d3 modified `sim.mk` to [dump out `sim-inputs.yml` without `tb_name`](https://github.com/ucb-bar/chipyard/blob/10c26250c8ad71a6bb7ff5c30f365f8655338c8f/vlsi/sim.mk#L11), which removes the `-top <>` option from VCS invocation when running `make sim-rtl`.
@harrisonliew What is the reason for this? And is there a way to specify `-top`? I tried setting `sim.inputs.tb_name` in my yaml but it looks like `sim-inputs.yml` takes precedence.

I know I can try to avoid having duplicated definitions but it's often involving PDKs so takes more work to resolve.

allpan3, 2023-04-19:
Fix for this: add `-error=noMPD` option in vcs to suppress the error.
Not sure if this is a good long term solution.

harrisonliew, 2023-04-19:
[`-top` was removed](https://github.com/ucb-bar/hammer-synopsys-plugins/pull/34) to remove a different VCS error, one which essentially said that two different designs were declared as the top.

You can add back `sim.inputs.tb_name` to the `sim.mk` so that it appears in `sim-inputs.yml`. If you specify `-error=noMPD`, I presume it just ignores redeclarations and assumes the last one is valid - this would work but could hide some problems later, so it's not good long-term.

Anyways, I think I see the root issue - it's that the SS and FF corners have different Verilog files (I presume this is the output of the TSMC memory compiler). I did a comparison of the two and they're only different by some timing params, which is expected. For RTL sim, these params don't get used; for gate-level timing-annotated sim, you would only want to use the file corresponding to the corner you want to check against. TL;DR, only 1 memory Verilog file should be used for simulation - perhaps you can modify the plugin's `sram_compiler` tool for now to only include the TT corner Verilog?

allpan3, 2023-04-19:
Yea that makes sense. I will make the change in `sram_compiler`. 
The same issue also exists for the standard cell library. The Decap cells (`*OD*CAP*)` have the same names across different Vt flavors. The plan is to shred them from the lvt and hvt files.

closed 2023-04-20 by allpan3

YiKangOY, 2023-12-19:
@allpan3 
Hi, 
I'm currently trying to run simulation with TSMC SRAM but encounter some problems. During (both RTL and Gate-level) simulation, it keeps showing 
```
Port 0 write address unknown at time xxx ns
Entire memory set to unknown.
```
May I contact you directly and maybe we could share some ideas on implementing Boom with TSMC PDK?

allpan3, 2023-12-20:
Hi @YiKangOY, did you try simulating the same code using Rocket core? My guess is that this is caused by the dual-port SRAMs used in Boom. Rocket core doesn't use any dual-port memory so your code should work.

I faced a similar issue with TSMC HPM n28. It has something to do with x-pessimism. The bottom line is that Boom expects no unknown data coming out from the SRAM.
Check you SRAM macro's verilog model and see if there's a way to initialize its data for simulation.
(I had a discussion with Jerry Zhao about a similar issue to this, and this seems to be the fix needed, but I haven't tested on my end.)

YiKangOY, 2023-12-20:
@allpan3 Thanks! I will try with Rocket core and see if there is further issue we can discuss on!

YiKangOY, 2024-01-11:
@allpan3  Hi, thanks for your advice, I have integrate TSMC SRAM into Rocket successfully. But problem occured when I simulated Boom. In dual port SRAM, sometimes the two ports will access the same address, with one writing and one reading, thus triggering memory contention. May I ask have you resolved this issue?

I think the memory accessing logic is controlled by barstool.  Is there any chance that we modify barstool so reading and writing to the same address within a cycle will not occur?

allpan3, 2024-01-14:
@YiKangOY Yes, that was the exact same problem I encountered.
From my understanding, this happens because Boom branch predictor sometimes allows reading and writing the same address in the same cycle, because those are just predictions and won't affect anything. However, the Verilog model of TSMC SRAM macro often sets memory content to X whenever "abnormal" operations take place. In my case, Boom eventually reads X from the BTB and uses that as the read address to the dcache SRAM, which triggers the logic to set all memory data to X. 

The specific behaviors may be different for your IP. I didn't get a chance to test the solution for this since my access to the TSMC IP has expired, but try to see if your Verilog model offers some flag to initialize all memory content in simulation.

YiKangOY, 2024-01-18:
@allpan3  Hi, thanks for your advice! We modify the TSMC sram verilog model slightly and then we can run RTL simulation on Boom with TSMC sram.

allpan3, 2024-01-19:
> @allpan3 Hi, thanks for your advice! We modify the TSMC sram verilog model slightly and then we can run RTL simulation on Boom with TSMC sram.

@YiKangOY Nice, glad to hear that. Can you share what change you made to the Verilog model?

YiKangOY, 2024-01-22:
@allpan3 Sure, but the detail of TSMC PDK is confidential, maybe we can discuss in private? My email address: oyykyook[at]gmail.com.