ucb-bar/chipyard #1444: vcu118 verilog genration failed in chipyard repository

ashutgupta28, 2023-04-17:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.8.1
Hash: a6a6a6

### OS Setup

Linux 5.15.0-69-generic #76~20.04.1-Ubuntu SMP Mon Mar 20 15:54:19 UTC 2023 x86_64 GNU/Linux

### Current Behavior

```
make SUB_PROJECT=vcu118 verilog 
Running with RISCV=/home/ssddls1/riscvl/riscvlyard/.conda-env/riscv-tools
mkdir -p /home/ssddls1/riscvl/riscvlyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config
cd /home/ssddls1/riscvl/riscvlyard && java -jar /home/ssddls1/riscvl/riscvlyard/generators/rocket-chip/sbt-launch.jar  ";project fpga_platforms; runMain chipyard.Generator  --target-dir /home/ssddls1/riscvl/riscvlyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config --name chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config --top-module chipyard.fpga.vcu118.VCU118FPGATestHarness --legacy-configs chipyard.fpga.vcu118:RocketVCU118Config "
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/ssddls1/riscvl/riscvlyard/.java_tmp
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by sbt.TrapExit$ (file:/home/ssddls1/.sbt/boot/scala-2.12.14/org.scala-sbt/sbt/1.5.5/run_2.12-1.5.5.jar)
WARNING: Please consider reporting this to the maintainers of sbt.TrapExit$
WARNING: System::setSecurityManager will be removed in a future release
[info] welcome to sbt 1.5.5 (N/A Java 17.0.3-internal)
[info] loading settings for project riscvlyard-build from plugins.sbt ...
[info] loading project definition from /home/ssddls1/riscvl/riscvlyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project constellation from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project tapeout from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/ssddls1/riscvl/riscvlyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (36697 settings) ...
[info] set current project to chipyardRoot (in build file:/home/ssddls1/riscvl/riscvlyard/)
[info] set current project to fpga_platforms (in build file:/home/ssddls1/riscvl/riscvlyard/)
[info] compiling 29 Scala sources to /home/ssddls1/riscvl/riscvlyard/fpga/target/scala-2.12/classes ...
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/TestHarness.scala:8:15: object fpgashells is not a member of package sifive
[error] import sifive.fpgashells.shell.xilinx.artyshell.{ArtyShell}
[error]               ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/TestHarness.scala:14:72: not found: type ArtyShell
[error] class ArtyFPGATestHarness(override implicit val p: Parameters) extends ArtyShell with HasHarnessSignalReferences {
[error]                                                                        ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/TestHarness.scala:30:23: not found: value clock_32MHz
[error]   val buildtopClock = clock_32MHz
[error]                       ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/HarnessBinders.scala:13:15: object fpgashells is not a member of package sifive
[error] import sifive.fpgashells.ip.xilinx.{IBUFG, IOBUF, PULLUP, PowerOnResetFPGAOnly}
[error]               ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/HarnessBinders.scala:74:26: value clock_32MHz is not a member of chipyard.fpga.arty.ArtyFPGATestHarness
[error]     withClockAndReset(th.clock_32MHz, th.ck_rst) {
[error]                          ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/arty/HarnessBinders.scala:74:42: value ck_rst is not a member of chipyard.fpga.arty.ArtyFPGATestHarness
[error]     withClockAndReset(th.clock_32MHz, th.ck_rst) {
[error]                                          ^
[... 273 lines cut ...]
[error]                           ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/vcu118/TestHarness.scala:38:47: not found: type PCIeVCU118EdgeShellPlacer
[error]   val edge      = Overlay(PCIeOverlayKey, new PCIeVCU118EdgeShellPlacer(this, PCIeShellInput()))
[error]                                               ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/vcu118/TestHarness.scala:38:79: not found: value PCIeShellInput
[error]   val edge      = Overlay(PCIeOverlayKey, new PCIeVCU118EdgeShellPlacer(this, PCIeShellInput()))
[error]                                                                               ^
[error] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/vcu118/TestHarness.scala:40:20: not found: value Overlay
[error]   val ddr2       = Overlay(DDROverlayKey, new DDR2VCU118ShellPlacer(this, DDRShellInput()))
[error]                    ^
[warn] /home/ssddls1/riscvl/riscvlyard/fpga/src/main/scala/vcu118/bringup/HarnessBinders.scala:58:39: non-variable type argument freechips.rocketchip.tilelink.TLBundle in type pattern freechips.rocketchip.util.HeterogeneousBag[freechips.rocketchip.tilelink.TLBundle] is unchecked since it is eliminated by erasure
[warn]       ports.head match { case tlPort: HeterogeneousBag[TLBundle] =>
[warn]                                       ^
[warn] one warning found
[error] 325 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 9 s, completed 17-Apr-2023, 12:00:22 pm
make: *** [/home/ssddls1/riscvl/riscvlyard/common.mk:107: /home/ssddls1/riscvl/riscvlyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config.fir] Error 1

```

### Expected Behavior

I have set up the chipyard repo and trying to generate verilog for vcu118 fpga. But after sourcing the environment I am getting error in various scala files. 
After sourcing the environment, for generating verilog file the command is `make SUB_PROJECT=vcu118 verilog`

michael-etzkorn, 2023-05-05:
It looks like your fpga-shells module was not initalized. Did you run the init-fpga script?

https://chipyard.readthedocs.io/en/stable/Prototyping/General.html#sources-and-submodule-setup

jerryhethatday, 2023-05-06:
@michael-etzkorn  Michael, Do you mind taking a look at my post in chipyard maillist? https://groups.google.com/g/chipyard/c/udqI1C0yU8k I want to create a baremetal workload for my vcu118 design. But I have no idea if FireMarshal is needed? Thanks

ashutgupta28, 2023-05-08:
@michael-etzkorn , 
I did what you said but it is showing me error 
```
/chipyard/generators/nvdla/nvdla.mk: No such file or directory
make: *** No rule to make target '/home/ashutosh/chipyard/generators/nvdla/nvdla.mk'.  Stop.
```

closed 2023-05-23 by ashutgupta28