ucb-bar/chipyard #1462: VCS compiliation errors with VLSI flow

krabo0om, 2023-05-06:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.0
Hash: 7475bfb1a05802ac4dfc3990a889f93164b8d798

### OS Setup

```Linux 4.18.0-425.19.2.el8_7.x86_64 #1 SMP Tue Apr 4 05:30:47 EDT 2023 x86_64 GNU/Linux```

```
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 8.5.2111
Release:        8.5.2111
Codename:       n/a
```
plain 1.9 conda

### Other Setup

VCS Version 2022-23/2022.06-SP2

### Current Behavior

VLSI command: `make sim-syn CONFIG=TinyRocketConfig BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-simple`

The `sim-rtl` target works fine but `sim-syn` and `sim-par` targets produce errors, e.g.:

```
Error-[URMI] Unresolved modules
/....../chipyard/vlsi/generated-src/chipyard.TestHarness.TinyRocketConfig/gen-collateral/TestHarness.sv, 175
"AsyncQueue bits_in_queue( .io_enq_clock (clock),  .io_enq_reset (reset),  .io_enq_valid (_ram_io_ser_in_valid),  .io_enq_bits (_ram_io_ser_in_bits),  .io_deq_clock (_chiptop_serial_tl_clock),  .io_deq_reset (reset),  .io_deq_ready (_chiptop_serial_tl_bits_in_ready),  .io_enq_ready (_bits_in_queue_io_enq_ready),  .io_deq_valid (_bits_in_queue_io_deq_valid),  .io_deq_bits (_bits_in_queue_io_deq_bits));"
  Module definition of above instance is not found in the design.
```

Similar errors are reported for other files, including 
```
..../gen-collateral/SerialAdapter.sv 
..../gen-collateral/SerialRAM.sv
..../gen-collateral/TestHarness.sv 
..../gen-collateral/TLRAM.sv
..../gen-collateral/UARTAdapter.sv
/chipyard/generators/rocket-chip/src/main/resources/vsrc/TestDriver.v
```

If those files are removed from the VCS command, it compiles the simulator successfully. These files are listed in `vlis/build/chipyard.TestHarness.TinyRocketConfig-ChipTop/sim-inputs.yml` for the parameter `sim.input.input_files`. 

### Expected Behavior

The simulator is build successfully. 

### Other Information

[hammer.log](https://github.com/ucb-bar/chipyard/files/11413214/hammer.log)

Standard ASAP7 flow, worked for 1.7.1, only changes where to the `env.yml` to set path and version of VCS/Cadence tools.

harrisonliew, 2023-05-07:
We just fixed this in #1452.

krabo0om, 2023-05-07:
Thanks for the fast reply, I can confirm that the error is fixed with those changes merged (01a03f5b5f06de8ede03e79bfadbb68a919f74db). I saw the PR but though it was unrelated because of the different error messages. 

Now I get #1265 when running with `*-debug`, but that should be unrelated and a setup problem on my end. With the VPD format it works.

closed 2023-05-07 by krabo0om