ucb-bar/chipyard #1488: `conda install -n base conda-lock` no longer works with existing Lockfiles due to new `conda-lock` release

rohanku, 2023-05-28:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Darwin rohanmbp.local 22.4.0 Darwin Kernel Version 22.4.0: Mon Mar  6 21:00:17 PST 2023; root:xnu-8796.101.5~3/RELEASE_X86_64 x86_64

conda-lock version 2.0.0

### Other Setup

Followed documentation for setting up Chipyard repository.

### Current Behavior

conda-lock is unable to read Lockfiles during `./build-setup.sh`

### Expected Behavior

conda-lock should be able to parse Lockfiles (works when constrained to version 1.4.0 using `conda install -n base conda-lock=1.4.0`)

jerryz123, 2023-05-29:
Are you saying that the amending the instructions to be `conda install -n base conda-lock=1.4.0` (forcing a version) will fix the issue? If so, can you just PR it to main?

fangrouli, 2023-06-17:
> I can verify that downgrading conda-lock from v2.0.0 to v1.4.0 stops the issue as described 
> in fact occurred when attempting to run `build-setup.sh`.

Exactly same behaviour. Recommends quoting this in the documentation.

jspkay, 2023-06-26:
I had the same issue and confirm that downgrading conda-lock to 1.4.0 solved the problem