ucb-bar/chipyard #1493: Validation errors

notooth1, 2023-05-31:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: stable

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`
(base) notooth@192:chipyard$ uname -a
Linux 192.168.0.101 4.18.0-425.3.1.el8.x86_64 #1 SMP Tue Nov 8 14:08:25 EST 2022 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	AlmaLinux
Description:	AlmaLinux release 8.7 (Stone Smilodon)
Release:	8.7
Codename:	StoneSmilodon

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

(base) notooth@192:Downloads$ git clone https://github.com/ucb-bar/chipyard.git
(base) notooth@192:Downloads$ cd chipyard
(base) notooth@192:chipyard$ git checkout stable
(base) notooth@192:chipyard$ ./build-setup.sh riscv-tools
...
pydantic.error_wrappers.ValidationError: 2 validation errors for Lockfile
package -> 383 -> optional
  field required (type=value_error.missing)
package -> 387 -> optional
  field required (type=value_error.missing)

### Expected Behavior

build successfully

michael-etzkorn, 2023-06-01:
I am seeing the same thing on my end.

michael-etzkorn, 2023-06-01:
Seems to be the same as this issue: https://github.com/ucb-bar/chipyard/issues/1488

Trying the solution there momentarily.