ucb-bar/chipyard #1512: Malformed input or input contains unmappable characters: shapeless/$tilde$qmark$greater$λ.class when running VCS simulation

T-K-233, 2023-06-14:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.1+
Hash: [d287fed](https://github.com/ucb-bar/chipyard/commit/d287fed654d1d5f62d1ddae97b6c56956edf7dad)

### OS Setup

BWRC cluster machine:

```
> uname -a
Linux bwrcix-1.eecs.berkeley.edu 3.10.0-1160.83.1.el7.x86_64 #1 SMP Mon Dec 19 10:44:06 UTC 2022 x86_64 GNU/Linux

> lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	RedHatEnterpriseServer
Description:	Red Hat Enterprise Linux Server release 7.9 (Maipo)
Release:	7.9
Codename:	Maipo

> printenv
CONDA_BACKUP_LD_LIBRARY_PATH=/bwrcq/C/chiyufeng/tapeout/chipyard-ide/.conda-env/riscv-tools/lib:/tools/support/lsf/9.1/linux2.6-glibc2.3-x86_64/lib
SYNPLCTYD_LICENSE_FILE=1701@bwrcflex-1.eecs.berkeley.edu:1701@bwrcflex-2.eecs.berkeley.edu
MANPATH=/tools/support/lsf/9.1/man:
REMCOMFDTD_LICENSE_FILE=1713@sunv210-2.eecs.berkeley.edu
XDG_SESSION_ID=13190
JAVA_LD_LIBRARY_PATH=/bwrcq/C/chiyufeng/tapeout/chipyard/.conda-env/lib/jvm/lib/server
GUESTFISH_INIT=\e[1;34m
HOSTNAME=bwrcix-1.eecs.berkeley.edu
MAGMA_LICENSE_FILE=1701@bwrcflex-1.eecs.berkeley.edu:1701@bwrcflex-2.eecs.berkeley.edu
TERM=xterm
SHELL=/bin/bash
SYNPLICITYD_LICENSE_FILE=1701@sunv40z-1.eecs.berkeley.edu:1701@sunv20z-1.eecs.berkeley.edu
HISTSIZE=1000
ANSYSLMD_LICENSE_FILE=1706@bwrcflex-1.eecs.berkeley.edu:1706@bwrcflex-2.eecs.berkeley.edu:1055@lsr.me.berkeley.edu
SSH_CLIENT=128.32.62.222 58450 22
CONDA_SHLVL=2
LORENTZ_LICENSE_FILE=27017@bwrcflex-1.eecs.berkeley.edu
CONDA_PROMPT_MODIFIER=(/bwrcq/C/chiyufeng/tapeout/chipyard/.conda-env) 
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=/bwrcq/C/chiyufeng/tapeout/chipyard-ide/.conda-env/share/glib-2.0/schemas
LM_PROJECT=bwrc_users
LSF_SERVERDIR=/tools/support/lsf/9.1/linux2.6-glibc2.3-x86_64/etc
QTDIR=/usr/lib64/qt-3.3
OLDPWD=/tools/C/chiyufeng/tapeout/chipyard/tests
QTINC=/usr/lib64/qt-3.3/include
CONDA_BACKUP_RISCV=/bwrcq/C/chiyufeng/tapeout/chipyard-ide/.conda-env/riscv-tools
SSH_TTY=/dev/pts/110
LSF_LIBDIR=/tools/support/lsf/9.1/linux2.6-glibc2.3-x86_64/lib
QT_GRAPHICSSYSTEM_CHECKED=1
SYNFORAD_LICENSE_FILE=1726@sunv210-2.eecs.berkeley.edu
VERPLEX_LICENSE_FILE=5280@sunv210-2.eecs.berkeley.edu
AVANTD_LICENSE_FILE=1701@bwrcflex-1.eecs.berkeley.edu
USER=chiyufeng
TAVEREN_LICENSE_FILE=1701@bwrcflex-1.eecs.berkeley.edu:1701@bwrcflex-2.eecs.berkeley.edu
LD_LIBRARY_PATH=/bwrcq/C/chiyufeng/tapeout/chipyard/.conda-env/riscv-tools/lib:/bwrcq/C/chiyufeng/tapeout/chipyard-ide/.conda-env/riscv-tools/lib:/tools/support/lsf/9.1/linux2.6-glibc2.3-x86_64/lib
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=01;05;37;41:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=01;36:*.au=01;36:*.flac=01;36:*.mid=01;36:*.midi=01;36:*.mka=01;36:*.mp3=01;36:*.mpc=01;36:*.ogg=01;36:*.ra=01;36:*.wav=01;36:*.axa=01;36:*.oga=01;36:*.spx=01;36:*.xspf=01;36:
SYNPLIFY_LICENSE_TYPE=synplifypremierdp
CONDA_EXE=/tools/C/chiyufeng/documents/miniconda3/bin/conda
XILINXD_LICENSE_FILE=2200@bwrcflex-1.eecs.berkeley.edu:2200@bwrcflex-2.eecs.berkeley.edu
APACHEDA_LICENSE_FILE=27019@sunv40z-1.eecs.berkeley.edu:27019@sunv20z-1.eecs.berkeley.edu
SNPSLMD_LICENSE_FILE=1701@bwrcflex-1.eecs.berkeley.edu:1701@bwrcflex-2.eecs.berkeley.edu
FTP_PASSIVE_MODE=yes
_CE_CONDA=
GUESTFISH_PS1=\[\e[1;32m\]><fs>\[\e[0;31m\] 
[... 544 lines cut ...]
[error] 	at scala.collection.parallel.Task.tryLeaf(Tasks.scala:56)
[error] 	at scala.collection.parallel.Task.tryLeaf$(Tasks.scala:50)
[error] 	at scala.collection.parallel.ParIterableLike$FlatMap.tryLeaf(ParIterableLike.scala:1078)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.internal(Tasks.scala:170)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.internal$(Tasks.scala:157)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.internal(Tasks.scala:440)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute(Tasks.scala:150)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute$(Tasks.scala:149)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.compute(Tasks.scala:440)
[error] 	at java.util.concurrent.RecursiveAction.exec(RecursiveAction.java:189)
[error] 	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:289)
[error] 	at java.util.concurrent.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1056)
[error] 	at java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1692)
[error] 	at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:175)
[error] (assembly) java.nio.file.InvalidPathException: Malformed input or input contains unmappable characters: shapeless/$tilde$qmark$greater$λ.class
[error] Total time: 52 s, completed Jun 13, 2023 2:01:01 PM
make: *** [/bwrcq/C/chiyufeng/tapeout/chipyard-ide/common.mk:119: /bwrcq/C/chiyufeng/tapeout/chipyard-ide/.classpath_cache/chipyard.jar] Error 1
(/bwrcq/C/chiyufeng/tapeout/chipyard-ide/.conda-env) chiyufeng@bwrcrdsl-4:/tools/C/chiyufeng/tapeout/chipyard-ide/sims/vcs $ 
```

[image]

### Expected Behavior

Runs VCS simulation

jerryz123, 2023-06-19:
If you can still reproduce this, can you try #1528? That should fix this

chithize, 2023-07-15:
I tried this, the problem is still there

chithize, 2023-07-16:
when I start the build, I noticed that there is the below warning information related to locale setting:
bin/bash: warning: setlocale: LC_ALL: cannot change locale ("en_US.UTF-8") 

cd /data2/chipyard && java -jar /data2/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.ivy.home=/data2/chipyard/.ivy2 -Dsbt.global.base=/data2/chipyard/.sbt -Dsbt.boot.directory=/data2/chipyard/.sbt/boot/ -Dsbt.color=always -Dsbt.supershell=false -Dsbt.server.forcestart=true ";project fpga_platforms; set assembly / assemblyOutputPath := file(\"/data2/chipyard/.classpath_cache/chipyard.jar\"); assembly" && touch /data2/chipyard/.classpath_cache/chipyard.jar 

/bin/bash: warning: setlocale: LC_ALL: cannot change locale ("en_US.UTF-8")

krabo0om, 2023-11-07:
I just got the same error. Was trying to run a VCS simulation but our license server is down. I ctrl-C the waiting (i.e., VCS) and after that I got the above error when runnning the VLSI flow. Switching into a new terminal solved the problem. 

I am using Chipyard 1.10 without #1528.

csgxiong, 2023-11-20:
@chithize Did you solved your problems now?
I met the same errors when using Chipyard V1.10.0.

chithize, 2023-11-20:
I had resolved , it is a bit long ago now, let me get it back for you.     my email is  hcruise@acemicro.cn

csgxiong, 2023-11-20:
> I had resolved , it is a bit long ago now, let me get it back for you. my email is [hcruise@acemicro.cn](mailto:hcruise@acemicro.cn)

Thanks. Counld you show me more information about it or shall I contact you with your email ?