ucb-bar/chipyard #1557: cann't down load br.xxxx.img.zip from web

Yuxin-Yu, 2023-07-15:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

I used the command "build_setup.sh riscv-tools" to complete the installation of the Chipyard runtime environment, and installed the firemarshal conda environment through the README file. The configuration in the Marshal config. yaml file is as follows:
`Board dir: 'boards/prototype'  `
When I run the command:
`./Marshal - v - d build boards/prototype/base-workloads/br-base. json. `
Error in downloading the br.222b.img.zip file. The error is as follows:
```
DEBUG: Running task/home/yyx/riscv/projects/chipard/software/firmware/images/prototype/br.22b/br.222b.img because one
Of its targets does not exist any more:/home/yyx/riscv/projects/chipard/software/fireMarshal/images/prototype/br.2222b/br
222b.img
Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/pro
Prototype/br.222b/br.222b.img.zip
DEBUG: urlretrieve exception: HTTP Error 404: Not Found
Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/prototype/br.222b/br.222b.img.zip 
DEBUG: urlretrieve exception: HTTP Error 404: Not Found
Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/pro
Prototype/br.222b/br.222b.img.zip
DEBUG: urlretrieve exception: HTTP Error 404: Not Found
......
DEBUG: already applied: /home/yyx/riscv/projects/chipyard/software/firemarshal/boards/default/distros/br/buildroot/dl/ncurs
es/ncurses-6.1-20190609-patch.sh.bz2
DEBUG: to be applied  : /home/yyx/riscv/projects/chipyard/software/firemarshal/boards/default/distros/br/buildroot/dl/ncurs
es/ncurses-6.1-20190609-patch.sh.bz2
DEBUG: make[1]: *** [package/pkg-generic.mk:250: /home/yyx/riscv/projects/chipyard/software/firemarshal/boards/default/dist
ros/br/buildroot/output/build/host-ncurses-6.1/.stamp_patched] Error 1
DEBUG: make: *** [Makefile:84: _all] Error 2
TaskError - taskid:/home/yyx/riscv/projects/chipyard/software/firemarshal/images/prototype/br.222b/br.222b.img
PythonAction Error
Traceback (most recent call last):
  File "/home/yyx/miniforge3/envs/firemarshal/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/yyx/riscv/projects/chipyard/software/firemarshal/boards/prototype/distros/br/br.py", line 255, in buildBaseIm
age
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/home/yyx/riscv/projects/chipyard/software/firemarshal/wlutil/wlutil.py", line 527, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.

ERROR: Failed to build workload br-base.json
Log available at: /home/yyx/riscv/projects/chipyard/software/firemarshal/logs/br-base-build-2023-07-15--02-34-20-3XGQWP4BPR
GW9NTJ.log
```
Based on the error message, I found that only the corresponding images of the fireship development board exist in the firemarshal-public-br-images repository, and there are no images of prototype and chipboard types. Personally, I suspect that this is a code bug?
And this is the log file:
[br-base-build-2023-07-15--02-34-20-3XGQWP4BPRGW9NTJ.log](https://github.com/ucb-bar/chipyard/files/12060136/br-base-build-2023-07-15--02-34-20-3XGQWP4BPRGW9NTJ.log)

### Other Setup

no

### Current Behavior

no

### Expected Behavior

no

### Other Information

no

jerryz123, 2023-07-15:
@abejgonzalez  cache problem?

Yuxin-Yu, 2023-07-15:
In addition, I found a similar issue with version 1.8.1, with the following error:
```
DEBUG: make[1]: *** [package/pkg-generic.mk:282: /home/yyx/riscv/projects/Chipyard-1.8.1/software/firemarshal/boards/default/distros/br/buildroot/output/build/toolchain-external-custom/.stamp_configured] Error 1
DEBUG: make: *** [Makefile:84: _all] Error 2
TaskError - taskid:/home/yyx/riscv/projects/Chipyard-1.8.1/software/firemarshal/images/br.603d.img
PythonAction Error
Traceback (most recent call last):
  File "/home/yyx/miniforge3/envs/firemarshal1.8/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/yyx/riscv/projects/Chipyard-1.8.1/software/firemarshal/boards/prototype/distros/br/br.py", line 184, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/home/yyx/riscv/projects/Chipyard-1.8.1/software/firemarshal/wlutil/wlutil.py", line 531, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.

ERROR: Failed to build workload br-base.json
Log available at: /home/yyx/riscv/projects/Chipyard-1.8.1/software/firemarshal/logs/br-base-build-2023-07-15--09-31-01-43S4CNHV7NK00L3N.log
ERROR: FAILURE: 1 builds failed
```
And this i
[br-base-build-2023-07-15--09-31-01-43S4CNHV7NK00L3N.log](https://github.com/ucb-bar/chipyard/files/12061082/br-base-build-2023-07-15--09-31-01-43S4CNHV7NK00L3N.log)
s the log file:

abejgonzalez, 2023-07-16:
The error is not related to caching but with buildroot based on this error message.

```
Applying ncurses-6.1-20190609-patch.sh.bz2 using bzip2:
2023-07-15 10:34:35,269 [run         ] [DEBUG]  Error: duplicate filename 'ncurses-6.1-20190609-patch.sh.bz2'
```

In the buildroot submodule I would try running `git clean -dfx` to clean it entirely

Yuxin-Yu, 2023-07-17:
Hi @abejgonzalez .I have ran `git clean -dfx` under buildroot as you said.But I got a new error,which like:
```
DEBUG: patching file doc/html/ada/funcs/S.htm
DEBUG: patching file doc/html/ada/funcs/T.htm
DEBUG: patch: **** Can't rename file doc/html/ada/funcs/T.htm.ozNt6bm to doc/html/ada/funcs/T.htm : Too many open files
DEBUG: make[1]: *** [package/pkg-generic.mk:250: /home/yyx/riscv/projects/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/host-ncurses-6.1/.stamp_patched] Error 2
DEBUG: make: *** [Makefile:84: _all] Error 2
TaskError - taskid:/home/yyx/riscv/projects/chipyard/software/firemarshal/images/prototype/br.222b/br.222b.img
PythonAction Error
Traceback (most recent call last):
  File "/home/yyx/miniforge3/envs/firemarshal/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/yyx/riscv/projects/chipyard/software/firemarshal/boards/prototype/distros/br/br.py", line 255, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/home/yyx/riscv/projects/chipyard/software/firemarshal/wlutil/wlutil.py", line 527, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.

ERROR: Failed to build workload br-base.json
Log available at: /home/yyx/riscv/projects/chipyard/software/firemarshal/logs/br-base-build-2023-07-17--01-04-01-06JFHXLKG9T9RGXK.log
ERROR: FAILURE: 1 builds failed
(firemarshal) yyx@yyx-virtual-machine:~/riscv/projects/chipyard/software/firemarshal$ 
```

In addition,here is the log file:
[br-base-build-2023-07-17--01-04-01-06JFHXLKG9T9RGXK.log](https://github.com/ucb-bar/chipyard/files/12065522/br-base-build-2023-07-17--01-04-01-06JFHXLKG9T9RGXK.log)

abejgonzalez, 2023-07-17:
That is a build-root Linux issue. To work around this you can run the following script: https://github.com/ucb-bar/chipyard/blob/main/scripts/fix-open-files.sh. If that script prints the warning then you need to fix your open files limit, restart your terminal, and run the build again (remember to source the Chipyard `env.sh` script before you do anything in Chipyard).

closed 2023-07-17 by Yuxin-Yu

Jerryy959, 2024-10-27:
> ### Background Work
> * [x]  Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
> * [x]  Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
> * [...]

I met the same issue, https://github.com/firesim/FireMarshal/issues/312, how can you solved this?