ucb-bar/chipyard #1562: Error during Rocket Core build for verilator on Chipyard 1.9.1:: 'std::optional' is only available from C++17 onwards

Nerotos, 2023-07-19:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.1
Hash: 968b20

### OS Setup

LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	Rocky
Description:	Rocky Linux release 8.7 (Green Obsidian)
Release:	8.7
Codename:	GreenObsidian

g++ --version: 
g++ (conda-forge gcc 12.2.0-19) 12.2.0

### Other Setup

Chipyard v1.9.1 was built following the documentation by running ./build-setup.sh riscv-tools

### Current Behavior

When calling make in the sims/verilator folder the build process fails with this error:

testversion_cy/chipyard/.conda-env/riscv-tools/include/fesvr/../riscv/cfg.h:99:8: error: 'optional' in namespace 'std' does not name a template type
   99 |   std::optional<reg_t>               start_pc;
      |        ^~~~~~~~
testversion_cy/chipyard/.conda-env/riscv-tools/include/fesvr/../riscv/cfg.h:99:3: note: 'std::optional' is only available from C++17 onwards
   99 |   std::optional<reg_t>               start_pc;
      |   ^~~
make[1]: *** [VTestHarness.mk:74: SimDTM.o] Error 1

I had another issue with verilator, but that was on the 1.10.0 release, which required a newer version of verilator. Here it seems like the -std=c++17 flag should be placed somewhere but is missing, and I am not sure where it should go here.

### Expected Behavior

Build completes without errors

jerryz123, 2023-07-19:
Can you try running `make clean` in the sims/verilator directory first?

Nerotos, 2023-07-22:
It turns out that something in my system was setting VERILATOR_ROOR, even though I removed it from my bashrc. Unsetting the variable fixed my issues.

closed 2023-07-22 by Nerotos