ucb-bar/chipyard #1565: Error during Verilog generation (firtool pass) for a simple SInt-based Queue peripheral

milovanovic, 2023-07-21:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.0 and subsequent ones (1.9.1 and 1.10.0).
Hash: 7475bfb

Works fine on 1.8.1 and before.

### OS Setup

Any Linux

### Other Setup

N/A

### Current Behavior

When using the Chipyard's makefile flow to generate Verilog for `Queue` peripheral with `SInt` (or `FixedPoint`) data type, the following problem occurs during the execution of `firtool` command inside [common.mk](https://github.com/ucb-bar/chipyard/blob/7475bfb1a05802ac4dfc3990a889f93164b8d798/common.mk):
```scala
Decoupled.scala:273:95: error: memories should be flattened before running LowerMemory
Decoupled.scala:273:95: note: see current operation: %4:2 = "firrtl.mem"() {annotations = [], depth = 2 : i64, name = "ram", nameKind = #firrtl<name_kind droppable_name>, portAnnotations = [[], []], portNames = ["MPORT", "io_deq_bits_MPORT"], readLatency = 0 : i32, ruw = 0 : i32, writeLatency = 1 : i32} : () -> (!firrtl.bundle<addr: uint<1>, en: uint<1>, clk: clock, data: sint<16>, mask: uint<1>>, !firrtl.bundle<addr: uint<1>, en: uint<1>, clk: clock, data flip: sint<16>>)
```

The Queue code example is just a simple `Queue` "connected" to the bus, the code can be found [here](https://github.com/milovanovic/chipyard/blob/1.9.0_queue/generators/chipyard/src/main/scala/Queue/QueueExample.scala).

This problem occurred somewhere between tags 1.8.1 and 1.9.0 because the same makefile flow works fine on Chipyard version 1.8.1 but fails on version 1.9.0 and above. It should be noted that no problem is observed when `UInt` data type is supplied to `Queue`. The main issue is that the flow for `SInt`-based Queue is broken.

Please see [this](https://github.com/milovanovic/chipyard/tree/1.9.0_queue) for more details on how to reproduce the error.

### Expected Behavior

Generates Verilog for `Queue` peripheral with `SInt` (or `FixedPoint`) input data type, just as it does for `UInt`.

On Chipyard version [1.8.1](https://github.com/milovanovic/chipyard/tree/1.8.1_queue) everything worked as expected.

### Other Information

See Chipyard [v1.9.0](https://github.com/milovanovic/chipyard/tree/1.9.0_queue) to reproduce the SInt Queue error.

See Chipyard [v1.8.1](https://github.com/milovanovic/chipyard/tree/1.8.1_queue) to observe the expected behavior.

jerryz123, 2023-07-25:
I was able to reproduce this pretty easily, any `Queue[SInt]` seems to break for me. Can you work around this for now by creating a `SIntQueue` wrapper that converts type to `UInt` before instantiating the queue?

milovanovic, 2024-01-02:
Actually, besides `Queue` the very same problem occurs also for memories, i.e. `SInt`-based `Mem` and `SyncReadMem`. Please see [this](https://github.com/milovanovic/QueueDemo/tree/MemDemo) for more details on how to reproduce the error.