ucb-bar/chipyard #1579: Error facing while compiling the Tiny Rocket Core

ashutgupta28, 2023-08-22:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I am trying to run the Tiny Rocket Core in Xilinx zcu102 FPGA. I tried with Default Rocket Core and it is working fine.
Is this Tiny Core required additional changes in the Scala Files.

### Expected Behavior

I am using this configuration for Tiny Rocket Core.

[image]

Error I am getting

```
zcu102.ZCU102FPGATestHarness.RISCVTinyZCU102Config --name chipyard.fpga.zcu102.ZCU102FPGATestHarness.RISCVTinyZCU102Config --top-module chipyard.fpga.zcu102.ZCU102FPGATestHarness --legacy-configs chipyard.fpga.zcu102:RISCVTinyZCU102Config
make[1]: Entering directory '/home/ssddls1/ashutosh/ashufpga/fpga/src/main/resources/zcu102/sdboot'
make[1]: Nothing to be done for 'bin'.
make[1]: Leaving directory '/home/ssddls1/ashutosh/ashufpga/fpga/src/main/resources/zcu102/sdboot'
ZCU102 FPGA Base Clock Freq: 100.0 MHz
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] 	at ... ()
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36
[error] 	at ... ()
[error] 	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
[error] Caused by: java.util.NoSuchElementException: None.get
[error] 	at scala.None$.get(Option.scala:529)
[error] 	at scala.None$.get(Option.scala:527)
[error] 	at chipyard.fpga.zcu102.ZCU102FPGATestHarness.<init>(TestHarness.scala:84)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
[error] 	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
[error] 	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36
[error] 	at chisel3.Module$.do_apply(Module.scala:53)
[error] 	at chisel3.stage.phases.Elaborate.$anonfun$transform$2(Elaborate.scala:33)
[error] 	at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:720)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:715)
[error] 	at chisel3.stage.phases.Elaborate.$anonfun$transform$1(Elaborate.scala:33)
[error] 	at scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:245)
[error] 	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error] 	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error] 	at scala.collection.TraversableLike.flatMap(TraversableLike.scala:245)
[error] 	at scala.collection.TraversableLike.flatMap$(TraversableLike.scala:242)
[error] 	at scala.collection.AbstractTraversable.flatMap(Traversable.scala:108)
[error] 	at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:28)
[error] 	at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:21)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[... 50 lines cut ...]
[error] 	at firrtl.options.Translator.transform(Phase.scala:248)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:248)
[error] 	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error] 	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at logger.Logger$.makeScope(Logger.scala:135)
[error] 	at firrtl.options.Stage.transform(Stage.scala:47)
[error] 	at firrtl.options.Stage.execute(Stage.scala:58)
[error] 	at firrtl.options.StageMain.main(Stage.scala:71)
[error] 	at chipyard.Generator.main(Generator.scala)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
[error] 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 7 s, completed 17-Aug-2023, 1:54:27 pm
make: *** [/home/ssddls1/ashutosh/ashufpga/common.mk:107: /home/ssddls1/ashutosh/ashufpga/fpga/generated-src/chipyard.fpga.zcu102.ZCU102FPGATestHarness.RISCVTinyZCU102Config/chipyard.fpga.zcu102.ZCU102FPGATestHarness.RISCVTinyZCU102Config.fir] Error 1
```

jerryz123, 2023-08-22:
This is a very old Chipyard (1.5.0), and our ability to support it is limited. Can you try again with the most recent version?

ashutgupta28, 2023-08-22:
@jerryz123 
I am using chipard version 1.9.1

jerryz123, 2023-08-22:
Did you add ZCU102 support to your own branch? I don't believe the upstream has that board supported.

ashutgupta28, 2023-08-22:
@jerryz123  
Yes. I have add ZCU102 supported files. I have also tested Rocket Core in FPGA and it is working also . 
The command I am using to rocket core on FPGA is `make SUB_PROEJECT=zcu102 verilog` and it is working. 
Similarly I have tested different configuration.

But in Tiny core even the default configuration is not working . Here I am giving make `SUB_PROEJECT=zcu102 CONFIG=TInyRocketConfig verilog` but it is giving me error. I am not able to find out what is the error .

jerryz123, 2023-08-22:
It is difficult for others to help you without being able to see what you implemented.
`[error] 	at chipyard.fpga.zcu102.ZCU102FPGATestHarness.<init>(TestHarness.scala:84)` suggests that line 84 in a TestHarness.scala you implemented is the source of the problem.

ashutgupta28, 2023-08-23:
But if this would be the case, then my different configuration is working ? Not only this I changed configuration from big core to Medium core , thats also working. 
Only it is getting problem while running in Tiny Core.

I found somewhere that  Rocket TinyCore uses a default scratchpad instead of a backing memory. This scratchpad 0x80000000 to 0x80003fff is overlapping with the memport's address range and I have remove the backing memory and L2 cache as well. like

```
class TinyRocketConfig extends Config(
  new chipyard.config.WithTLSerialLocation(
    freechips.rocketchip.subsystem.FBUS,
    freechips.rocketchip.subsystem.PBUS) ++                          // attach TL serial adapter to f/p busses
  new freechips.rocketchip.subsystem.WithIncoherentBusTopology ++    // use incoherent bus topology
  new freechips.rocketchip.subsystem.WithNBanks(0) ++                // remove L2$
  new freechips.rocketchip.subsystem.WithNoMemPort ++                // remove backing memory
  new freechips.rocketchip.subsystem.With1TinyCore ++                // single tiny rocket-core
  new chipyard.config.AbstractConfig)
```

But still not able to figure out what is the issue. 
**Note** :  
My design is compiling using verilator.

T-K-233, 2023-12-18:
I have run into similar issue when building for Arty FPGA designs.

I think TinyCore requires a vastly different configuration for the memory system compared to other Core Configs. If the resource is limited, could you try use the SmallCore? 

```scala
new freechips.rocketchip.subsystem.WithNSmallCores(1)
```

It's still larger than TinyCore, but it shares the same system configuration as Medium and Big.

Hope this is helpful.

lyxxcz, 2024-01-29:
Hello，I have a Xilinx ZCU102 and want to run the Rocket Core on it ，but don't know how to support it，can you tell me how you did it? Thank you.

Rhythmicc, 2024-06-21:
I'm also trying to simulate Rocket Core on ZCU102, using the peripheral code provided by the guy who opened the issue. Then, I fixed some syntax errors that occur in newer versions, but I still can't generate the bitstream. The error is as follows:

```
Exception in thread "main" java.lang.IllegalArgumentException: requirement failed: Clock group uncore has non-homogeneous requested ClockParameters List(ClockParameters(100.0,50.0), ClockParameters(100.0,50.0), ClockParameters(500.0,50.0), ClockParameters(100.0,50.0), ClockParameters(100.0,50.0))
	at ... ()
	at chipyard.clocking.ClockGroupCombiner.$anonfun$sinkFn$3(ClockGroupCombiner.scala:53)
	at scala.collection.immutable.ArraySeq.foldLeft(ArraySeq.scala:222)
	at chipyard.clocking.ClockGroupCombiner.$anonfun$sinkFn$1(ClockGroupCombiner.scala:46)
	at scala.collection.immutable.List.map(List.scala:246)
	at scala.collection.immutable.List.map(List.scala:79)
	at freechips.rocketchip.diplomacy.MixedAdapterNode.mapParamsU(Nodes.scala:1492)
	at freechips.rocketchip.diplomacy.MixedNode.liftedTree3$1(Nodes.scala:1176)
	at freechips.rocketchip.diplomacy.MixedNode.uiParams$lzycompute(Nodes.scala:1173)
	at freechips.rocketchip.diplomacy.MixedNode.uiParams(Nodes.scala:1172)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$uoParams$1(Nodes.scala:1171)
	at scala.collection.immutable.List.map(List.scala:246)
	at scala.collection.immutable.List.map(List.scala:79)
	at freechips.rocketchip.diplomacy.MixedNode.uoParams$lzycompute(Nodes.scala:1171)
	at freechips.rocketchip.diplomacy.MixedNode.uoParams(Nodes.scala:1171)
	at freechips.rocketchip.diplomacy.MixedNode.liftedTree3$1(Nodes.scala:1176)
	at freechips.rocketchip.diplomacy.MixedNode.uiParams$lzycompute(Nodes.scala:1173)
	at freechips.rocketchip.diplomacy.MixedNode.uiParams(Nodes.scala:1172)
	at freechips.rocketchip.diplomacy.MixedNode.edgesIn$lzycompute(Nodes.scala:1195)
	at freechips.rocketchip.diplomacy.MixedNode.edgesIn(Nodes.scala:1195)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$bundleIn$2(Nodes.scala:1213)
	at chisel3.experimental.prefix$.apply(prefix.scala:50)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$bundleIn$1(Nodes.scala:1213)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at freechips.rocketchip.diplomacy.MixedNode.bundleIn$lzycompute(Nodes.scala:1213)
	at freechips.rocketchip.diplomacy.MixedNode.bundleIn(Nodes.scala:1213)
	at freechips.rocketchip.diplomacy.MixedNode.in(Nodes.scala:1270)
	at freechips.rocketchip.diplomacy.MixedNode.instantiate(Nodes.scala:1281)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$16(LazyModule.scala:343)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:343)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:307)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:406)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$23$1(LazyModule.scala:423)
	at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:420)
	at freechips.rocketchip.prci.ClockGroupResetSynchronizer$Impl.<init>(ResetSynchronizer.scala:33)
	at freechips.rocketchip.prci.ClockGroupResetSynchronizer.$anonfun$module$2(ResetSynchronizer.scala:32)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at freechips.rocketchip.prci.ClockGroupResetSynchronizer.module$lzycompute(ResetSynchronizer.scala:32)
	at freechips.rocketchip.prci.ClockGroupResetSynchronizer.module(ResetSynchronizer.scala:32)
	at freechips.rocketchip.prci.ClockGroupResetSynchronizer.module(ResetSynchronizer.scala:30)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$15(LazyModule.scala:336)
	at chisel3.Module$.do_apply(Module.scala:53)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$14(LazyModule.scala:336)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$13(LazyModule.scala:336)
	at scala.Option.getOrElse(Option.scala:201)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:334)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:310)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:307)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:406)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$23$2(LazyModule.scala:421)
	at chisel3.withClockAndReset$.apply(MultiClock.scala:26)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$23$1(LazyModule.scala:421)
	at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:420)
	at freechips.rocketchip.prci.Domain$Impl.<init>(ClockDomain.scala:12)
	at freechips.rocketchip.prci.Domain.$anonfun$module$1(ClockDomain.scala:11)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at freechips.rocketchip.prci.Domain.module$lzycompute(ClockDomain.scala:11)
	at freechips.rocketchip.prci.Domain.module(ClockDomain.scala:11)
[... cut ...]
	at chipyard.fpga.zcu102.ZCU102FPGATestHarness.module(TestHarness.scala:92)
	at chipyard.fpga.zcu102.ZCU102FPGATestHarness.module(TestHarness.scala:26)
	at chipyard.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:37)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
make: *** [/root/chipyard/common.mk:140: /root/chipyard/fpga/generated-src/chipyard.fpga.zcu102.ZCU102FPGATestHarness.RocketZCU102Config/chipyard.fpga.zcu102.ZCU102FPGATestHarness.RocketZCU102Config.fir] Error 1
```