ucb-bar/chipyard #1604: Tracing the instruction being executed from VCS simulation waveform

YiKangOY, 2023-09-22:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hi, I want to trace the execution of the instruction being executed from the VCS simulation Waveform. Is there any RTL signal corresponds to the instruction that can be traced during simulation? Thanks!that 

### Motivating Example

The instruction (machine code) that being executed.

jerryz123, 2023-09-26:
This depends on the core implementation. 
For the chisel-based cores (Rocket/Shuttle/Boom), enabling the Trace out will expose IO at the ChipTop corresponding to instruction commigs

Michael-Doran, 2025-04-23:
I've enabled the flag "enableTraceCoreIngress" in the RocketCore.scala file, and I've added "WithTraceIOPunchtrhough" to my configuration, but chisel seems to be optimizing away these signals.  Any thoughts?