ucb-bar/chipyard #1606: Issue with guestmount building on WSL2

matsbror, 2023-09-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: commit b7644b2455cc4bae190e811a5d8085f3aad85b87 (HEAD, tag: 1.10.0, origin/stable)

### OS Setup

Trying to build under WSL2, Ubuntu 22.04. According to a closed issue, this should work.

```
$ uname -a
Linux BRONSON 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ conda list
# packages in environment at /home/mats/hpca/conda:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
annotated-types           0.5.0              pyhd8ed1ab_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
boltons                   23.0.0             pyhd8ed1ab_0    conda-forge
brotli-python             1.1.0           py310hc6cd4ac_0    conda-forge
brotlipy                  0.7.0           py310h5764c6d_1005    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cachecontrol              0.13.1             pyhd8ed1ab_0    conda-forge
cachecontrol-with-filecache 0.13.1             pyhd8ed1ab_0    conda-forge
cachy                     0.3.0              pyhd8ed1ab_1    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2fee648_5    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
click-default-group       1.2.4              pyhd8ed1ab_0    conda-forge
clikit                    0.6.2              pyhd8ed1ab_2    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     23.7.4          py310hff52083_0    conda-forge
conda-libmamba-solver     23.7.0             pyhd8ed1ab_0    conda-forge
conda-lock                1.4.0              pyhd8ed1ab_2    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
crashtest                 0.4.1              pyhd8ed1ab_0    conda-forge
cryptography              41.0.4          py310h75e40e8_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distlib                   0.3.7              pyhd8ed1ab_0    conda-forge
ensureconda               1.4.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
filelock                  3.12.4             pyhd8ed1ab_0    conda-forge
fmt                       10.1.1               h00ab1b0_0    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
html5lib                  1.1                pyh9f0ad1d_0    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
idna                      3.4                pyhd8ed1ab_0    conda-forge
importlib-metadata        6.8.0              pyha770c72_0    conda-forge
importlib_metadata        6.8.0                hd8ed1ab_0    conda-forge
jaraco.classes            3.3.0              pyhd8ed1ab_0    conda-forge
jeepney                   0.8.0              pyhd8ed1ab_0    conda-forge
jinja2                    3.1.2              pyhd8ed1ab_1    conda-forge
jsonpatch                 1.32               pyhd8ed1ab_0    conda-forge
jsonpointer               2.4             py310hff52083_3    conda-forge
keyring                   24.2.0          py310hff52083_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.21.2               h659d440_0    conda-forge
[... 223 lines cut ...]
supermin: kernel: looking for kernels in /lib/modules/*/vmlinuz ...
supermin: kernel: looking for kernels in /boot ...
supermin: failed to find a suitable kernel (host_cpu=x86_64).

I looked for kernels in /boot and modules in /lib/modules.

If this is a Xen guest, and you only have Xen domU kernels
installed, try installing a fullvirt kernel (only for
supermin use, you shouldn't boot the Xen guest with it).
libguestfs: error: /usr/bin/supermin exited with error status 1, see debug messages above
libguestfs: trace: launch = -1 (error)
libguestfs: trace: close
libguestfs: closing guestfs handle 0x55e4054e1770 (state 0)
libguestfs: command: run: rm
libguestfs: command: run: \ -rf /tmp/libguestfsOWrNLq
``` 

### Expected Behavior

The build setup should work.

### Other Information

Log file:
[build-setup.log](https://github.com/ucb-bar/chipyard/files/12711107/build-setup.log)

jerryz123, 2023-09-26:
Hmm, this will take some time to debug.
For now, a workaround would be to skip the firemarhshal initialization step using a flag to build-setup (use -h to see the flags).

This only prevents you from building firemarshal linux images

matsbror, 2023-09-26:
Thanks @jerryz123 , I also tried to perform the build setup in an Ubuntu 22.04 docker container and it failed in the same step.

jerryz123, 2023-09-26:
Could this be the same issue? https://github.com/steigr/docker-hipchat-server/issues/1

matsbror, 2023-09-26:
> Could this be the same issue? [steigr/docker-hipchat-server#1](https://github.com/steigr/docker-hipchat-server/issues/1)

I will try

matsbror, 2023-10-02:
Have you had a chance to look at this?

barrettsmits, 2023-11-28:
When using virt-customize on a ubuntu 22.04 WSL2 instance I was getting the same error, I had to get a Linux image installed to address the above error. 

While searching I stumbled up on this issue: [Github Issue Link.](https://github.com/steigr/docker-hipchat-server/issues/1) As my WSL env is fully ephemeral, I went ahead and installed the kernel, and got vert-customize working as expected.  

Hopefully this helps someone in need.

closed 2026-08-19 by matsbror