ucb-bar/chipyard #1607: Fail to make buildfile for CustomGemminiSoCConfig in vlsi using sky130

lu0de, 2023-09-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.1
Hash: 968b207

### OS Setup

Linux localhost.localdomain 3.10.0-1160.92.1.el7.x86_64 #1 SMP Tue Jun 20 11:48:01 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.9.2009 (Core)
Release:        7.9.2009
Codename:       Core

XDG_SESSION_ID=15408
GUESTFISH_INIT=\e[1;34m
HOSTNAME=localhost.localdomain
TERM=xterm
SHELL=/bin/bash
HISTSIZE=1000
SSH_CLIENT=10.19.102.127 14909 22
CONDA_SHLVL=1
CONDA_PROMPT_MODIFIER=(base)
QTDIR=/usr/lib64/qt-3.3
QTINC=/usr/lib64/qt-3.3/include
SSH_TTY=/dev/pts/51
QT_GRAPHICSSYSTEM_CHECKED=1
USER=luode2
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=01;05;37;41:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=01;36:*.au=01;36:*.flac=01;36:*.mid=01;36:*.midi=01;36:*.mka=01;36:*.mp3=01;36:*.mpc=01;36:*.ogg=01;36:*.ra=01;36:*.wav=01;36:*.axa=01;36:*.oga=01;36:*.spx=01;36:*.xspf=01;36:
CONDA_EXE=/home/luode2/anaconda3/bin/conda
SNPSLMD_LICENSE_FILE=27020@10.15.89.162
_CE_CONDA=
GUESTFISH_PS1=\[\e[1;32m\]><fs>\[\e[0;31m\]
MAIL=/var/spool/mail/luode2
PATH=/home/luode2/anaconda3/bin:/home/luode2/anaconda3/condabin:/usr/local/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/opt/cadence/GENUS/GENUS201/bin:/opt/cadence/INNOVUS201/bin:/opt/cadence/SSV202/bin:/opt/synopsys/vcs/R-2020.12-SP2/bin:/opt/synopsys/verdi/R-2020.12-SP2/bin:/opt/mentor/aoi_cal_2021.1_16.10/bin:/home/luode2/.local/bin:/home/luode2/bin
CONDA_PREFIX=/home/luode2/anaconda3
CALIBRE_HOME=/opt/mentor/aoi_cal_2021.1_16.10
PWD=/home/luode2
VCS_HOME=/opt/synopsys/vcs/R-2020.12-SP2
LANG=en_US.UTF-8
GUESTFISH_OUTPUT=\e[0m
_CE_M=
HISTCONTROL=ignoredups
SHLVL=1
HOME=/home/luode2
CONDA_PYTHON_EXE=/home/luode2/anaconda3/bin/python
LOGNAME=luode2
QTLIB=/usr/lib64/qt-3.3/lib
MGLS_LICENSE_FILE=1717@10.15.89.162
XDG_DATA_DIRS=/home/luode2/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share
SSH_CONNECTION=10.19.102.127 14909 10.15.88.43 22
CDS_LIC_FILE=5280@10.15.89.162
CONDA_DEFAULT_ENV=base
LESSOPEN=||/usr/bin/lesspipe.sh %s
XDG_RUNTIME_DIR=/run/user/1010
DISPLAY=localhost:11.0
GUESTFISH_RESTORE=\e[0m
_=/usr/bin/printenv

### Current Behavior

Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Dsbt.supershell=false -Djava.io.tmpdir=/home/luode2/chipyard/cy4/chipyard/.java_tmp
[info] welcome to sbt 1.8.2 (N/A Java 17.0.3-internal)
[... 98 lines cut ...]
[info]   -hir, --hammer-ir: Hammer-IR output currently only needed for IP compilers
[info]   -c, --cost-func: Cost function to use. Optional (default: "default")
[info]   -cp, --cost-param: Cost function parameter. (Optional depending on the cost function.). e.g. -c ExternalMetric -cp path /path/to/my/cost/script
[info]   --force-compile [mem]: Force the given memory to be compiled to target libs regardless of the mode
[info]   --force-synflops [mem]: Force the given memory to be compiled via synflops regardless of the mode
[info]   --mode:
[info]     default: Select the default option from below.
[info]     strict: Compile all memories to library or return an error.
[info]     synflops: Produces synthesizable flop-based memories for all memories (do not map to lib at all); likely useful for simulation purposes.
[info]     compileandsynflops: Compile all memories and create mock versions of the target libs with synflops; likely also useful for simulation purposes.
[info]     fallbacksynflops: Compile all memories to library when possible and fall back to synthesizable flop-based memories when library synth is not possible.
[info]     compileavailable: Compile all memories to library when possible and do nothing in case of errors. (default)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 1 s, completed Sep 25, 2023, 3:45:55 PM
make: *** No rule to make target '/home/luode2/chipyard/cy4/chipyard/vlsi/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/gen-collateral/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v', needed by '/home/luode2/chipyard/cy4/chipyard/vlsi/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/syn.f'.  Stop.

### Expected Behavior

Should make buildfile for CustomGemminiSoCConfig which can be synthesized.

### Other Information

I ran this command in chipyard/vlsi/
"make buildfile CONFIG=CustomGemminiSoCConfig tech_name=sky130"

lu0de, 2023-10-12:
That happens because chipyard failed to find dual-port sram to build Gemmini, while sram22 currently doesn't support generating dual-port sram (openram does).

closed 2024-08-12 by lu0de

aneskovic, 2024-08-30:
I'm facing the same problem. 

How did you resolve this issue? Were you able to generate true dual-port SRAM with OpenRAM, or did you find pre-compiled examples for sky130?