ucb-bar/chipyard #1609: Regression: Verilator emulator options no longer recognized or displayed in help text, e.g. --cycle-count, --vcd

JL102, 2023-09-27:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Affected release: 1.10.0
Affected hash: b7644b2455cc4bae190e811a5d8085f3aad85b87

Working release: 1.9.1
Working hash: 968b207ccade6f61d6f9bb2e19a54781d4a96f26

### OS Setup

```
Linux JDESK 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

SHELL=/bin/bash
NVM_INC=/home/drak/.nvm/versions/node/v18.12.1/include/node
WSL2_GUI_APPS_ENABLED=1
CONDA_EXE=/home/drak/.local/miniforge3/bin/conda
_CE_M=
WSL_DISTRO_NAME=Ubuntu
WT_SESSION=b40d23d0-abd8-44aa-b51d-565b4fc43608
JAVA_HOME=/home/drak/chipyard_origin/.conda-env/lib/jvm
JAVA_LD_LIBRARY_PATH=/home/drak/chipyard_origin/.conda-env/lib/jvm/lib/server
XML_CATALOG_FILES=file:///home/drak/chipyard_origin/.conda-env/etc/xml/catalog file:///etc/xml/catalog
NAME=JDESK
PWD=/home/drak/chipyard_origin/sims/verilator
GSETTINGS_SCHEMA_DIR=/home/drak/chipyard_origin/.conda-env/share/glib-2.0/schemas
LOGNAME=drak
CONDA_PREFIX=/home/drak/chipyard_origin/.conda-env
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
CONDA_BACKUP_RISCV=/home/drak/chipyard/esp-tools-install
HOME=/home/drak
LANG=C.UTF-8
WSL_INTEROP=/run/WSL/3328_interop
LS_COLORS=<redacted for brevity>
WAYLAND_DISPLAY=wayland-0
CONDA_PROMPT_MODIFIER=(/home/drak/chipyard_origin/.conda-env)
cbench=/home/drak/chipyard/generators/customtensor/benchmarks
NVM_DIR=/home/drak/.nvm
JAVA_LD_LIBRARY_PATH_BACKUP=
BENCH=/home/drak/chipyard/esp-tools-install/riscv64-unknown-elf/share/riscv-tests/benchmarks
LESSCLOSE=/usr/bin/lesspipe %s %s
MAKEFLAGS=-j12
TERM=xterm-256color
_CE_CONDA=
LESSOPEN=| /usr/bin/lesspipe %s
USER=drak
CONDA_SHLVL=2
DISPLAY=:0
SHLVL=1
NVM_CD_FLAGS=
CONDA_PYTHON_EXE=/home/drak/.local/miniforge3/bin/python
LD_LIBRARY_PATH=/home/drak/chipyard_origin/.conda-env/riscv-tools/lib
XDG_RUNTIME_DIR=/mnt/wslg/runtime-dir
CONDA_DEFAULT_ENV=/home/drak/chipyard_origin/.conda-env
WSLENV=WT_SESSION::WT_PROFILE_ID
JAVA_HOME_CONDA_BACKUP=
CONDA_BACKUP_PATH=<redacted for brevity>
PATH=<redacted for brevity>
[... 580 lines cut ...]

and for the emulator options to work, for example `--cycle-count` and `--vcd`:

```
$ ./simulator-chipyard-RocketConfig-debug --vcd=out.vcd --cycle-count ../../tests/hello.riscv
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 44791
[UART] UART0 is here (stdin/stdout).
Hello world from core 0, a rocket
*** PASSED *** Completed after 52394 cycles

$ ls -lah
total 210M
drwxr-xr-x 3 drak drak 4.0K Sep 27 12:48 .
drwxr-xr-x 6 drak drak 4.0K Sep 27 00:58 ..
-rw-r--r-- 1 drak drak   52 Sep 27 00:56 .gitignore
-rw-r--r-- 1 drak drak  11K Sep 27 00:58 Makefile
drwxr-xr-x 3 drak drak 4.0K Sep 27 01:50 generated-src
-rw-r--r-- 1 drak drak 195M Sep 27 13:26 out.vcd
-rwxr-xr-x 1 drak drak 8.8M Sep 27 12:46 simulator-chipyard-RocketConfig-debug
```

### Other Information

I tested this on release 1.10.0, 1.9.1, and on HEAD (from last night). 1.9.1 works as expected, 1.10.0 and HEAD do not.

jerryz123, 2023-09-27:
The verilator backend now no longer uses the `emulator.cc` driver... it now uses the same `TestDriver.v` as verilog. So that help message, as well as many of the flags it describes are out-of-date.

JL102, 2023-09-28:
> The verilator backend now no longer uses the `emulator.cc` driver... it now uses the same `TestDriver.v` as verilog. So that help message, as well as many of the flags it describes are out-of-date.

Understood. I looked at `TestDriver.v` and I see the `vcdfile` plusarg, but when trying this:
```
$ ./simulator-chipyard.harness-RocketConfig-debug +vcdfile=./out.vcd ../../tests/hello.riscv
```
I get this output:
```
[UART] UART0 is here (stdin/stdout).
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open +vcdfile=./out.vcd (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
```
How do I output a vcd?

jerryz123, 2023-09-28:
I believe the `run-binary` make target should output VCDs. The files might not have .vcd extension, but they should be VCDs IIRC?

JL102, 2023-09-28:
Got it, thanks! After passing `BINARY=../../tests/hello.riscv` into the make arguments too, it outputted a vcd and some other stuff into `output/chipyard.harness.TestHarness.RocketConfig`

JL102, 2023-09-29:
Hm... There seems to be a significant reduction in performance between the Verilator testharnesses of Chipyard 1.9.1 and Chipyard 1.10.0.

In my local copy of both v1.9.1 and v1.10.0, I ran the same `make debug CONFIG=RocketConfig`, and then after building, ran `time make run-binary-debug BINARY=../../tests/hello.riscv`.

- The 1.9.1 time took about 14 seconds, and at the bottom of `output/chipyard.harness.TestHarness.RocketConfig/hello.out`, I see that it completed after 53,235 cycles.
- The 1.10.1 time took about 46 seconds, and at the bottom of `output/chipyard.harness.TestHarness.RocketConfig/hello.out`, I see that it completed after 306,076 simulation cycles`.

Just to check and make sure the variable was not the testbench compilation, I ran 1.9.1's testharness with 1.10.0's binary and vice versa. Switching the binaries had no impact on the simulation time (1.9.1's testharness took ~45 seconds and completed after 306,076 no matter which test binary it ran)

In my own test that I'm working on in my fork, I was used to the "actual" code starting to run after about 80k-100k cycles, but after syncing up my fork with the upstream, the same test takes about 280k cycles before the "actual" code starting to run. Is it known why it's taking so much longer for the testharnesses to run after the update?

jerryz123, 2023-09-29:
Yes. We switched the verilator backend to use `--timing`, which uses a completely different simulation driver than the old mechanism. The timing-driven simulation is more faithful to real hardware, as it can simulate arbitrary clocks and delays.

To improve performance, you can try `make run-binary-debug LOADMEM=1`. This will use a fast loadmem mechanism to instantaneously copy the binary into target memory.

JL102, 2023-09-29:
Understood, thank you!

matsbror, 2023-10-02:
@JL102 I have an issue you may have a solution to (unrelated to this issue). I see you are on WSL2. Did you have a problem getting FireSim installed as part of the Chipyard setup? I can't get it to work in WSL2, but on a pure Linux host, it works.

JL102, 2023-10-03:
> @JL102 I have an issue you may have a solution to (unrelated to this issue). I see you are on WSL2. Did you have a problem getting FireSim installed as part of the Chipyard setup? I can't get it to work in WSL2, but on a pure Linux host, it works.

@matsbror Unfortunately I don't have a very satisfying solution. I was having the same issue as described in #1441, and after seeing this comment - https://github.com/ucb-bar/chipyard/issues/1441#issuecomment-1607767964 - I've been using using that command ever since. I haven't needed to use Firesim within Chipyard in my development environment.

At my workplace they have a VM built for using Firesim, and earlier this summer I was even having the same issue on the VM. (When Firesim was at the project root, with Chipyard as a submodule in `target-design/chipyard`, it was fine, but when Chipyard was at the project root with a Firesim submodule at `software/firesim`, it didn't work)

I tested it again now; when I ran the full `build-setup.sh` on `1.10.0`, the FireMarshal build step failed with some weird error I'm unsure of. I then wiped chipyard, re-cloned, and checked out `main`. Then, the full `build-setup.sh` actually worked.

jerryz123, 2023-10-04:
Can you post the full error message for failed to build workload?

BhavikOstwal, 2024-11-26:
Can someone tell me how to view cycle-counts and use verbose flag in Verilator commands?

I have chipyard v1.11.0 release version.