ucb-bar/chipyard #1613: Verilator generates a simulation executable that produces segmentation fault in v1.10.0

danivz, 2023-10-04:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: [b7644b2](https://github.com/ucb-bar/chipyard/commit/b7644b2455cc4bae190e811a5d8085f3aad85b87)

### OS Setup

```
Linux pececito 6.2.0-33-generic #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 10:33:52 UTC 2 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
SHELL=/bin/bash
SESSION_MANAGER=local/pececito:@/tmp/.ICE-unix/2234,unix/pececito:/tmp/.ICE-unix/2234
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GTK_IM_MODULE=ibus
CONDA_EXE=/home/dani/miniforge3/bin/conda
_CE_M=
LC_ADDRESS=es_ES.UTF-8
JAVA_HOME=/home/dani/work/berkeley/chipyard/.conda-env/lib/jvm
GNOME_SHELL_SESSION_MODE=ubuntu
LC_NAME=es_ES.UTF-8
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
JAVA_LD_LIBRARY_PATH=/home/dani/work/berkeley/chipyard/.conda-env/lib/jvm/lib/server
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
LC_MONETARY=es_ES.UTF-8
XML_CATALOG_FILES=file:///home/dani/work/berkeley/chipyard/.conda-env/etc/xml/catalog file:///etc/xml/catalog
GTK_MODULES=gail:atk-bridge
PWD=/home/dani/work/berkeley/chipyard
GSETTINGS_SCHEMA_DIR=/home/dani/work/berkeley/chipyard/.conda-env/share/glib-2.0/schemas
LOGNAME=dani
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
CONDA_PREFIX=/home/dani/work/berkeley/chipyard/.conda-env
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
SYSTEMD_EXEC_PID=2256
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
XAUTHORITY=/run/user/1000/gdm/Xauthority
WINDOWPATH=2
HOME=/home/dani
USERNAME=dani
LC_PAPER=es_ES.UTF-8
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6800
CONDA_PROMPT_MODIFIER=(/home/dani/work/berkeley/chipyard/.conda-env) 
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/7a4764ad_3d5a_443e_8022_73d47cb2d104
JAVA_LD_LIBRARY_PATH_BACKUP=
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
LC_IDENTIFICATION=es_ES.UTF-8
_CE_CONDA=
LESSOPEN=| /usr/bin/lesspipe %s
USER=dani
[... 450 lines cut ...]
```
- **Command 3**
```
(/home/dani/work/berkeley/chipyard/.conda-env) dani@pececito:~/work/berkeley/chipyard/sims/verilator$ ./simulator-chipyard.harness-HwachaRocketConfig-debug -h
Segmentation fault (core dumped)
```

It also happens to me with the debug simulation of a custom coprocessor (RoCC) of similar complexity as Hwacha.

### Expected Behavior

The simulation works for the `RocketConfig` and `GemminiRocketConfig`. It should work as well for `HwachaRocketConfig` and my `CustomRoCCRocketConfig`.

### Other Information

In order to compile the simulations successfully, I followed the GCC suggestion:
```
/home/dani/work/berkeley/chipyard/.conda-env/bin/x86_64-conda-linux-gnu-c++  -I.  -MMD -I/home/dani/work/berkeley/chipyard/.conda-env/share/verilator/include -I/home/dani/work/berkeley/chipyard/.conda-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O3 -std=c++17 -I/home/dani/work/berkeley/chipyard/.conda-env/esp-tools/include -I/home/dani/work/berkeley/chipyard/tools/DRAMSim2 -I/home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral   -DVERILATOR -include /home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig.plusArgs -include /home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/chipyard.harness.TestHarness.RocketConfig/VTestDriver.h -DVL_TIME_CONTEXT   -fcoroutines -Os -c -o SimDRAM.o /home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral/SimDRAM.cc
/home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral/SimDRAM.cc: In function 'void* memory_init(long long int, long long int, long long int, long long int, long long int, long long int)':
/home/dani/work/berkeley/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.RocketConfig/gen-collateral/SimDRAM.cc:64:9: error: 'endianness_t' does not name a type; did you mean 'memif_endianness_t'?
   64 |         endianness_t get_target_endianness() const override {
      |         ^~~~~~~~~~~~
      |         memif_endianness_t

```

jerryz123, 2023-10-04:
The FESVR library in esp-tools is out-of-date, and will no longer be updated. 

Switch to the `riscv-tools` conda environment, then rebuild the RTL simulations and rerun them.

If you want to compile programs for hwacha, you should maintain a separate `esp-tools` conda environment that is used *only* for compiling hwacha programs.

danivz, 2023-10-05:
I've built a clean setup executing `./build-setup.sh riscv-tools -s 6 -s 7 -s 8 -s 9` and I'm having the same problem: the simulation executable of `HwachaRocketConfig` and my `CustomRoCCRocketConfig` generate segmentation fault.

The difference with the previous issue is that I didn't have to change `endianness_t` of `fesvr/memif.h`, it doesn't produce an error with `riscv-tools`