ucb-bar/chipyard #1617: [help] An "Error: Option --top-module failed" error occurs when performing "Prototyping Flow"

luming-xyz, 2023-10-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

1. I completed the "1.4.1 Initial Repository Setup" by following the documentation 

2. Then run the dhrystone program to test the environment and get the following results
```
[UART] UART0 is here (stdin/stdout).
Microseconds for one run through Dhrystone: 451
Dhrystones per Second:                      2215
mcycle = 225682
minstret = 187526
```

3.  When I tried running "10. Prototyping Flow" I got the error

_cd fpga_
_make SUB_PROJECT=vcu118 bitstream_
```
(/data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/.conda-env) [luming@ming fpga]$ make SUB_PROJECT=vcu118 bitstream
Running with RISCV=/data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/.conda-env/riscv-tools
mkdir -p /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config
(set -o pipefail && 	cd /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard && java -cp /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/.classpath_cache/chipyard.jar chipyard.Generator  --target-dir /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config --name chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config --top-module chipyard.fpga.vcu118.VCU118FPGATestHarness --legacy-configs chipyard.fpga.vcu118:RocketVCU118Config  | tee /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config.chisel.log)
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/.java_tmp
------------------------------------------------------------------------------
Error: Option --top-module failed when given 'chipyard.fpga.vcu118.VCU118FPGATestHarness'. chipyard.fpga.vcu118.VCU118FPGATestHarness
Try --help for more information.
------------------------------------------------------------------------------
make: *** [/data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/common.mk:132: /data/prj/2_rocket_chip/1_learn/chipyard_try0926/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config.fir] Error 1
```

### Current Behavior

```
------------------------------------------------------------------------------
Error: Option --top-module failed when given 'chipyard.fpga.vcu118.VCU118FPGATestHarness'. chipyard.fpga.vcu118.VCU118FPGATestHarness
Try --help for more information.
------------------------------------------------------------------------------
```

### Expected Behavior

I have searched the relevant materials but failed to find a solution. May I ask how I should troubleshoot this problem :)

luming-xyz, 2023-10-07:
Added: Command to use dhrystone test environment:

jerryz123, 2023-10-07:
Delete Chipyard/.classpath_cache and retry?

luming-xyz, 2023-10-07:
> Delete Chipyard/.classpath_cache and retry?
It works, and now vivado is compiling, thank you very much!!!

closed 2023-10-07 by luming-xyz