ucb-bar/chipyard #1621: ERROR: Failed to build workload br-base.json

zzulb, 2023-10-11:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

### Current Behavior

DEBUG: Running: "e2fsck -f -p /chipyard/software/firemarshal/images/firechip/br-base/br-base.img" in /chipyard/software/firemarshal
DEBUG: /chipyard/software/firemarshal/images/firechip/br-base/br-base.img 已挂载。
DEBUG: e2fsck: 无法继续，已中止。
DEBUG: 
DEBUG: 
TaskError - taskid:/chipyard/software/firemarshal/images/firechip/br-base/br-base.img
PythonAction Error
Traceback (most recent call last):
  File "/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/chipyard/software/firemarshal/wlutil/build.py", line 640, in makeImage
    wlutil.resizeFS(config['img'], 0)
  File "/chipyard/software/firemarshal/wlutil/wlutil.py", line 625, in resizeFS
    raise sp.CalledProcessError(ret, " ".join(chkfsCmd))
subprocess.CalledProcessError: Command 'e2fsck -f -p /chipyard/software/firemarshal/images/firechip/br-base/br-base.img' returned non-zero exit status 8.

ERROR: Failed to build workload br-base.json
Log available at: /chipyard/software/firemarshal/logs/br-base-build-2023-10-11--13-39-44-MJ7EXEW4UMYON35S.log
ERROR: FAILURE: 1 builds failed

### Expected Behavior

No error reported, continue execution！

### Other Information

This is my first execution and the above error is displayed. I don't know how to solve it！

./marshal -v -d build br-base.json

JL102, 2023-10-14:
Did you get this error while running `./build-setup.sh`, or did the setup process work correctly? If yes to the former, then this is a duplicate of #1441. If yes to the latter, then I'm not sure.

zzulb, 2023-10-22:
Thank you very much. I have resolved this issue. I unmount the br-base.img image and then re execute the command. Now, it has been executed normally.

otoomey, 2023-11-28:
It seems that getting firemarshal to work properly is not particularly easy. I had the same "ERROR: Failed to build workload br-base.json", but for totally different reasons. 

Doing the following resolved this issue on my Ubuntu machine:
1. Ensure that `guestmount` is installed (it wasn't on my machine): `sudo apt install guestmount`
2. Modify the permissions on your kernel to allow read without root. You can do this from the file explorer.

The last step is necessary because Canonical removed read permissions from the `vmlinuz-x.x.x-generic` kernel images several years ago (See relevant report [here](https://bugs.launchpad.net/fuel/+bug/1467579)).

otoomey, 2024-01-07:
@xiaodaome if you look in that log file it should tell you where it tries, and fails, to mount the ```vmlinuz-x.x.x-generic``` kernel image, as well as the path to the image that you need to change permissions for. It is also possible that you are running into other issues, there are a number of issues that seem to trigger the ```ERROR: Failed to build workload br-base.json``` error message.

imemmaj, 2025-07-08:
What worked for me was to unmount the br-base image.
```
sudo umount /mnt/br-base

```