ucb-bar/chipyard #1623: Retire usage of deprecated SFC and use MFC exclusively

oharboe, 2023-10-14:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Retire SFC usage and use MFC exclusively. SFC is deprecated already in Chisel and mixing SFC and MFC is not supported.

### Motivating Example

The mix of SFC and MFC creates problems with dontTouch e.g. in MegaBoom: https://github.com/llvm/circt/issues/6143#issuecomment-1740224708

Another example of SFC/MFC trouble #1309

oharboe, 2023-10-19:
> What command are you running? I believe we should have isolated the SFC flow so the standard verilog generation step doesn't invoke SFC at all, but there could be a bug we overlooked

This command fails in firtool:

```
cd vlsi && make buildfile tutorial=sky130-openroad CONFIG=MegaBoomConfig
```

jerryz123, 2023-10-19:
I see. The sky-130 tutorial you are attempting relies on Yosy, which (at the time at least) had trouble parsing CIRCT-generated Verilog. Running SFC in the sky130 flow was a workaround for this. @nayiri-k or @abejgonzalez do you remember what the precise limitation was? 

Perhaps with newer CIRCT, or some emission option, we can bypass the SFC flow entirely for this tutorial. 
If you'd like, you can comment out the ENABLE_YOSYS flag in vlsi/tutorial.mk to run with a MFC-only flow, and see what happens.

oharboe, 2023-10-19:
> I see. The sky-130 tutorial you are attempting relies on Yosy, which (at the time at least) had trouble parsing CIRCT-generated Verilog.

The failure is before Yosys is invoked, the failure is in firtool. Also, Chipyard has a YOSYS_FLOW configuration option that seems to pass an option to firtool so that the Verilog from firtool is parsed by yosys without trouble.

So the only problem that needs to be solved, as far as I gather, is to make firtool work with the commandline above.

oharboe, 2023-10-19:
Trying:

```
$ git diff common.mk vlsi/
diff --git a/common.mk b/common.mk
index c8c29b1f..27956e53 100644
--- a/common.mk
+++ b/common.mk
@@ -190,7 +190,8 @@ SFC_MFC_TARGETS = \
        $(GEN_COLLATERAL_DIR)
 
 SFC_REPL_SEQ_MEM = --infer-rw --repl-seq-mem -c:$(MODEL):-o:$(SFC_SMEMS_CONF)
-MFC_BASE_LOWERING_OPTIONS ?= emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket
+# disallowPackedArrays is needed to be compatible with Yosys
+MFC_BASE_LOWERING_OPTIONS ?= emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays
 
 # DOC include start: FirrtlCompiler
 # There are two possible cases for this step. In the first case, SFC
diff --git a/vlsi/tutorial.mk b/vlsi/tutorial.mk
index 6b970fcb..d8636576 100644
--- a/vlsi/tutorial.mk
+++ b/vlsi/tutorial.mk
@@ -36,8 +36,6 @@ ifeq ($(tutorial),sky130-openroad)
                         $(if $(filter $(VLSI_TOP),RocketTile), \
                             example-designs/sky130-openroad-rockettile.yml, )
     VLSI_OBJ_DIR      ?= build-sky130-openroad
-    # Yosys compatibility for CIRCT-generated Verilog, at the expense of elaboration time.
-    ENABLE_YOSYS_FLOW  = 1
 endif
 
 INPUT_CONFS       ?= $(TOOLS_CONF) $(TECH_CONF) $(DESIGN_CONFS) $(EXTRA_CONFS)
 ```

However, it still fails:

```
firtool \
	--format=fir \
	--dedup \
	--export-module-hierarchy \
	--emit-metadata \
	--verify-each=true \
	--warn-on-unprocessed-annotations \
	--disable-annotation-classless \
	--disable-annotation-unknown \
	--mlir-timing \
	--lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays \
	--repl-seq-mem \
	--repl-seq-mem-file=/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.mems.conf \
	--repl-seq-mem-circuit=TestHarness \
	--annotation-file=/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.anno.json \
	--split-verilog \
	-o /home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/gen-collateral \
	/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: warning: unprocessed annotation:'firrtl.transforms.NoCircuitDedupAnnotation$' still remaining after LowerToHW
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: warning: unprocessed annotation:'logger.LogLevelAnnotation' still remaining after LowerToHW
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:48196:10: warning: unprocessed annotation:'freechips.rocketchip.util.RegFieldDescMappingAnnotation' still remaining after LowerToHW
  module PeripheryBus :
         ^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:287585:10: warning: unprocessed annotation:'freechips.rocketchip.util.RetimeModuleAnnotation' still remaining after LowerToHW
  module IntToFP :
         ^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:562371:10: warning: unprocessed annotation:'freechips.rocketchip.util.ParamsAnnotation' still remaining after LowerToHW
  module BoomTile :
         ^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:671534:10: warning: unprocessed annotation:'freechips.rocketchip.util.AddressMapAnnotation' still remaining after LowerToHW
[... 34 lines cut ...]
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %30 = "hw.array_concat"(%6, %29) : (!hw.array<3xi3>, !hw.array<5xi3>) -> !hw.array<8xi3>
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %29 = "hw.array_create"(%arg15, %10, %10, %arg4, %10) : (i3, i3, i3, i3, i3) -> !hw.array<5xi3>
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %35 = "hw.array_concat"(%12, %34) : (!hw.array<3xi1>, !hw.array<5xi1>) -> !hw.array<8xi1>
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %34 = "hw.array_create"(%arg14, %18, %18, %arg3, %18) : (i1, i1, i1, i1, i1) -> !hw.array<5xi1>
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
[... cut ...]
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %12 = "hw.aggregate_constant"() {fields = [false, false, false]} : () -> !hw.array<3xi1>
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %11 = "hw.aggregate_constant"() {fields = [0 : i9, 0 : i9, 0 : i9]} : () -> !hw.array<3xi9>
```

Though, I think this is might be fixed in the latest version of firtool... Testing firtool 1.53.0...

Alas not...

```

oharboe, 2023-11-09:
@jerryz123 FYI https://github.com/llvm/circt/pull/6402 fixes https://github.com/llvm/circt/issues/5355 and https://github.com/llvm/circt/issues/6324 and https://github.com/ucb-bar/chipyard/issues/1623#issuecomment-1771455376

oharboe, 2023-11-10:
@jerryz123  Trying with firtool after https://github.com/llvm/circt/pull/6402 was merged, the build now makes it further, but runs into some smaller and some bigger snags.

Deleting this option, gets it a bit further...

```
firtool: for the --lowering-options option: unknown style option 'locationInfoStyle=wrapInAtSquareBracket'
```

Deleting the first line in the .fir file, gets it a bit futher...

```
chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: expected a top-level 'circuit' definition
FIRRTL version 1.2.0
^
```

Before finally it fails on these annotations:

```
chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: unapplied annotations with target '~TestHarness|BoomTile>auto' and payload '[{class = "chisel3.experimental.EnumAnnotations$EnumVecAnnotation", fields = [["itype"]], target = ["trace_core_source_out", "group"], typeName = "freechips.rocketchip.util.TraceItype"}]'

circuit TestHarness :
^
chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: unapplied annotations with target '~TestHarness|ChipTop>jtag' and payload '[{class = "firrtl.transforms.DontTouchAnnotation", target = ["TDO"]}, {class = "firrtl.transforms.DontTouchAnnotation", target = ["TDI"]}, {class = "firrtl.transforms.DontTouchAnnotation", target = ["TMS"]}, {class = "firrtl.transforms.DontTouchAnnotation", target = ["TCK"]}]'

circuit TestHarness :
^
chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: unapplied annotations with target '~TestHarness|BoomCore>debug_brs' and payload '[{class = "firrtl.transforms.DontTouchAnnotation"}]'

circuit TestHarness :
^
chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: unapplied annotations with target '~TestHarness|BoomCore>debug_jals' and payload '[{class = "firrtl.transforms.DontTouchAnnotation"}]'
[deleted the rest]
```

seldridge, 2023-11-10:
That error looks to be using a pretty old message. However, I think it's saying that the target specified by the annotation doesn't exist. E.g., inside `circuit TestHarness`, inside `module BoomTile`, there is no port/wire/reg/node/memory/instance named `auto`.

The SFC was looser around this as it treated annotations differently. They were a separate data structure that was updated while the circuit is compiled. MFC actually inserts these annotations into the circuit, so they have to have legal values.

Also, given the old error message, I'd expect it to complain like:

```
1623.fir:1:1: error: cannot find name 'x' in Foo
circuit Foo: %[[
^
1623.fir:1:1: error: Unable to resolve target of annotation: {class = "hello", target = "~Foo|Foo>x"}
circuit Foo: %[[
^
```

This may indicate that you're on an old version of `firtool` and updating may provide either more information or may fix it (I think just the former).

oharboe, 2023-11-10:
@seldridge  To reproduce, unzip [firtool2.zip](https://github.com/ucb-bar/chipyard/files/13323800/firtool2.zip)

```
firtool       --format=fir    --export-module-hierarchy       --verify-each=true  --warn-on-unprocessed-annotations       --disable-annotation-classless  --disable-annotation-unknown    --mlir-timing   --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,disallowPackedArrays      --repl-seq-mem  --repl-seq-mem-file=chipyard.harness.TestHarness.MegaBoomConfig.mems.conf   --annotation-file=chipyard.harness.TestHarness.MegaBoomConfig.sfc.anno.json     --split-verilog     -o gen-collateral chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
```

```
$ firtool --version
LLVM (http://llvm.org/):
  LLVM version 18.0.0git
  Optimized build with assertions.
CIRCT unknown git version
(/home/oyvind/chipyard/.conda-env) $ git describe 
firtool-1.59.0-19-g97c770c09
```

```
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'debug_brs' in BoomCore
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_brs"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'debug_jals' in BoomCore
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_jals"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'debug_jalrs' in BoomCore
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_jalrs"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.out.bits"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.out.valid"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.out.ready"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.in.bits"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.in.valid"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl.bits.in.ready"}
circuit TestHarness :
[... 247 lines cut ...]
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'jtag' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>jtag.TMS"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'jtag' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>jtag.TCK"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'uart_0' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>uart_0.rxd"}
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: cannot find name 'uart_0' in ChipTop
circuit TestHarness :
^
chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:1:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>uart_0.txd"}
circuit TestHarness :
^
```

seldridge, 2023-11-11:
This looks like a situation where the annotation file is from Chisel and the FIRRTL has been run through SFC to the point of lowering types. That should produce an annotation file that has the lowered types in it. However, the original annotation file will not work.

E.g., there's this annotation targeting `debug_jals` inside module `BoomCore`:

```json
  {
    "class": "firrtl.transforms.DontTouchAnnotation",
    "target": "~TestHarness|BoomCore>debug_jals"
  }
```

Inside `BoomCore`, I see what looks like a type-lowered `debug_jals[4]` that has been split into four separate registers:

```
    reg debug_jals_0 : UInt<64>, clock with :
      reset => (UInt<1>("h0"), debug_jals_0) 
    reg debug_jals_1 : UInt<64>, clock with :
      reset => (UInt<1>("h0"), debug_jals_1) 
    reg debug_jals_2 : UInt<64>, clock with :
      reset => (UInt<1>("h0"), debug_jals_2) 
    reg debug_jals_3 : UInt<64>, clock with :
      reset => (UInt<1>("h0"), debug_jals_3) 
```

This is likely indicating that there is a problem with the Chipyard flow such that it is not sending the SFC-output annotation file to MFC.

Alternatively, and more work, moving fully to an MFC flow would fix this. We've recently added support for both custom annotations and out-of-tree passes injected at specific locations in the pipeline. This may ease the migration.

oharboe, 2023-11-12:
With the hack below, which disables the SFC step of the flow and only uses MFC, and the latest firtool with https://github.com/llvm/circt/pull/6402 and firtool completes without errors when I run:

```
make buildfile tutorial=sky130-openroad CONFIG=MegaBoomConfig
```

```
diff --git a/common.mk b/common.mk
index f6bc19c1..7023bc1e 100644
--- a/common.mk
+++ b/common.mk
@@ -28,7 +28,6 @@ EXTRA_SIM_CXXFLAGS   ?=
 EXTRA_SIM_LDFLAGS    ?=
 EXTRA_SIM_SOURCES    ?=
 EXTRA_SIM_REQS       ?=
-ENABLE_CUSTOM_FIRRTL_PASS += $(ENABLE_YOSYS_FLOW)
 
 ifneq ($(ASPECTS), )
        comma = ,
@@ -190,7 +189,8 @@ SFC_MFC_TARGETS = \
        $(GEN_COLLATERAL_DIR)
 
 SFC_REPL_SEQ_MEM = --infer-rw --repl-seq-mem -c:$(MODEL):-o:$(SFC_SMEMS_CONF)
-MFC_BASE_LOWERING_OPTIONS ?= emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket
+MFC_BASE_LOWERING_OPTIONS ?= emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays
+
 
 # DOC include start: FirrtlCompiler
 # There are two possible cases for this step. In the first case, SFC
@@ -205,13 +205,8 @@ MFC_BASE_LOWERING_OPTIONS ?= emittedLineLength=2048,noAlwaysComb,disallowLocalVa
 # hence we remove them manually by using jq before passing them to firtool
 
 $(SFC_LEVEL) $(EXTRA_FIRRTL_OPTIONS) &: $(FIRRTL_FILE)
-ifeq (,$(ENABLE_CUSTOM_FIRRTL_PASS))
-       echo $(if $(shell grep "Fixed<" $(FIRRTL_FILE)), low, none) > $(SFC_LEVEL)
+       echo none > $(SFC_LEVEL)
        echo "$(EXTRA_BASE_FIRRTL_OPTIONS)" $(if $(shell grep "Fixed<" $(FIRRTL_FILE)), "$(SFC_REPL_SEQ_MEM)",) > $(EXTRA_FIRRTL_OPTIONS)
-else
-       echo low > $(SFC_LEVEL)
-       echo "$(EXTRA_BASE_FIRRTL_OPTIONS)" "$(SFC_REPL_SEQ_MEM)" > $(EXTRA_FIRRTL_OPTIONS)
-endif
 
 $(MFC_LOWERING_OPTIONS):
        mkdir -p $(dir $@)
diff --git a/vlsi/Makefile b/vlsi/Makefile
index b3901fe6..89b5a44a 100644
--- a/vlsi/Makefile
+++ b/vlsi/Makefile
@@ -29,9 +29,9 @@ SMEMS_CACHE        ?= $(tech_dir)/sram-cache.json
 SMEMS_HAMMER       ?= $(build_dir)/$(long_name).mems.hammer.json
 
 ifdef USE_SRAM_COMPILER
-       TOP_MACROCOMPILER_MODE ?= -l $(SMEMS_COMP) --use-compiler -hir $(SMEMS_HAMMER) --mode strict
+       TOP_MACROCOMPILER_MODE ?= -l $(SMEMS_COMP) --use-compiler -hir $(SMEMS_HAMMER) --mode synflops
 else
-       TOP_MACROCOMPILER_MODE ?= -l $(SMEMS_CACHE) -hir $(SMEMS_HAMMER) --mode strict
+       TOP_MACROCOMPILER_MODE ?= -l $(SMEMS_CACHE) -hir $(SMEMS_HAMMER) --mode synflops
 endif
 
 ENV_YML            ?= $(vlsi_dir)/env.yml
```

The flow does fail later, but I think that is probably because I mangled Chipyard to get the build past the firtool error.

stefanottili, 2025-06-10:
This bug from Oct 2013 is still being referenced to in May 2015 by https://github.com/The-OpenROAD-Project/megaboom.
Has this been solved ?
```
About the MegaBoom RTL code

Based on: https://chipyard.readthedocs.io/en/stable/VLSI/Sky130-OpenROAD-Tutorial.html#initial-setup

Note: Chipyard main does not work smoothly with MegaBoom as of writing as Chipyard is mixing SFC and MFC.
Follow https://github.com/ucb-bar/chipyard/issues/1623 for latest updates.
```

closed 2025-06-10 by oharboe

oharboe, 2025-06-10:
> https://github.com/The-OpenROAD-Project/megaboom README led me here, shouldn’t that be updated ?

I think a fresh try/issue without the noise from ancient versions would clarify things.