ucb-bar/chipyard #1624: MultiSoC Configs should handle uniquification between harness and multiple ChipTops

abejgonzalez, 2023-10-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Currently, both `firtool` and `scripts/uniquify-*` support having 1 ChipTop to uniquify modules (`firtool` can only mark 1 module underneath the harness as a DUT - used to generate the corresponding json)(`scripts/uniquify-*` directly checks if the `--dut` arg matches a specific module).

This should be supported somehow.

### Motivating Example

Can't generate multi-SoC config. where modules might need to be uniquified between the harness and multiple chiptops.

jerryz123, 2023-10-16:
Shouldn't the multi-soc config tread everything that is not ChipTop0 as within the harness, from a unifuification perspectivice?

abejgonzalez, 2023-10-16:
Correct, it does. However, I think all "ChipTop{_*}"'s should be "bundled" together. Otherwise, there are modules that are uniquified incorrectly causing missing Verilog errors in VCS/Verilator.

jerryz123, 2023-10-16:
Hmmm, does the default compile for some multi-top config fail because of this? I thought the uniquification would never cause modules to be "lost" like this

abejgonzalez, 2023-10-17:
Yea I was seeing errors with my specific SoC config (fairly beefly dual-Rocket-SoC + rocc accelerators + mmio devices config). I'll try to recreate this, but this issue serves as a reminder that this is an issue.

abejgonzalez, 2023-10-25:
Also, I'm not sure this is related exactly to this issue but also comes up in a similar setup...

If you have two `addPath` (from `HasBlackBoxPath`) calls that are the same for two modules firtool errors saying "can't find <path to one of the files>" even though it exists. This can be bypassed by using `addResource` (from `HasBlackBoxResource`) which inlines the aforementioned file and then re-emits it.

jerryz123, 2023-10-25:
> If you have two addPath (from HasBlackBoxPath) calls that are the same for two modules firtool errors saying "can't find " even though it exists. This can be bypassed by using addResource (from HasBlackBoxResource) which inlines the aforementioned file and then re-emits it.

Does this happen for a single-soc build as well?

abejgonzalez, 2023-10-26:
Untested right now but this should be easy to replicate both in/out of Chipyard.

PR #1640 (2023-11-03): Fix regex for similar module names that are uniquified
When there is a module that is uniquified between the model and top, the regex in this file will improperly over-rename things. Ex: `MemLoader` is renamed to `MemLoader_UNIQUE`... however `ReverseMemLoader` would also be renamed. This PR fixes this issue by fixing the regex and adding more capture groups to properly capture spaces.

Fixes #1624 

**Related PRs / Issues**:

**Type of change**:
- [x] Bug fix
- [ ] New feature
- [ ] Other enhancement

**Impact**:
- [ ] RTL change
- [ ] Software change (RISC-V software)
- [x] Build system chan [... cut]
closed 2023-11-03 by abejgonzalez