ucb-bar/chipyard #1634: Outdated doc about Configs-Parameters-Mixins.rst

Jerry-Tianchen, 2023-10-22:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.7.0 or 1.10.0
Hash: a6a6a6

### OS Setup

N/A

### Other Setup

N/A

### Current Behavior

N/A

### Expected Behavior

N/A

### Other Information

[image]
I realized that this portion might be outdated. We don't have `val dut = p(BuildTop)(p)` anymore in version 1.10.0 or 1.7.0(The version I am working on)

Thanks~

jerryz123, 2023-10-25:
That's been pulled into `HasHarnessInstantiators`. The DUT instantiation path is a bit more complex now to support multi+heterogeneous-DUT simulation.

Jerry-Tianchen, 2023-10-25:
I see.... I think quite a lot of other docs are also out of date. I have went through the **frontend.scala** and the flow diagram below does not picture the correct frontend pipeline anymore. For example, TLB is same cycle hit rather than the next cycle hit as shown on the diagram. Fetch buffer should also be in F5 instead of F4....
https://docs.boom-core.org/en/latest/sections/instruction-fetch-stage.html