ucb-bar/chipyard #1637: firtool and GLIBC version

nishanthsmurthy24, 2023-10-26:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.9.1
Hash: 968b207

### OS Setup

Linux ece-buttonbox 3.10.0-1160.31.1.el7.x86_64 #1 SMP Wed May 26 20:18:08 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

(Date/Time is set properly)

LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: n/a
Description:    NAME="Red Hat Enterprise Linux Workstation"
Release:        n/a
Codename:       n/a

OS: RHEL 7.9

### Other Setup

Followed Documentation for all the Initial Setup. 

Downloaded firtool binary from 
https://github.com/llvm/circt/releases/tag/firtool-1.56.0

### Current Behavior

`common.mk` has firtool command, I run the `make` command inside `sims/verilator` 

firtool spits out this error
```firtool: /lib64/libc.so.6: version `GLIBC_2.18' not found (required by firtool)```

I currently have GLIBC_2.16 on RHEL 7.9. I have tried fixing the GLIBC_2.18 issue, by compiling a higher version of GLIBC. But it was not fruitful.

Is there something I am missing about firtool usage?

### Expected Behavior

Can I build the firtool binary from the source using a lower GLIBC version?

Shouldn't firtool run out of the box somehow?

jerryz123, 2023-10-26:
When using the conda install flow, which sets up a conda-managed environment and sysroot, the firtool that is installed should match the GLIBC installed by Conda. External firtools may not work in this environment.

nishanthsmurthy24, 2023-10-26:
Thanks for that. It is working now.

closed 2023-10-26 by nishanthsmurthy24

exhaust-create, 2024-03-10:
My issue is "firtool: command not found". I tried to download the binary file of firtool and glibc-2.18, but it didn't work because of many new issues. I have tried many approaches until found that firtool can be installed straightly with command `conda install firtool `.

exhaust-create, 2024-07-11:
> Thanks for that. It is working now.

Hello, could you tell me how you resolved this problem? Which version of firtool did you choose, or did you update GLIBC?

suhwanhong999, 2024-12-08:
See the solution here:
https://groups.google.com/g/chipyard/c/MIuCl8HhCIk

karthik5102001, 2024-12-10:
During the process of doing build-setup.sh skip the circt setup ( ./build-setup.sh --skip-circt) so the firtool wont be inside riscv-tools binary dirctory, so if its in that directory it will access the library of the current system you are using (which is why you are getting the error "firtool: /lib64/libc.so.6: version `GLIBC_2.18' not found (required by firtool)") so when you need to generate a RTL files for example Rocket-chip, make sure you are in conda environment and do "$ conda install firtool" then the firtool will be in the .conda-env's binary directory, hence it will access the sysroot's library files, you can check it by ldd <path to firtool> (to debug dynamic linking issues) and check where your firtool is pointing to. You can check by giving "which firtool" it should show .conda-env/bin/firtool not the .conda-env/riscv-tools/bin/firtool. This is how i cleared this issue, probably you might find even better approach for this error.