ucb-bar/chipyard #1641: Monolithic Blackbox Uniquification Not Fully Supported

abejgonzalez, 2023-11-02:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

N/A

### Other Setup

N/A

### Current Behavior

Blackbox Verilog files that are combinations of multiple Verilog files (i.e. like Ibex's single `*.preprocessed.sv` file) cannot be uniquified since the `scripts/uniquify*` script requires 1 Verilog file with 1 module definition only. 

### Expected Behavior

Uniquification either can be turned off or it can handle the situation above.

jerryz123, 2023-11-02:
The point of uniquification js to support VLSI flows that synthesize the Top, but not the Harness right? 

If so, I wonder if moving forwards, we can enforce unique module names at the Chisel level, to avoid using sed scripts to edit .sv files.

abejgonzalez, 2023-11-02:
> The point of uniquification js to support VLSI flows that synthesize the Top, but not the Harness right?

Correct. It is so that modules shared between the harness/top are unique and only top modules are synthesized.

> If so, I wonder if moving forwards, we can enforce unique module names at the Chisel level, to avoid using sed scripts to edit .sv files.

This seems tricky to enforce. Also isn't this separate from this issue, no?

jerryz123, 2023-11-02:
> This seems tricky to enforce. Also isn't this separate from this issue, no?

I'm not sure... there might be some chisel naming scope API that can prepend/append a prefix to the module names of all modules instantiated within the scope. Sounds like something that could/should exist. If it does exist, we can just wrap all the HarnessBinder instantiation within this "name scope".

If we had that... then we wouldn't need the uniquification script, and your multi-top sims would work correctly, even when a monolithic blackbox is instantiated in multiple places.

abejgonzalez, 2023-11-06:
Posted something on the gitter for this: https://matrix.to/#/!jSbgpvvpsPbLpsLZOC:matrix.org/$YW8mWMqiEN67_Q69TmeTRIUA_SN_LR1HdfyPK9JjELU?via=matrix.org&via=gitter.im&via=austin-harris.com

jackkoenig, 2023-11-07:
Responded on Gitter, we would like to add a module prefixing API as a first class feature in Chisel but there are a few minor speedbumps. There is an existing but undocumented API via FIRRTL annotations in firtool, you can use https://github.com/llvm/circt/blob/main/docs/Dialects/FIRRTL/FIRRTLAnnotations.md#nestedprefixmodulesannotation.

abejgonzalez, 2023-11-08:
Thanks. I'll mess around with it and see if it works nicely for our use case. Thanks again

lsteveol, 2023-11-08:
I would be interested in this as well if you get something working @abejgonzalez

abejgonzalez, 2023-11-09:
For the record, I added the annotation here: https://github.com/ucb-bar/chipyard/blob/8b6adb1525625f3b5bc0cfe84b79a99822e1ef85/common.mk#L147 targeting ChipTop and it seemed to work great. However, we still need to think a bit more on how BB's interact with our flows Chipyard/Hammer before implementing it and upstreaming it.