ucb-bar/chipyard #1656: Building Linux with FireMarshal: invalid password for 'root' on 'console'

luming-xyz, 2023-11-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: b7644b2455cc4bae190e811a5d8085f3aad85b87

### OS Setup

when building linux with FIreMarshal, Some errors occurred while signing in as' root 'with password' FPGA '.

1. Setup FireMarshal 
cd software/fireMarshal
./init-submodules.sh
2. echo "board-dir : 'boards/prototype'" > $PATH_TO_FIREMARSHAL/marshal-config.yaml
3. ./marshal -v -d build br-base.json # here the -d indicates --nodisk or initramfs
4. ./marshal -v -d install -t prototype br-base.json
5. spike images/prototype/br-base/br-base-bin-nodisk

### Current Behavior

OpenSBI v1.2-78-g5ccebf0
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|___/_____|
        | |
        |_|

Platform Name             : ucbbar,spike-bare
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 10000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : htif
Platform Shutdown Device  : htif
Platform Suspend Device   : ---
Firmware Base             : 0x80000000
Firmware Size             : 232 KB
Firmware RW Offset        : 0x20000
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*
Domain0 Region00          : 0x0000000080000000-0x000000008001ffff M: (R,X) S/U: ()
Domain0 Region01          : 0x0000000080020000-0x000000008003ffff M: (R,W) S/U: ()
Domain0 Region02          : 0x0000000002080000-0x00000000020bffff M: (I,R,W) S/U: ()
Domain0 Region03          : 0x0000000002000000-0x000000000207ffff M: (I,R,W) S/U: ()
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000081800000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.12
Boot HART Base ISA        : rv64imafdc
Boot HART ISA Extensions  : time
[... 205 lines cut ...]
launching firemarshal workload run/command
firemarshal workload run/command done

Welcome to Buildroot
buildroot login: root
root
Password: fpga

Login incorrect
login[90]: invalid password for 'root' on 'console'
buildroot login: root
root
Password: fpga

Login incorrect
login[90]: invalid password for 'root' on 'console'
buildroot login: 
Login timed out after 60 seconds

Welcome to Buildroot
buildroot login: 

### Expected Behavior

The same problem occurred in[#1558](https://github.com/ucb-bar/chipyard/issues/1558) .but the root cause was not found. Do you have any suggestions for this issue？thankyou :)

luming-xyz, 2023-11-08:
I found a way to avoid it by modifying BR2 in fireMarshal/prototype/base workloads/br base/builderroot configuration_ TARGET_ GENERIC_ ROOT_ PASSWD="" to clear the password. And passed the testing of spike and FPGA prototypes.
But the root cause of the problem has not been found yet, is it due to coding issues or something else?

T-K-233, 2024-01-07:
Encountered same issue here. Setting the password to `""` hack works.

jerryz123, 2024-01-07:
Can someone PR removing the password to Firemarshal?