ucb-bar/chipyard #1658: IOBinders.scala:428:53: no arguments allowed for nullary constructor BlockDeviceIO

oharboe, 2023-11-09:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

```
$ cd vlsi
$ make buildfile tutorial=sky130-openroad
Running with RISCV=/home/oyvind/chipyard/.conda-env/riscv-tools
mkdir -p /home/oyvind/chipyard/.classpath_cache/
cd /home/oyvind/chipyard && java -jar /home/oyvind/chipyard/scripts/sbt-launch.jar -Dsbt.ivy.home=/home/oyvind/chipyard/.ivy2 -Dsbt.global.base=/home/oyvind/chipyard/.sbt -Dsbt.boot.directory=/home/oyvind/chipyard/.sbt/boot/ -Dsbt.color=always -Dsbt.supershell=false -Dsbt.server.forcestart=true ";project chipyard; set assembly / assemblyOutputPath := file(\"/home/oyvind/chipyard/.classpath_cache/chipyard.jar\"); assembly" && touch /home/oyvind/chipyard/.classpath_cache/chipyard.jar
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/oyvind/chipyard/.java_tmp
[info] welcome to sbt 1.8.2 (N/A Java 20-internal)
[info] loading settings for project chipyard-build-build-build from metals.sbt ...
[info] loading project definition from /home/oyvind/chipyard/project/project/project
[info] loading settings for project chipyard-build-build from metals.sbt ...
[info] loading project definition from /home/oyvind/chipyard/project/project
[success] Generated .bloop/chipyard-build-build.json
[success] Total time: 0 s, completed Nov 9, 2023, 12:43:21 PM
[info] loading settings for project chipyard-build from metals.sbt,plugins.sbt ...
[info] loading project definition from /home/oyvind/chipyard/project
[success] Generated .bloop/chipyard-build.json
[success] Total time: 0 s, completed Nov 9, 2023, 12:43:22 PM
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project barf from build.sbt ...
[info] loading settings for project constellation from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project tapeout from build.sbt ...
[info] loading settings for project fixedpoint from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/oyvind/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (39519 settings) ...
[info] set current project to chipyardRoot (in build file:/home/oyvind/chipyard/)
[info] set current project to chipyard (in build file:/home/oyvind/chipyard/)
[info] Defining assembly / assemblyOutputPath
[info] The new value will be used by assembly
[info] Reapplying settings...
[info] set current project to chipyard (in build file:/home/oyvind/chipyard/)
[info] compiling 21 Scala sources to /home/oyvind/chipyard/generators/chipyard/target/scala-2.13/classes ...
[error] /home/oyvind/chipyard/generators/chipyard/src/main/scala/iobinders/IOBinders.scala:428:53: no arguments allowed for nullary constructor BlockDeviceIO: ()(implicit p: org.chipsalliance.cde.config.Parameters): testchipip.BlockDeviceIO
[error]       val port = IO(new ClockedIO(new BlockDeviceIO(bdParams))).suggestName("blockdev")
[error]                                                     ^
[error] /home/oyvind/chipyard/generators/chipyard/src/main/scala/harness/HarnessBinders.scala:84:60: type mismatch;
[error]  found   : icenet.NICIOvonly
[error]  required: Option[icenet.NICIOvonly]
[error]     withClock(port.io.clock) { NicLoopback.connect(port.io.bits, port.params) }
[error]                                                            ^
[error] /home/oyvind/chipyard/generators/chipyard/src/main/scala/harness/HarnessBinders.scala:84:71: type mismatch;
[error]  found   : icenet.NICConfig
[error]  required: Option[icenet.NICConfig]
[error]     withClock(port.io.clock) { NicLoopback.connect(port.io.bits, port.params) }
[error]                                                                       ^
[error] /home/oyvind/chipyard/generators/chipyard/src/main/scala/harness/HarnessBinders.scala:66:53: no arguments allowed for nullary constructor SimBlockDevice: ()(implicit p: org.chipsalliance.cde.config.Parameters): testchipip.SimBlockDevice
[error]     val sim_blkdev = Module(new SimBlockDevice(port.params))
[error]                                                     ^
[error] /home/oyvind/chipyard/generators/chipyard/src/main/scala/harness/HarnessBinders.scala:75:61: too many arguments (found 2, expected 1) for constructor BlockDeviceModel: (nSectors: Int)(implicit p: org.chipsalliance.cde.config.Parameters): testchipip.BlockDeviceModel
[error]     val blkdev_model = Module(new BlockDeviceModel(16, port.params))
[error]                                                             ^
[warn] /home/oyvind/chipyard/generators/chipyard/src/main/scala/iobinders/IOBinders.scala:86:52: non-variable type argument chisel3.Data in type pattern chipyard.iobinders.Port[chisel3.Data] is unchecked since it is eliminated by erasure
[warn]       portMap.values.flatten.foreach { case (port: Port[Data]) => dontTouch(port.io) }
[warn]                                                    ^
[warn] /home/oyvind/chipyard/generators/chipyard/src/main/scala/iobinders/Ports.scala:7:35: imported `I2CPort` is permanently hidden by definition of class I2CPort in package iobinders
[warn] import sifive.blocks.devices.i2c.{I2CPort}
[warn]                                   ^
[warn] two warnings found
[error] 5 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 6 s, completed Nov 9, 2023, 12:43:34 PM
make: *** No rule to make target '/home/oyvind/chipyard/.classpath_cache/chipyard.jar', needed by '/home/oyvind/chipyard/.classpath_cache/tapeout.jar'.  Stop.
```

### Expected Behavior

No error

oharboe, 2023-11-09:
@jerryz123 FYI

jerryz123, 2023-11-09:
Have you updated your submodules after bumping chipyard? If not, can you rerun `scripts/init-submodules-no-riscv-tools.sh` again in the root of your repo?

oharboe, 2023-11-09:
> Have you updated your submodules after bumping chipyard? If not, can you rerun `scripts/init-submodules-no-riscv-tools.sh` again in the root of your repo?

Same result.

jerryz123, 2023-11-09:
Can you run `git status` in the root of your repo and paste the output here?

oharboe, 2023-11-10:
> Can you run `git status` in the root of your repo and paste the output here?

Running `git status` and  `git submodule foreach git status` and cleaning out  everything and checking out latest origin/main fixed it.

Thanks!

closed 2023-11-10 by oharboe