ucb-bar/chipyard #1659: No Object name Hwacha

HakamAtassi, 2023-11-10:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: b7644b24

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

I'm trying to get Hwacha added to the base Rocket-chip config provided in chipyard. 
Based on 

https://chipyard.readthedocs.io/en/stable/Generators/Hwacha.html 

adding 
``` hwacha.DefaultHwachaConfig```
 to 

chipyard/generators/rocket-chip/src/main/scala/system/Configs.scala under 

class BaseConfig extends Config(
  new WithDefaultMemPort ++
  new WithDefaultMMIOPort ++
  new WithDefaultSlavePort ++
  new WithTimebase(BigInt(1000000)) ++ // 1 MHz
  new WithDTS("freechips,rocketchip-unknown", Nil) ++
  new WithNExtTopInterrupts(2) ++
  new hwacha.DefaultHwachaConfig ++                          // <= NEW 
  new BaseSubsystemConfig 
)

Should work 

### Current Behavior

Running  

make VERILATOR_THREADS=8 NUMACTL=1

in generators/rocket-chip/emulator 

causes 

[error] /home/hakam/Repos/chipyard/generators/rocket-chip/src/main/scala/system/Configs.scala:21:7: not found: value hwacha
[error]   new hwacha.DefaultHwachaConfig ++
[error]       ^
[error] one error found

Which makes sense, since hwacha is not really defined anywhere. 

Now the question is, how and where is hwacha supposed to be defined?

adding 

lazy val hwacha = (project in file("hwacha"))
  .settings(commonSettings)
  .settings(publishArtifact := false)

to build.sbt has not helped

I'm not entirely sure how to proceed. I have not been able to find other repos that publicly use hwacha nor have I found any documentation on adding hwacha to a rocket core within chipyard that demonstrates how to instantiate a "hwacha" object. Rocket-chip by default uses hardfloat, but hardfloat is already included within the rocket-chip file, where as hwacha is not.

I have also tried moving the hwacha file to the Rocket-chip file, but that just causes all sorts of package namespace issues, and am also pretty confident that that is not the correct way to instantiate a hwacha module. 

Sort of stumped. Would really appreciate some help. Thanks

### Expected Behavior

verilator make runs fine and executres custom hwacha opcodes/tests.

jerryz123, 2023-11-10:
Without making any modifications to Chipyard or rocket chip, run `make CONFIG=HwachaRocketConfig`. 

https://chipyard.readthedocs.io/en/stable/Generators/Hwacha.html

HakamAtassi, 2023-11-10:
Thanks for the quick response. Running make CONFIG=HwachaRocketConfig successfully builds the the .jar but seemlingly not the emulator:

[image]

The make is to be run in rocket-chip/emulator, correct?

jerryz123, 2023-11-10:
No. Please follow the Chipyard documentation, which instructs you to run it in Chipyard/sims/verilator

HakamAtassi, 2023-11-10:
[image]

It seems to be segmenting on a test that normally runs sucessfully on a default rocketchip. I recloned the repo to verify that this was not a result of a config that I modified. Same problem. 

I referred back to the docs and retried the entire build using esp tools (which I believe is correct) and now have this issue: 

[image]

Which kills the make. I do actually think this is a typo. I fixed the type names manually and continued the make process. It completes but causes the same segmentation issue, even when I build the hello world example in 

https://chipyard.readthedocs.io/en/stable/Software/Baremetal.html

Any further suggestions? I appreciate the help. Thanks.

HakamAtassi, 2023-11-10:
Ended up building a custom config with rocket and hwacha using the guide provided here: 

https://fires.im/micro19-slides-pdf/03_building_custom_socs.pdf

Which seems to work. Thanks.

closed 2023-11-10 by HakamAtassi