ucb-bar/chipyard #1663: missing submodule

Gorgeous0215, 2023-11-14:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

https://github.com/ucb-bar/hammer-mentor-plugins.git is missing ,so i can't git submodules init, is that a problem?

### Expected Behavior

if the submodule doesn't matter maybe delete it ?

harrisonliew, 2023-11-14:
The submodule is needed if you are using the VLSI flow with Siemens Mentor Graphics tools, and it is a private repository that [access must be requested ](https://chipyard.readthedocs.io/en/stable/VLSI/Basic-Flow.html) for. In the normal way that the repo is supposed to be set up (i.e. do not `git submodule update -init`), you will not encounter a problem with this missing submodule.

closed 2023-11-14 by harrisonliew

Gorgeous0215, 2023-11-15:
then what should i do to get submodule init

harrisonliew, 2023-11-15:
Please refer to the documentation for how to set up the repository: https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#initial-repository-setup