ucb-bar/chipyard #1666: Error: Could not find or load main class `.Generator`

T-K-233, 2023-11-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

```bash
> uname -a
Linux PaperWeight 6.2.0-36-generic #37~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct  9 15:34:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
> lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
```

```bash
> printenv
SHELL=/bin/bash
SESSION_MANAGER=local/PaperWeight:@/tmp/.ICE-unix/1624,unix/PaperWeight:/tmp/.ICE-unix/1624
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
TERM_PROGRAM_VERSION=1.84.2
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GTK_IM_MODULE=ibus
CONDA_EXE=/home/tk/Documents/mambaforge/bin/conda
_CE_M=
JAVA_HOME=/home/tk/Desktop/chipyard/.conda-env/lib/jvm
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
JAVA_LD_LIBRARY_PATH=/home/tk/Desktop/chipyard/.conda-env/lib/jvm/lib/server
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
XML_CATALOG_FILES=file:///home/tk/Desktop/chipyard/.conda-env/etc/xml/catalog file:///etc/xml/catalog
GTK_MODULES=gail:atk-bridge
PWD=/home/tk/Desktop/chipyard
GSETTINGS_SCHEMA_DIR=/home/tk/Desktop/chipyard/.conda-env/share/glib-2.0/schemas
XDG_SESSION_DESKTOP=ubuntu
LOGNAME=tk
XDG_SESSION_TYPE=x11
CONDA_PREFIX=/home/tk/Desktop/chipyard/.conda-env
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
SYSTEMD_EXEC_PID=1644
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
XAUTHORITY=/run/user/1000/gdm/Xauthority
VSCODE_GIT_ASKPASS_NODE=/usr/share/code/code
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
WINDOWPATH=2
HOME=/home/tk
USERNAME=tk
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=Unity
CONDA_PROMPT_MODIFIER=(/home/tk/Desktop/chipyard/.conda-env) 
[... 466 lines cut ...]
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstandard                 0.22.0          py310h1275a96_0    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
```

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

```bash
Running with RISCV=/home/tk/Desktop/chipyard/.conda-env/riscv-tools
mkdir -p /home/tk/Desktop/chipyard/fpga/generated-src/..
(set -o pipefail &&     cd /home/tk/Desktop/chipyard && java -cp /home/tk/Desktop/chipyard/.classpath_cache/chipyard.jar .Generator  --target-dir /home/tk/Desktop/chipyard/fpga/generated-src/.. --name .. --top-module . --legacy-configs :   | tee /home/tk/Desktop/chipyard/fpga/generated-src/../...chisel.log)
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/tk/Desktop/chipyard/.java_tmp
Error: Could not find or load main class .Generator
Caused by: java.lang.ClassNotFoundException: /Generator
make: *** [/home/tk/Desktop/chipyard/common.mk:137: /home/tk/Desktop/chipyard/fpga/generated-src/../...fir] Error 1
```

### Expected Behavior

Chipyard should compile the chip source code and generate FPGA bitstream

jerryz123, 2023-11-16:
Delete chipyard/.classpath_cache and retry.

T-K-233, 2023-11-16:
Oh I'm dumb...

For FPGA it should be `SUB_PROJECT=[arty|arty100t]`

so

```bash
make SUB_PROJECT=arty bitstream
```

Then it raises the following error

```bash
> make SUB_PROJECT=arty100t bitstream
Running with RISCV=/home/tk/Desktop/chipyard/.conda-env/riscv-tools
mkdir -p /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig
cp -f /home/tk/Desktop/chipyard/generators/testchipip/src/main/resources/testchipip/bootrom/bootrom.rv64.img /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig/bootrom.rv64.img
cp -f /home/tk/Desktop/chipyard/generators/testchipip/src/main/resources/testchipip/bootrom/bootrom.rv32.img /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig/bootrom.rv32.img
mkdir -p /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig
(set -o pipefail &&     cd /home/tk/Desktop/chipyard && java -cp /home/tk/Desktop/chipyard/.classpath_cache/chipyard.jar chipyard.Generator  --target-dir /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig --name chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig --top-module chipyard.fpga.arty100t.Arty100THarness --legacy-configs chipyard.fpga.arty100t:RocketArty100TConfig   | tee /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig.chisel.log)
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/tk/Desktop/chipyard/.java_tmp
------------------------------------------------------------------------------
Error: Option --top-module failed when given 'chipyard.fpga.arty100t.Arty100THarness'. chipyard.fpga.arty100t.Arty100THarness
Try --help for more information.
------------------------------------------------------------------------------
make: *** [/home/tk/Desktop/chipyard/common.mk:137: /home/tk/Desktop/chipyard/fpga/generated-src/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig/chipyard.fpga.arty100t.Arty100THarness.RocketArty100TConfig.fir] Error 1
```

And deleting the .classpath_cache directory solves the error:

```bash
rm -rf ./chipyard/.classpath_cache
```

closed 2023-11-16 by T-K-233