ucb-bar/chipyard #1673: make CONFIG=SmallBoomConfig

aoliaoICer, 2023-11-29:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.8.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

An error is reported when make CONFIG=SmallBoomConfig is reported in chipyard
[error] Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/odin/chipyard2/chipyard/.java_tmp
[error] Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! Please file an issue at github.com/ucb-bar/firr
[error] at firrtl.Utils$.error(Utils.scala:471)
[error] at firrtl.Utils$.throwInternalError(Utils.scala:175)
[error] at firrtl.stage.phases.CatchExceptions.transform(CatchExceptions.scala:31)
...59more
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 107 s (01:47), completed Nov 29, 2023, 2:27:54 PM
make: *** [/home/odin/chipyard2/chipyard/common.mk:137: firrtl_temp] Error 1

### Expected Behavior

To resolve this bug, configure the boom environment

jerryz123, 2024-01-04:
This is unusual. Can you try with the latest Chipyard?

waseem-10xe, 2024-08-09:
make CONFIG=SmallBoomV3Config
kindly use the above commands errors are resolved.Thanks

Shahed-ur-Rahman, 2025-03-12:
Hello, Wassem. Can you please  share the SmallBoom RTL code?

> make CONFIG=SmallBoomV3Config kindly use the above commands errors are resolved.Thanks

Shahed-ur-Rahman, 2025-07-10:
> File "/home/shahed/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute returned_value = self.py_callable(*self.args, **kwargs) File "/home/shahed/chipyard/software/firemarshal/wlutil/build.py", line 603, in makeImage wlutil.applyOverlay(config['img'], config['overlay']) File "/home/shahed/chipyard/software/firemarshal/wlutil/wlutil.py", line 753, in applyOverlay copyImgFiles(img, flist, 'in') File "/home/shahed/chipyard/software/firemarshal/wlutil/wlutil.py", line 702, in copyImgFiles with mountImg(img, getOpt('mount-dir')): File "/home/shahed/chipyard/.conda-env/lib/python3.10/contextlib.py", line 135, in **enter** return next(self.gen) File "/home/shahed/chipyard/software/firemarshal/wlutil/wlutil.py", line 629, in mountImg run(['guestmount', '--pid-file', pidPath, '-o', f'uid={uid}', '-o', f'gid={gid}', '-a', imgPath, '-m', '/dev/sda', mntPath]) File "/home/shahed/chipyard/software/firemarshal/wlutil/wlutil.py", line 521, in run p = sp.Popen(*args, universal_newlines=True, stderr=sp.STDOUT, stdout=sp.PIPE, **kwargs) File "/home/shahed/chipyard/.conda-env/lib/python3.10/subprocess.py", line 971, in **init** self._execute_child(args, executable, preexec_fn, close_fds, File "/home/shahed/chipyard/.conda-env/lib/python3.10/subprocess.py", line 1863, in _execute_child raise child_exception_type(errno_num, err_msg, err_filename) FileNotFoundError: [Errno 2] No such file or directory: 'guestmount'
> 
> ERROR: Failed to build workload br-base.json Log available at: /home/shahed/chipyard/software/firemarshal/logs/br-base-build-2025-07-10--11-10-28-RYS6U6EZR1345DV1.log ERROR: FAILURE: 1 builds failed build-setup.sh: Build script failed with exit code 1 at step 9: Pre-compiling FireMarshal buildroot sources
> 
> What is the reason for this failure?

Solved