ucb-bar/chipyard #1677: Add Python support for uart_tsi

T-K-233, 2023-12-01:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Add Python support for uart_tsi so that the bringup program can run on Windows/Mac. 

### Motivating Example

Use pyserial to enable unified UART interface in Python program. This will also allow us to interface with other applications running on the host PC, such as RoSE simulations.

jerryz123, 2023-12-01:
Yes a python uart_tsi library would be good.

T-K-233, 2023-12-29:
A preliminary implementation for those who need the functionality now:

```python
import struct

import serial

port = "/dev/ttyUSB3"
baudrate = 115200

ser = serial.Serial(port=port, baudrate=baudrate)

CMD_READ = 0x00
CMD_WRITE = 0x01

def readWord(address):
    command = CMD_READ
    header = struct.pack("<LQQ", command, address, 0)
    ser.write(header)

    received = ser.read(4)
    rx_data, = struct.unpack("<L", received)
    return rx_data

def writeWord(address, data):
    command = CMD_WRITE
    header = struct.pack("<LQQ", command, address, 0)
    payload = struct.pack("<L", data)
    buffer = header + payload
    ser.write(buffer)

if __name__ == "__main__":

    writeWord(0x08000000, 0xdeadbeef)
    
    print(hex(readWord(0x08000000)))

```