ucb-bar/chipyard #1693: java.lang.OutOfMemoryError: Java heap space

csgxiong, 2023-12-13:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

ubuntu 18

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I try to add vector instructions to rocket-chip decode_table using chipyard V1.10.0.
But it throws "OutOfMemoryError: Java heap space" error when using "make verilog CONFIG=RocketConfig".
I have add '-Xms512M -Xmx64G -Xss8M' option to JAVA_TOOL_OPTIONS, but the errors are still the same.
If I delete some instructions from vector decode_table,such as "vwsub_vx", the verilog can be generated sucessfully.

### Expected Behavior

Anyone knowns how to fix this ?

jerryz123, 2023-12-13:
Try using the JAVA_HEAP_SIZE flag

csgxiong, 2023-12-13:
> JAVA_HEAP_SIZE
Thanks for your reply.

 I have modified  the  JAVA_HEAP_SIZE and JAVA_TOOL_OPTIONS at vaiiables.mk. but the errors are still the same.
I did a little experiment that  If I deleted some instructions from the vector decode_table,such as "vwsubu_vx / vmaddu_vx / vmfle_vf", the verilog could be generated sucessfully.
It seems some specific instructions may cause these errors
So, I'm confused.

csgxiong, 2023-12-19:
The problem has been solved when I increased JAVA_HEAP_SIZE to 512G.

closed 2023-12-19 by csgxiong