ucb-bar/chipyard #1708: Request for example to connect normal UART to IO pin on Arty100T FPGA

T-K-233, 2023-12-19:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Could someone provide an example of connecting the normal UART and JTAG pins on the Arty100T design?

### Motivating Example

Something similar to the one in Arty 35T:

```scala

class WithArtyUARTHarnessBinder extends HarnessBinder({
  case (th: Arty35THarness, ports: UARTPort) => {
    withClockAndReset(th.clock_32MHz, th.ck_rst) {
      IOBUF(th.uart_rxd_out,  ports.io.txd)
      ports.io.rxd := IOBUF(th.uart_txd_in)    
    }
  }
})

class WithArtyJTAGHarnessBinder extends HarnessBinder({
  case (th: Arty35THarness, port: JTAGPort) => {
    val jtag_wire = Wire(new JTAGIO)
    jtag_wire.TDO.data := port.io.TDO
    jtag_wire.TDO.driven := true.B
    port.io.TCK := jtag_wire.TCK
    port.io.TMS := jtag_wire.TMS
    port.io.TDI := jtag_wire.TDI

    val io_jtag = Wire(new JTAGPins(() => new BasePin(), false)).suggestName("jtag")

    JTAGPinsFromPort(io_jtag, jtag_wire)

    io_jtag.TCK.i.ival := IBUFG(IOBUF(th.jd_2).asClock).asBool

    IOBUF(th.jd_5, io_jtag.TMS)
    PULLUP(th.jd_5)

    IOBUF(th.jd_4, io_jtag.TDI)
    PULLUP(th.jd_4)

    IOBUF(th.jd_0, io_jtag.TDO)

    // mimic putting a pullup on this line (part of reset vote)
    th.SRST_n := IOBUF(th.jd_6)
    PULLUP(th.jd_6)

    // ignore the po input
    io_jtag.TCK.i.po.map(_ := DontCare)
    io_jtag.TDI.i.po.map(_ := DontCare)
    io_jtag.TMS.i.po.map(_ := DontCare)
    io_jtag.TDO.i.po.map(_ := DontCare)
  }
})
```

jerryz123, 2023-12-19:
The arty100t design uses the on-board USB-UART for the UART-TSI link. 
To support another UART/JTAG interface, we'd need to allocate one of the PMODs.
Which PMOD would you prefer?

T-K-233, 2023-12-19:
Could we use the following mapping on PMOD JD?

- JD0 - TDO
- JD1 - (nTRST)
- JD2 - TCK
- JD3 - UART_TXD
- JD4 - TDI
- JD5 - TMS
- JD6 - (nRESET)
- JD7 - UART_RXD

T-K-233, 2023-12-19:
Also here's the bump signal - pin names mapping:
[image]

jerryz123, 2023-12-20:
@T-K-233 can you fix the mapping in what you posted? It looks like the available pins are 1-4, and 7-10, not 0-7.

T-K-233, 2023-12-20:
Oh, I'm using the Chisel port IO index. Corrected above.

closed 2023-12-30 by T-K-233