ucb-bar/chipyard #1713: How to mount the second partition on SD card with buildroot Linux

T-K-233, 2023-12-21:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: 32f79a8

### OS Setup

```
doesn't matter?
```

### Other Setup

Following [this](https://chipyard.readthedocs.io/en/stable/Prototyping/VCU118.html#setting-up-the-sdcard) documentation.
The Linux can boot up, but it cannot discover the second partition. 

### Current Behavior

`ls /dev/` does not show any sd* or mmcblk* entry

### Expected Behavior

Be able to read contents in the second partition.

Chenning-Tao, 2024-01-11:
I have the same problem

wadidf, 2024-09-22:
Some tips to help :

- **Device tree (DTS):** Check in the SPI block that you have a mmc-slot there.
Example (the Lxx labels might have different numbers in your DTS file):
`
L29: spi@64001000 {
			#address-cells = <1>;
			#size-cells = <0>;
			clocks = <&L0>;
			compatible = "sifive,spi0";
			interrupt-parent = <&L19>;
			interrupts = <2>;
			reg = <0x64001000 0x1000>;
			reg-names = "control";
			L37: mmc@0 {
				compatible = "mmc-spi-slot";
				disable-wp;
				reg = <0x0>;
				spi-max-frequency = <1000000>;
				voltage-ranges = <3300 3300>;
			};
		};
`

- **in Firemarshal:** check your linux configuration file (linux-config) and be sure that the options below are active (or add/change them) to :
`
CONFIG_MMC=y
CONFIG_SPI=y
CONFIG_EXT2_FS=y
`

panjin12345, 2025-07-06:
> Some tips to help :
>
> * **Device tree (DTS):** Check in the SPI block that you have a mmc-slot there.
>   Example (the Lxx labels might have different numbers in your DTS file):
>
> ```
> L29: spi@64001000 [...]

Hello, thank you for your answer. First, I have the same problem. Second, I add CONFIG_MMC=y, but I can't even get into the linux system. I'm stuck here.

[image]

Biancaa-R, 2026-05-05:
> > Some tips to help :
> > 
> > * **Device tree (DTS):** Check in the SPI block that you have a mmc-slot there.
> >   Example (the Lxx labels might have different numbers in your DTS file):
> > 
> > ```
> > L29: spi@64001000 {
> > 			#address-cells = <1>;
> > 			#size-cells = <0>;
> > 			clocks = <&L0>;
> > 			compatible = "sifive,spi0";
> > 			interrupt-parent = <&L19>;
> > 			interrupts = <2>;
> > 			reg = <0x64001000 0x1000>;
> > 			reg-names = "control";
> > 			L37: mmc@0 {
> > 				compatible = "mmc-spi-slot";
> > 				disable-wp;
> > 				reg = <0x0>;
> > 				spi-max-frequency = <1000000>;
> > 				voltage-ranges = <3300 3300>;
> > 			};
> > 		};
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * **in Firemarshal:** check your linux configuration file (linux-config) and be sure that the options below are active (or add/change them) to :
> > 
> > ```
> > CONFIG_MMC=y
> > CONFIG_SPI=y
> > CONFIG_EXT2_FS=y
> > ```
> 
> Hello, thank you for your answer. First, I have the same problem. Second, I add CONFIG_MMC=y, but I can't even get into the linux system. I'm stuck here.
> <img alt="Image" width="623" height="810" src="https://private-user-images.githubusercontent.com/141385610/462889832-a8d85da3-5a18-441d-a0d5-ef6774804a66.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Nzc5NTM3OTIsIm5iZiI6MTc3Nzk1MzQ5MiwicGF0aCI6Ii8xNDEzODU2MTAvNDYyODg5ODMyLWE4ZDg1ZGEzLTVhMTgtNDQxZC1hMGQ1LWVmNjc3NDgwNGE2Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwNTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDUwNVQwMzU4MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NTNjZmQ5NjJmYWZmZmM5OGVjYTkwODRiYzljMTg5NGNjZWQ2ZGNiMzBhOWRiNDdjOGQzMWI3NDUyNmUwNDUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZwbmcifQ.vzJLRLR7OWqyixzRrGCPQRfrKa8d3jBd_5clPDKQSjQ">

I think your issue is something else as even u-boot proper is not showing up and the proceeding of the boot is not happening in the system here . The u boot version should be displayed (provided you are using it ) . Incase of direct jump to kernel (lets say you dont use any DDR and just PSRAM or SRAM ) ,The issue could be the console device ,if its not mapped correctly hvc0 or ttyS0 if you have drivers for it in your system.